hoeken/PsychicHttp #1: Memory leak

ninja-, 2023-11-24:
Continuing discussion https://github.com/me-no-dev/ESPAsyncWebServer/issues/1334, even with latest espressif, websockets seem to permanently leak memory.
Calling sendAll with 0 clients -> constant memory usage, nothing happens.
Calling sendAll with say 5 clients connected will leak memory that will never recover after connections are closed.
Receiving WS messages does not seem to cause the same type of leak.

I found one leak in PsychicHttp already, but it's on a rare code path that does not trigger in this case.

```cpp
esp_err_t PsychicHttpWebSocketConnection::queueMessage(httpd_ws_frame_t *ws_pkt)
{
  // create a copy of this packet as its getting queued to the http server
  struct async_resp_arg *resp_arg = (async_resp_arg *)malloc(sizeof(struct async_resp_arg));

  // did we get the memory?
  if (resp_arg == NULL)
  {
    ESP_LOGE(PH_TAG, "queueMessage malloc failed to allocate");
    return ESP_ERR_NO_MEM;
  }

  resp_arg->hd = this->_server;
  resp_arg->fd = this->_fd;
  resp_arg->data = (char *)malloc(ws_pkt->len + 1);

  // did we get the memory?
  if (resp_arg->data == NULL)
  {
    free(resp_arg); // added
    ESP_LOGE(PH_TAG, "httpd_queue_work malloc failed to allocate %d", ws_pkt->len + 1);
    return ESP_ERR_NO_MEM;
  }

  // copy it over and send it off
  memcpy(resp_arg->data, ws_pkt->payload, ws_pkt->len + 1);
  esp_err_t err = httpd_queue_work(resp_arg->hd, PsychicHttpWebSocketConnection::queueMessageCallback, resp_arg);
  if (err != ESP_OK)
  {
    free(resp_arg->data); // added
    free(resp_arg); // added
    ESP_LOGE(PH_TAG, "httpd_queue_work failed with %s", esp_err_to_name(err));
  }
  return err;
}
```

ninja-, 2023-11-24:
Comment out httpd_queue_work - leak stops.
Leave it be with queueMessageCallback, but from queueMessageCallback remove httpd_ws_send_frame_async - leak still happens O_O

ninja-, 2023-11-24:
I think the work queue is lagging behind, so data is not freed for a very long time - so resp_arg and it's data. Maybe it keeps sending data to closed connections...

I enabled CONFIG_LWIP_SO_LINGER=y and added
  this->config.lru_purge_enable = true;
  this->config.enable_so_linger = true;

but memory still never recovers...

ninja-, 2023-11-24:
httpd uses a UDP socket for sending things to the "work queue", which means it can get randomly lost under load(what a terrible default setting o.o), and that means data will leak forever, as free happens inside the work function
[image]

there is a setting to "fix" that called CONFIG_HTTPD_QUEUE_WORK_BLOCKING...
unfortunately it causes a deadlock so it's not a direct fix....

at least I found the cause...

ninja-, 2023-11-24:
that code CONFIG_HTTPD_QUEUE_WORK_BLOCKING looks broken, looks like it still leaks, doesn't return semaphores, and ultimately deadlocks. hmm

I got rid of httpd_queue_work and now it works fine, no leaks!!

hoeken, 2023-11-29:
I'm really sorry I didn't see this before, I didnt get the new issue notification for some reason.

That does indeed look like a pretty serious bug on the esp-idf side of things. If I'm understanding this correctly, it basically is only a memory leak when the UDP packet is not received.

My code heavily depends on this to send data back to the client.  I haven't encountered this yet, probably because the UDP packets are getting through. I will do some testing and see if we can just directly send the httpd_ws_send_frame_async function which the callback would call.

I've been running the latest code for a while now on my main project and its been up for 1 day 14 hours, handled 25m messages, and does appear to have a very slow memory leak... the response that uses this queueMessage is a tiny fraction of the messages.

hoeken, 2023-11-29:
I switched the code to call httpd_ws_send_frame_async directly and am running it on my end.  So far so good.  Thanks for pointing this out, I would have torn my hair out over it.

BTW, there were a couple other leaks that I fixed in this commit: https://github.com/hoeken/PsychicHttp/commit/7230f64b8ce255de6e796b7b6372a5f7b18da2c2

closed 2023-11-29 by hoeken

ninja-, 2023-11-30:
Thanks. I'll push my current state of PsychicHttp over the weekend for comparision, there are just a couple more fixes which might be useful.
I recently discovered https://github.com/cesanta/mongoose which sounds like a good future backend for Psychic, as it only has C API similiar to espressif.
It looks like most IoT products are using `mongoose` these days to get away from the bug hell of AsyncHttpServer and others :)

ninja-, 2023-11-30:
Note that mongoose license seems a little weird https://mongoose.ws/licensing/ so it might be a deal breaker for anything other than prototyping or home use...

hoeken, 2023-11-30:
Funny you should mention that.

This project is actually a fork of https://github.com/jeremypoulter/ArduinoMongoose

There's a bit more info on it on the current readme here, but I was using that library for a while.  The problem is that it uses a very old version of mongoose and its not straightforward to update.  I actually got fairly far in doing that (and the history is in this github actually) but my main problem is that with Mongoose 7 it uses an internal buffer for serving pages which limits your page size to the available heap.  I have a 120kb const char * index file i need to serve so it was a dealbreaker for me to have to copy that into a buffer just to send.  Their github support team was not very supportive either.

commit 17893551 (2023-11-29): fixes #1