hoeken/PsychicHttp #3: Wow! Incredible.

zekageri, 2023-12-05:
I'm yet to test this library but i had goosebumps when i seen it on the Async Webserver issues.
One question. Do you have plans to implement SSE?
I'm currently using an ESP Async Webserver fork  and heavily using SSE.

I'm excited to try this out!
Thank you in the name of all the open source ppl!

hoeken, 2023-12-05:
I'm glad that you're excited!  I wasn't sure what features were popular so
I implemented the minimum I needed for my own use. I'll look into SSE and
see how difficult it would be to add.

hoeken, 2023-12-05:
Oops, I just saw the rest of your list.

* event source: i'm looking into this and it seems possible, but it will definitely take a bit of work.  espasynctcp is tightly coupled with AsyncTCP so i need to change all that out + fix all the knock on effects
* I'm not sure what you mean by middlewares?
* AP filter would be cool, I can look into it if there is interest
* Yes, you can set a notfound handler
* Not exactly, this library is similar to ESPAsyncWebserver, but doesn't use the exact Handler class syntax that it does.  It uses a simpler .on(url, method, callback) style. Part of the reason is that the underlying ESP-IDF callbacks require a URL + method for each endpoint so you cant just loop over each handler and call canHandle for each request because of that.
* it does not support gzip automatically but that could probably be added.  I'm not sure exactly how useful that would be though.  The slowdown in processing would probably offset the speed you gain from sending smaller files.  it does support sending .gz files from SPIFFS/LittleFS automatically if it finds {request_url}.gz

Do you mind sending me some code how you use Event Source?  The example on the ESPAsyncWebserver site is pretty basic.

zekageri, 2023-12-05:
Yeah, that was it. It is working with https://psychic.local/

hoeken, 2023-12-05:
yes, wss is working.. just that demo code isn't checking the protocol of the page and adapting appropriately.

hoeken, 2023-12-05:
> Oh, it would be good if there were a callback to redirect to secure context if the user wants. Will test it thanks! Let me know if you need more examples or help on anything. I will fork the repo and see what i can do.

Yeah that would be great.  I think it *might* be possible to create a second server that redirects all requests to the https side.

> Would you consider splitting the library to more files? It is really hard to see through the code when everything is in one file. I would split it into it's corresponding h and cpp files.

Absolutely.  I agree its a pain to read through it all.  I'm also going to try and make a flowchart of how all the pieces fit together.

hoeken, 2023-12-05:
I am preparing the changes for the EventSource + handlers functionality and just pushed a big cleanup that splits out all the individual classes and files.

zekageri, 2023-12-05:
That is wonderful. More pll should hear about this lib. We are waiting so long for exactly this library. So many ppl tried to do what MeNoDev done but all of it failed. I have tested the reliability of the requests and the websocket. I couldn't crash it with minimal test cases. I will write a more real world experience soon. The websocket is slower than ESPAsyncWebserver as you said, it can be improved. But again, you are amazing.

hoeken, 2023-12-05:
Thanks man!  To be fair, most of the hard work is done in the ESP-IDF library and PsychicHttp just makes it easy to use. :)

hoeken, 2023-12-06:
Quick little update, I have a pretty viable plan for v1.1 to add in EventSource stuff, as well as the generic handler functionality that ESPAsyncServer has. Plan is on the homepage if you're curious.  It's a bit of a job since the code isn't actually based on ESPAsyncWebserver, but I'll get there.  Tomorrows forecast is rainy so looking good for getting work done.

zekageri, 2023-12-06:
Will check!

zekageri, 2023-12-06:
Well, it sounds like a massive rewrite but it will worth it. If you have separate handlers the user can organise their code much more. Like in my case.

zekageri, 2023-12-07:
Really cool. I will test it as soon as i can. I think I will start to implement this tomorrow side by side to my esp async webserver.

proddy, 2023-12-09:
Like @zekageri the world has been waiting years for this library, super excited to start the port on my code. EventSource is also important to me, so happy to contribute there.

hoeken, 2023-12-09:
@proddy thanks :)

I've got EventSource implemented in the main branch, seems to be working well.  I'm hoping to make a v1.1 release this week.   If you could test with the git version of the code and give feedback that would be great.

I'm going to start a "Porting From ESPAsyncWebserver" section on the docs.  I've already converted my code in bits and pieces, so a lot of it will be from memory.  It would be great if you could send me any issues you run into so I can add it to the list.

closed 2023-12-11 by zekageri