hoeken/PsychicHttp #5: Defines for maxUploadSize

Eszartek, 2023-12-05:
Would if be possible to add defines for maxUploadSize so that prior to including PsychicHttp.h I can just #define MAXUPLOADSIZE 1500000 in my code.

Something like this placed in PsychicHttp.h:
```
#ifndef MAXUPLOADSIZE
  #define MAXUPLOADSIZE 200 * 1024
#endif
...
...
unsigned long maxUploadSize = MAXUPLOADSIZE;
```

Since there is no multi-part POST yet, I'm using your example xhr post code to update firmware.bin and lifftlefs.bin, so I just adjusted all your maximums to 1.5mb and it works perfect. Moving forward I want to help test new releases of PsychicHttp and I'd like to being able to use defines for this and future tunables to make it easier.

hoeken, 2023-12-05:
Absolutely!  That should be very straightforward to add.

closed 2023-12-05 by hoeken

Eszartek, 2023-12-07:
Lots of changes, but its looking good. I have one segfault in the websockets stuff, but i havent gotten to that yet, I just wanted to get firmware updates via multipart working, and I did. The only thing that messed me up is that

` request->contentLength()`

is a larger value that the actual filesize. ie:

`Writing  1122672/1122879 bytes to: /firmware.bin`

The firmware.bin was in fact 1122672 bytes.

I haven't tracked down why that is, but I just ignored it and pressed on coding and I finally got the firmware update to complete successfully. I had to rearrage the order of  "if (!index)" and "if (last) " compared to your example due to the extra complexity of doing and update vs writing to littlefs.

Now on to fixing the websocket fault that's probably just due to the changes in v1.1 that need updating in my code.

Eszartek, 2023-12-08:
I was just using request->contentLength()  to check that I got all the bytes before I commit to shutting off wifi and doing a restart. I'll prob move that bit to multipartHandler->onRequest as you suggest, as it makes logical sense to wrap up the firmware update task there.

I tracked the segfault to the v1.1 changes regarding server.serveStatic  needing ->setFilter(ON_STA_FILTER) and all is working as before. It's always a challenge finding all the changes one error at a time, but it forces me to really understand them.

> Hopefully this will be the last big structural change.

lol, as if :)

One last note, I noticed that my firmware size increased 112k or so, then I realized that in the original PyschicHttp code I had commented out all the SSL bits, and that was the difference. 

It would be handy to have those few lines #ifdef USE_SSL for those that don't need to use it and need the flash savings.

hoeken, 2023-12-08:
Another good suggestion with the SSL size stuff.  I tried to do it with #defines but Arduino makes it very hard to pass in build flags, so I copied the esp-idf style and now if you want SSL you can include another file with the derived class:

```
#include <PsychicHttp.h>
#include <PsychicHttpsServer.h>
PsychicHttpsServer server;
```

Eszartek, 2023-12-11:
FYI,  if I see commits, I  update Pyschic once per day and I've had no issues. I will try out the send events soon.

commit c2485541 (2023-12-05): added #ifndefs around config parameters fixes #5