hoeken/PsychicHttp #10: [Question] About handler classes.

zekageri, 2023-12-08:
How would one implement a handler class to handle only requests with a specific endpoint?

**Begin**
```cpp
#include "Main/Main.h"
server.addHandler(new MainHandler());
```

**Main handler.h**
```cpp
#pragma once

#include <Arduino.h>
#include <PsychicHttp.h>

class MainHandler : public PsychicHandler{
    public:
        esp_err_t handleRequest(PsychicRequest *request) override;
    private:
};
```

**Main handler.cpp**
```cpp
#include "Main.h"

esp_err_t MainHandler::handleRequest(PsychicRequest *request){
    // ???
    /*request->on("/",HTTP_GET,[](PsychicRequest *request){
        PsychicFileResponse response(request,db.sysFS,INDEX_PATH,"index/html");
        response.send();
    });*/
}
```

hoeken, 2023-12-08:
There are a couple ways to do this.  Easiest way would be to instantiate your class and add it to the server.  Also the fastest way if its a single endpoint.

```
MainHandler myhandler();
server.on("/url", HTTP_GET, &myhandler);
```

If its just responding to a simple request, the server.on("/url", METHOD, requestCallback()) shorthand will internally create an endpoint, adds a WebHandler and sets the onRequest() callback to the one you passed in.

If its more complicated, you can add it to the list of global handlers.  That is going to be a bit slower since it loops through each one of those and calls canHandle() if no urls match.  You would want to override the canHandle() function with logic to determine what requests it can handler. and then the handleRequest() function to actually handle the response.

```
MainHandler myhandler();
server.addHandler(&myhandler);
```

zekageri, 2023-12-08:
So it would be more complex  like

```cpp
Settings::Settings(AsyncWebServer *server){
  server->on("/settings/setLocalTime", HTTP_GET, [](AsyncWebServerRequest *request){
   // handle request
  });

  server->on("/settings/setConfig", HTTP_POST, [this](AsyncWebServerRequest *request) {
   // handle request
  });
};
```

hoeken, 2023-12-08:
There's definitely multiple ways to do this, it all just comes down to your preferences and your needs.

That code would work perfectly fine, infact its my preferred way for simple requests.  You can also drop the HTTP_GET parameter as it defaults to that.

```
Settings::Settings(AsyncWebServer *server){
  server->on("/settings/setLocalTime", [](AsyncWebServerRequest *request){
   // handle request
  });

If you have a handler object, you can also pass that instead of a callback:
```

You could also make your own more generic handler:

```
PsychicWebHandler handler;
handler.onRequest([](PsychicRequest *request){
   // do common setup stuff here
  if (request->url() == "/settings/foo")
    doSomeStuff();
  else if (request->url() == "/settings/bar")
    doOtherStuff();
  else
    error();

  //clean up and return response->send();
});

server.on("/settings/foo", &handler);
server.on("/settings/bar", &handler);
```

Wildcards should also be supported:

```
server.on("/settings/*", &handler);
```

zekageri, 2023-12-11:
That is really cool.
But can i pass a class as a handler and deal with requests inside?
Something like this

```cpp
#include "Settings/Main.h"

SettingsHandler settingsHandler();

void HsHServer::begin() {
    server.on("/settings/*", &settingsHandler);
    server.listen(SERVER_PORT);
}
```

And handle the requests coming to `"/settings/` inside the handler

**Settings.h**

```cpp
#pragma once

#include <Arduino.h>
#include <PsychicHttp.h>

class SettingsHandler : public PsychicHandler{
    public:
        esp_err_t handleRequest(PsychicRequest *request) override;
    private:
};
```

**Settings.cpp**

```cpp
esp_err_t SettingsHandler::handleRequest(PsychicRequest *request){
    if (request->url() == "/foo"){

    } else if (request->url() == "/bar"){

    }
    return 0;
}
```

For this i get an error right now

`no instance of overloaded function "PsychicHttpServer::on" matches the argument list`

zekageri, 2023-12-11:
Okay it works with `PsychicWebHandler` instead of `PsychicHandler` as a public class

zekageri, 2023-12-11:
It is really elegant using it like this

```cpp
#include "Auth/Main.h"
AuthHandler authHandler;
void setup(){
  server.listen(SERVER_PORT);
  server.serveStatic("/", db.sysFS, "/")->setCacheControl(ONE_HOUR_CACHE_TIME);
  server.on("/auth/*", &authHandler);
}
```

```cpp
#pragma once

#include <Arduino.h>
#include <PsychicHttp.h>

class AuthHandler : public PsychicWebHandler{
    public:
        esp_err_t handleRequest(PsychicRequest *request) override;
    private:
};
```

```cpp
esp_err_t AuthHandler::handleRequest(PsychicRequest *request){
    Serial.printf("[AuthHandler] - %s\n",request->url().c_str());
    if( request->url() == "/auth/getCurrentUser" ){
    }else if( request->url() == "/auth/login" ){
    }
    return 0;
}
```

closed 2023-12-11 by zekageri