hoeken/PsychicHttp #11: [Question] About websockets

zekageri, 2023-12-08:
Previously i had a class for handling websocket messages.

```cpp
ws.onEvent([](AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len){
    hshSocket.handleEvent(server, client, type, arg, data, len);
});

void HsHSocket::handleEvent(AsyncWebSocket *server, AsyncWebSocketClient *client, AwsEventType type, void *arg, uint8_t *data, size_t len) {
    if (type == WS_EVT_DATA) {
        AwsFrameInfo *info = (AwsFrameInfo *)arg;
        for(int i = 0; i < len; i++){
            buffer[socketBuffIndex++] = data[i];
        }
        if (socketBuffIndex >= info->len) {
            std::string message(buffer, info->len);
            messages.push_back( std::move(message) );
            socketBuffIndex = 0;
        }
    }else if (type == WS_EVT_CONNECT) {
        if( debug ){
            logger.write("[HsH_Socket] - (%d) Client connected.\n", client->id());
        }
    } else if (type == WS_EVT_DISCONNECT) {
        if( debug ){
            logger.write("[HsH_Socket] - (%d) Client disconnected.\n", client->id());
        }
    } else if (type == WS_EVT_ERROR) {
        if( debug ){
            logger.write("[HsH_Socket] - (%d) Error: %s\n", client->id(),(char *)data);
        }
    } else if (type == WS_EVT_PONG) {
        if( debug ){
            logger.write("[HsH_Socket] - (%d) Pong received.\n", client->id());
        }
    }
}
```

I had a global buffer which was allocated in the external RAM so i can accept any size of request at once. For this i had to buffer the data which was separated by AsyncWebsocket like this

```cpp
if (type == WS_EVT_DATA) {
    AwsFrameInfo *info = (AwsFrameInfo *)arg;
    for(int i = 0; i < len; i++){
        buffer[socketBuffIndex++] = data[i];
    }
    if (socketBuffIndex >= info->len) {
        std::string message(buffer, info->len);
        messages.push_back( std::move(message) );
        socketBuffIndex = 0;
    }
}
```

I want to keep this class with PsychicHttp.
Does this lib passes the data chunked or it buffers the entire data and give it at once?

zekageri, 2023-12-08:
If it is almost the same, in theory i could do something like this

```cpp
void HsHSocket::handleEvent(PsychicWebSocketRequest *request, httpd_ws_frame *frame) {
    for(int i = 0; i < frame->len; i++){
        buffer[socketBuffIndex++] = frame->payload[i];
    }
    if (socketBuffIndex >= frame->len) {
        std::string message(buffer, frame->len);
        messages.push_back( std::move(message) );
        socketBuffIndex = 0;
    }
}
```

hoeken, 2023-12-10:
@zekageri if you are dealing with websocket requests outside of the onFrame callback, you need to store the client->socket() and use that to get your websocketclient pointer.  Otherwise you might have issues if a client disconnects between sending you a message and your response.

Docs are up here: https://github.com/hoeken/PsychicHttp#websockets

To check the socket:

```cpp
//make sure our client is still connected.
PsychicWebSocketClient *client = websocketHandler.getClient(socket);
if (client != NULL)
  client->send("Your Message")
```

Same applies to eventsource clients.  Always store that socket # and not the client pointer

zekageri, 2023-12-11:
Thank you very much. I will try your suggestions. Right now the websocket client can not connect for some reason. I investigate it first. And i need to adjust my session logic too. I want to serve requests only to the logged in users but i need some exception path like login and auth.

zekageri, 2023-12-11:
i don't want to deal with socket clients. I just receive a json message and put it into a lifo std::deque buffer.
I process the messages one by one in the main loop.

```cpp
void HsHSocket::loop(){
    if( messages.empty() ){return;}
    std::string message = messages.front();
    emit(message.c_str());
    messages.pop_front();
}

void HsHSocket::handleEvent(PsychicWebSocketRequest *request, httpd_ws_frame *frame) {
    if (frame->len <= sizeof(buffer)){
        memcpy(buffer, frame->payload, frame->len);
        std::string message(buffer, frame->len);
        messages.push_back( std::move(message) );
    }
}
```

zekageri, 2023-12-11:
At my server init method i do this

```cpp
hshSocket.init();
server.on(WS_SOURCE, &ws);
```

```cpp
#define WS_SOURCE "/hshSocket"

void HsHSocket::init(){
    if (!hshSystem.alloc(buffer, MAX_FRAME_SIZE) && debug) {
        logger.write("[HsH_Socket] - Failed to allocate memory!\n");
    }
    ws.onOpen([](PsychicWebSocketClient *client) {
        Serial.printf("[socket] connection #%u connected from %s\n", client->socket(), client->localIP().toString());
    });
    ws.onClose([](PsychicWebSocketClient *client) {
        Serial.printf("[socket] connection #%u closed from %s\n", client->socket(), client->localIP().toString());
    });
    ws.onFrame([this](PsychicWebSocketRequest *request, httpd_ws_frame *frame) {
        handleEvent(request, frame);
        return true;
    });
}
```

Client can not connect for some reason.
Events also can not connect.
Strange...

zekageri, 2023-12-11:
Okay the problem was that i added the handlers before server.listen and not after. Now my events and ws can connect.

closed 2023-12-11 by zekageri