hoeken/PsychicHttp #9: [Question] Reply with file to a request?

zekageri, 2023-12-08:
How do i reply to a request with a file from the FS?

Should i create a separate response object for this?

Or
```cpp
server.on("/",HTTP_GET,[this](PsychicRequest *request){
    request->reply(200,LittleFS,INDEX_PATH,"text/html");
});
```

zekageri, 2023-12-08:
I see. This is the way. Strange but okay.

```cpp
server.on("/",HTTP_GET,[this](PsychicRequest *request){
    PsychicFileResponse response(request,db.sysFS,INDEX_PATH,"index/html");
    response.send();
});
```

zekageri, 2023-12-08:
Well, it does not compile :D

zekageri, 2023-12-08:
There must be a way to serve files

```cpp
server.on("/",HTTP_GET,[](PsychicRequest *request){
    PsychicFileResponse response(request,db.sysFS,INDEX_PATH,"index/html");
    response.send();
});
```

This does not compile. It says that the lambda is not good for it

```
src/Server/Server.cpp:22:10: error: no matching function for call to 'PsychicHttpServer::on(const char [2], http_method, HsHServer::begin()::<lambda(PsychicRequest*)>)'
         });
          ^
In file included from lib/PsychicHttp-master/src/PsychicHttp.h:7,
                 from src/Server/Server.h:4,
                 from src/Server/Server.cpp:1:
lib/PsychicHttp-master/src/PsychicHttpServer.h:55:22: note: candidate: 'PsychicEndpoint* PsychicHttpServer::on(const char*)'
     PsychicEndpoint* on(const char* uri);
```

but i can see that

```cpp
PsychicEndpoint *PsychicHttpServer::on(const char* uri, http_method method, PsychicHttpRequestCallback fn)
{
  //these basic requests need a basic web handler
  PsychicWebHandler *handler = new PsychicWebHandler();
  handler->onRequest(fn);

  return on(uri, method, handler);
}
```

Where `PsychicHttpRequestCallback fn` is a `typedef std::function<esp_err_t(PsychicRequest *request)> PsychicHttpRequestCallback;` so it should work with a simple lambda

Eszartek, 2023-12-08:
also from the example:
```
//serve static files from LittleFS/www on / only to clients on same wifi network
//this is where our /index.html file lives
server.serveStatic("/", LittleFS, "/www/")->setFilter(ON_STA_FILTER);
```
and naturally you can change "/www/" to "/".

zekageri, 2023-12-08:
Yeah, that was my mistake. I already sorted this pointer thing. Thanks.

hoeken, 2023-12-08:
@zekageri serveStatic is the recommended way

zekageri, 2023-12-08:
What if i want to construct a file after a request and send it as a response?

zekageri, 2023-12-08:
I have a bunch of deferred requests where i pass an `AsyncWebServerRequest` to a class which saves it, creates a file and sends it back.

```cpp
void HwConfig::startReScann(AsyncWebServerRequest *request) {
    scannRequest = request;
    reScann = true;
}
```

// In the loop when the process is done

```cpp
AsyncWebServerResponse *response =  scannRequest->beginResponse(
    db.userFS,
    HW_CONFIG_PATH,
    "application/json"
);
scannRequest->send(response);
```

closed 2023-12-11 by zekageri