hoeken/PsychicHttp #13: [Question] About handlers

zekageri, 2023-12-11:
i have added my handler to the on method of the server.

server.on("/auth/*", &authHandler);

I have to support both HTTP_POST and HTTP_GET on the same path.

In the handler method

```cpp
if( request->url() == "/auth/login" ){ // This should be a GET
}else if( request->url() == "/auth/login" ){ // This should be a POST
}
```

How to do that?
Should i add my handler twice?

```cpp
server.on("/auth/*", HTTP_GET, &authHandler);
server.on("/auth/*", HTTP_POST, &authHandler);
```

And handle it inside the callback like this?

```cpp
if( request->url() == "/auth/login" && request->method() == "HTTP_GET ){ // This should be a GET
}else if( request->url() == "/auth/login" && request->method() == "HTTP_POST  ){ // This should be a POST
}
```

zekageri, 2023-12-12:
So far it looks like this for me

```cpp
server.on("/auth/*", HTTP_GET, &authHandler);
server.on("/auth/*", HTTP_POST, &authHandler);

server.on("/db/*", HTTP_GET, &dbHandler);
server.on("/db/*", HTTP_POST, &dbHandler);

server.on("/zone/*", HTTP_GET, &zoneHandler);
server.on("/zone/*", HTTP_POST, &zoneHandler);

server.on("/settings/*", HTTP_GET, &settingsHandler);
server.on("/settings/*", HTTP_POST, &settingsHandler);

server.on("/hardwares/*", HTTP_GET, &hardwareHandler);
server.on("/hardwares/*", HTTP_POST, &hardwareHandler);
```

And it is getting ugly.

Inside every handler i check for the method and the path

```cpp
if (request->path() == "/hardwares/ping") {
    return ping(request);
} else if (request->path() == "/hardwares/updateIndexes" && request->method() == HTTP_POST) {
    return updateIndexes(request);
}
```

Because there can be multiple methods for a given path.

hoeken, 2023-12-12:
Unfortunately the underlying ESP-IDF stuff does not support the HTTP_ANY method like ESPAsync does.

Implementing this would require skipping their endpoint callback entirely and abusing the notFound callback to implement it ourselves.  We are already doing that for the global handlers, so it's not a huge change, but I'd like to do a bit more research and see if there are any other downsides before doing that.

For now I think you're just going to have to live with a bit of ugliness in the setup area.  You're going to need to apply the logic of checking url and method somewhere.  ESP-IDF wants you to do it when you attach the callback, but you want to do it in the handler.

If you really want to push all that to the handler, you can add it as a global handler and then use the canHandle function to check the url and the method.

zekageri, 2023-12-13:
Yeah because like in express.js you can attach handlers for endpoints and handle any endpoint within that class.
This is an elegant solution.

For example:

```cpp
server.on("/user/*", HTTP_ANY, &userHandler);

class UserHandler: public PsychicWebHandler{
private:
  esp_err_t UserHandler::handleRequest(PsychicRequest *request) {
    // Handle any request coming to /user
    /*
    /add
    /remove
    /login
    /logout
    /etc...
    */
  }

};
inline UserHandler userHandler;
```
It would be even better if we could just skip the first path in the handler because we already know that for example this is a `/user` path so the path would only contain the relevant strings. This is purely for astethics and readability and line width optimisation.

But i guess we could do something like this

```
server.on("/user/*", HTTP_ANY, [](){
    userHandler(&server);
});

class UserHandler: public PsychicWebHandler{
 public:
    UserHandler::UserHandler( PsychicHttpServer* server ){
        server->on("/add",[]( PsychicRequest *request ){
            // Process request...
        });
    }
}
```

proddy, 2023-12-13:
I don't use HTTP_ANY, I handle all the routing to specific endpoints on the client (using react-router). But I can see a probable use case. I switched out express for `itty-router` a while ago for performance and they have a similar cascading path construct (https://itty.dev/itty-router/route-patterns#query).

zekageri, 2023-12-13:
With react router you still need to handle ANY because there are other endpoints than page routes.
Yes you have for example `/login` with `GET` so you can get a login page and you also have a `/login` `POST` so a user can login with creds.

proddy, 2023-12-13:
that's very true, valid point.

zekageri, 2023-12-13:
You can also set a `/user` with `PUT` or `DELETE` or `POST` so you can modify, remove or add a new user.
Never mind it still does not change the fact that the IDF does not support ANY. :D

hoeken, 2023-12-13:
I opened a feature request on the ESP-IDF side of things, lets see if there is any interest.  It shouldn't be too hard to implement.

https://github.com/espressif/esp-idf/issues/12794

zekageri, 2023-12-13:
Fingers crossed. I hope those guys are at least as half as active as you.

hoeken, 2023-12-22:
Dont hold your breath, lol

proddy, 2023-12-29:
The comment from espressif is correct; there is no HTTP_ANY in the RFC (https://www.rfc-editor.org/rfc/rfc2616#section-5.1.1) 

Having methods matching would be nice, like 'HTTP_GET | HTTP_POST'.

Each of my server.on() endpoints are taking up 1kb of free heap mem and I could reduce that to half but not having a single handler for get/read settings and post/saving settings.

zekageri, 2024-01-04:
I ended up adding separate on handlers to my endpoints and skipped class specific handling.

closed 2024-01-04 by zekageri

johnnytolengo, 2024-01-26:
@zekageri how did you fix/workaround that?

I have a single endpoint:  /getsomething
GET:  get some data.
PUT:  save some data.

I tried to use two entries (esp32) but crashes:
server.on("/getsometing", HTTP_GET, ...
server.on("/getsometing", HTTP_PUT, ...

zekageri, 2024-01-26:
Well, put and get should work out of the box. Try it with post instead of put or if it didnt work just add different url. But are you sure the crash is not in your code?

johnnytolengo, 2024-01-26:
you are right it worked but I found a possible bug or bad situation:

```
server.on("/dosomething", HTTP_PUT, [this](PsychicRequest *request, JsonVariant &jsonDoc){
});
```
if the body is empty it automatically reponse "400 Bad Request"  it doesn't give you a chance to run that endpoint.