hoeken/PsychicHttp #14: Request body is empty on POST?

zekageri, 2023-12-11:
```cpp
Serial.printf("[AuthHandler] - %s Method: %s\n",request->url().c_str(),request->method() == HTTP_GET ? "GET" : "POST");
if( request->url() == "/auth/login" && request->method() == HTTP_POST ){
  Serial.printf("Request body content: %s\n",request->body());
  DynamicJsonDocument doc(USERS_JSON_SIZE);
  DeserializationError error = deserializeJson(doc, request->body() );
  if( error ){
      Serial.printf("[AuthHandler] - Error: %s\n",error.c_str());
      return request->reply(500);
  }
}
```

**Log:**

```cpp
[AuthHandler] - /auth/login Method: POST
Request body content: 
[AuthHandler] - Error: EmptyInput  
```

It should be a json string

hoeken, 2023-12-11:
Yeah if you're extending webhandler and overriding the handleRequest() function you will need to re-implement some of that logic.  Or just use the onRequest callback.

zekageri, 2023-12-12:
I see. Thank you.

closed 2023-12-12 by zekageri