hoeken/PsychicHttp #15: Session key value

zekageri, 2023-12-11:
Can i use cookies to store session?

hoeken, 2023-12-11:
Not sure exactly what you mean, but you can access cookies from the request object using the functions:

```cpp
    bool hasCookie(const char * key);
    const String getCookie(const char * key);
```

If you want to set a cookie from the server side you will need to set the appropriate headers in the response body.  I could probably add a convenience function there if there is interest.

I added some session storage stuff since the esp-idf library supports it.  Basically just a way to store strings associated with an active connection.  Its very basic and honestly not that well tested.  The functions are in the request class as well:

```cpp
    bool hasSessionKey(const String& key);
    const String getSessionKey(const String& key);
    void setSessionKey(const String& key, const String& value);
```

zekageri, 2023-12-11:
Yeah, I have seen the session methods and I have tried it too. Currently big servers has a way to config the storage of the session string. With express you can choose if you store it in a db or a cookie. This way the server can keep track of the session even after restart. I used cookies with async server before. I tought that if it is implemented it would have been cool. For the login session i set up a 7 day cookie in the browser and if the server restarts the user is still logged in because the cookie is saved.

zekageri, 2023-12-11:
I will write an example tomorrow

zekageri, 2023-12-12:
So this was my setCookie method with AsyncServer

```cpp
void HsH_Server::setCookie(AsyncWebServerResponse* response, const char* name, const char* value, int secondsFromNow) {
    time_t now = time(nullptr);
    time_t expirationTimestamp = now + secondsFromNow;

    // Convert the expiration timestamp to a formatted string for the "expires" attribute
    struct tm* tmInfo = gmtime(&expirationTimestamp);
    char expires[128];
    strftime(expires, sizeof(expires), "%a, %d %b %Y %H:%M:%S GMT", tmInfo);

    // Set the cookie with the "expires" attribute
    char cookie[512];  // Adjust the buffer size as needed
    snprintf(cookie, sizeof(cookie), "%s=%s; expires=%s; path=/; HttpOnly;", name, value, expires);
    response->addHeader("Set-Cookie", cookie);
 
    // For debug
    Serial.printf("[Server] - Set cookie: %s\n", cookie);
}
```

It requires the correct system time set from ntp or whatever. Accepts the response pointer, a cookie name, a value and an integer value which represents when the cookie should expire in seconds from the time it was called.
Can it be integrated? I mean, it requires the system time and an internal cookie buffer so you can just leave it to the user. I don't know how much this is the server's responsibility

hoeken, 2023-12-12:
Take a look at the response object.  There is a setCookie() function there that takes the name, data, and a max age.  No need for a date or ntp on the esp (unless you want to)

On a side note, this is one of the things that bugs me about Async is that it confuses an HTTP request with an HTTP response and makes things more difficult than it needs to be.  The request is what the client sends to the server, and the response is what gets sent back.

zekageri, 2023-12-12:
Yeah but you also did it so we can answer to the request using the request. Express js has a request and a response object passed to the callback automatically

zekageri, 2023-12-13:
Max-Age is not a good solution for me.

**From the MDN docs:**

> Maximum number of seconds the results can be cached, as an unsigned non-negative integer. Firefox [caps this at 24 hours](https://searchfox.org/mozilla-central/source/netwerk/protocol/http/nsCORSListenerProxy.cpp#1207) (86400 seconds). Chromium (prior to v76) [caps at 10 minutes](https://source.chromium.org/chromium/chromium/src/+/main:services/network/public/cpp/cors/preflight_result.cc;drc=52002151773d8cd9ffc5f557cd7cc880fddcae3e;l=36) (600 seconds). Chromium (starting in v76) [caps at 2 hours](https://source.chromium.org/chromium/chromium/src/+/main:services/network/public/cpp/cors/preflight_result.cc;drc=49e7c0b4886cac1f3d09dc046bd528c9c811a0fa;l=31) (7200 seconds). The default value is 5 seconds.

Reference link: https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Access-Control-Max-Age

For example i want to set a session cookie on the client side to 7 days. It is a user login cookie. Max 24 hour is a really low number for something like this. I have to set an expires attribute so i can precisely tell the browser the expiration date of the cookie.

zekageri, 2023-12-13:
With my solution you can set it precisely to whatever date you want from now.
I rewrote so it uses the PsychicResponse

```cpp
void HsHServer::setCookie(PsychicResponse* response,const char* name, const char* value, int secondsFromNow) {
    time_t now = time(nullptr);
    time_t expirationTimestamp = now + secondsFromNow;

    // Convert the expiration timestamp to a formatted string for the "expires" attribute
    struct tm* tmInfo = gmtime(&expirationTimestamp);
    char expires[128];
    strftime(expires, sizeof(expires), "%a, %d %b %Y %H:%M:%S GMT", tmInfo);

    // Set the cookie with the "expires" attribute
    char cookie[512];  // Adjust the buffer size as needed
    snprintf(cookie, sizeof(cookie), "%s=%s; expires=%s; path=/; HttpOnly;", name, value, expires);
    response->addHeader("Set-Cookie", cookie);
}
```

hoeken, 2023-12-13:
That page you linked is about a completely different header.  Set-Cookie has a sub-field called Max-Age which doesn't have any of those limitations.

Regardless, I added your code in.  It checks to see if there is a system date set and will use an expiration date or max-age as appropriate.  I also added a 4th parameter of 'extras' since the Set-Cookie spec has a ton of options and I didn't want to bloat this function.  You can call it like this:

```cpp
response->setCookie(name, value, secondsFromNow, "path=/; HttpOnly");
```

The extras get concatenated on at the end.

zekageri, 2023-12-13:
Oh I see. This is a really nice addition. I tried to use setCookie but it didnt work for me for some reason. I will try this version!

zekageri, 2023-12-14:
It just does not work for me.

```cpp
PsychicResponse response(request);
response.setCookie(SESSION_KEY,sessionID.c_str(),USER_LOGIN_EXPIRATION_SECONDS); // Built in setCookie
//hshServer.setCookie(&response,SESSION_KEY,sessionID.c_str(),USER_LOGIN_EXPIRATION_SECONDS); // my setCookie
return response.send();
```

My setCookie is working but not the built in. It just does not set the header at all.
If i use my version it is working.

zekageri, 2023-12-14:
NVM. It was because of the path attribute.
It is really clever we can add additional attributes.

closed 2023-12-14 by zekageri