hoeken/PsychicHttp #18: Got an reboot with assert fail

zekageri, 2023-12-11:
```
assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)

Backtrace: 0x40083c7d:0x3ffde7b0 0x4008fc39:0x3ffde7d0 0x40095279:0x3ffde7f0 0x40092f27:0x3ffde920 0x40090d3a:0x3ffde960 0x4017897f:0x3ffde9a0 0x4017a1b8:0x3ffde9d0 0x4017a2d4:0x3ffdea00 0x4017a34a:0x3ffdea50 0x40166587:0x3ffdea70 0x401680b9:0x3ffdeab0 0x4016818e:0x3ffdeb10 0x40182791:0x3ffdeb30 0x4018227d:0x3ffdeb50 0x40181901:0x3ffdeb70 0x40182112:0x3ffdec10 0x40180dcc:0x3ffdec30 0x4021d5fe:0x3ffdec50 0x40180ef1:0x3ffdec70

ELF file SHA256: 211f9037e9582717

Rebooting...
```

hoeken, 2023-12-11:
My first guess is that the stack size on the server is too small.  The server runs in its own thread with its own stack, set to 4096 by default.  You can adjust that by setting server.config.stack_size to something bigger.  try something like 32768 and see if that fixes the problem

zekageri, 2023-12-11:
Oh yeah. That will be it. I will try thanks

zekageri, 2023-12-12:
I increased the stack size to the suggested size but i got an error again

```
CORRUPT HEAP: Bad head at 0x3ffbc530. Expected 0xabba1234 got 0x3ffb9a34 

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL) 

Backtrace: 0x40083c7d:0x3ffec100 0x4008fc39:0x3ffec120 0x40095279:0x3ffec140 0x40094e71:0x3ffec270 0x4008421d:0x3ffec290 0x400952a9:0x3ffec2b0 0x4013b852:0x3ffec2d0 0x40122441:0x3ffec2f0 0x401225e5:0x3ffec3f0 0x4013aa9d:0x3ffec440 0x40182ca1:0x3ffec4c0 0x40181bb0:0x3ffec4f0 0x401822c6:0x3ffec590 0x40180f80:0x3ffec5b0 0x4021d77a:0x3ffec5d0 0x401810a5:0x3ffec5f0   

  #0  0x40083c7d:0x3ffec100 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408

ELF file SHA256: 7d953e6b12d1c0e7

Rebooting...
```

zekageri, 2023-12-12:
I've got something in the backtrace now

```cpp
Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.

Core  0 register dump:
PC      : 0x40092e80  PS      : 0x00060433  A0      : 0x800908c4  A1     
 : 0x3ffd6c40
A2      : 0x70697ab3  A3      : 0xb33fffff  A4      : 0x0000abab  A5     
 : 0x00060423
A6      : 0x00060420  A7      : 0x0000cdcd  A8      : 0x0000cdcd  A9     
 : 0xffffffff
A10     : 0x00000003  A11     : 0x00060423  A12     : 0x00060420  A13    
 : 0x00000000
A14     : 0x30e97ab3  A15     : 0x003fffff  SAR     : 0x00000020  EXCCAUSE: 0x00000009
EXCVADDR: 0x70697ab3  LBEG    : 0x4008b064  LEND    : 0x4008b07a  LCOUNT 
 : 0xffffffff

Backtrace: 0x40092e7d:0x3ffd6c40 0x400908c1:0x3ffd6c80 0x40178ab1:0x3ffd6cc0 0x4017aa2c:0x3ffd6cf0 0x4016f48a:0x3ffd6d10 0x40174346:0x3ffd6d80 0x4017864a:0x3ffd6db0 0x40168ce1:0x3ffd6dd0

  #0  0x40092e7d:0x3ffd6c40 in compare_and_set_native at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/compare_set.h:25
      (inlined by) spinlock_acquire at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/spinlock.h:103    
      (inlined by) xPortEnterCriticalTimeout at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:301     

ELF file SHA256: ecbe8b7d917fef41

Rebooting...
ets Jul 29 2019 12:21:46
```

Still crashing when i serve files.

```cpp
esp_err_t DBHandler::handleDownload(PsychicRequest* request) {
    request->loadParams();

    if (!request->hasParam("path")) {
        return request->reply(400);
    }

    const String& partitionLabel = request->hasParam("partitionLabel") 
        ? request->getParam("partitionLabel")->value()
        : "user";
    
    fs::LittleFSFS partition = partitionLabel == "user" 
        ? db.userFS 
        : db.sysFS;

    const String& filePath = request->getParam("path")->value();

    if (!db.exists(filePath.c_str(), partition)) {
        return request->reply(404);
    }

    PsychicFileResponse response(request, partition, filePath, "application/octet-stream");

    if (partitionLabel != "user") {
        response.addHeader("Cache-Control", MAX_CACHE_CONTROL_TIME);
    }

    return response.send();
}
```

zekageri, 2023-12-12:
Okay it was my fault. There were some problem with Strings..

closed 2023-12-12 by zekageri