hoeken/PsychicHttp #21: How to get proper content length for json deserialization?

zekageri, 2023-12-12:
```cpp
esp_err_t ZoneHandler::testJsonPost(PsychicRequest *request){
    Serial.println( request->contentLength() );
    Serial.println( request->body().length() );

    DynamicJsonDocument doc( request->contentLength() );
    DeserializationError error = deserializeJson(doc,request->body());
    if(error){
        Serial.println(error.c_str());
        return request->reply(500);
    }
    return request->reply(200);
}
```

Log

```cpp
393
393
NoMemory
```

Both contentLength and body().length() returns the same size.
Should i know ahead of time the json body length?

zekageri, 2023-12-13:
i have enough heap. I just tought that it would be so easy. I'm currently doing this

```cpp
esp_err_t ZoneHandler::testJsonPost(PsychicRequest *request){
    DynamicJsonDocument doc( request->contentLength() * 2 ); // No error that way so far
    DeserializationError error = deserializeJson(doc,request->body());
    if(error){
        Serial.println(error.c_str());
        return request->reply(500);
    }
    return request->reply(200);
}
```

hoeken, 2023-12-13:
Yeah ArduinoJson has a few quirks like that, glad its working though.

closed 2023-12-14 by zekageri