hoeken/PsychicHttp #24: Something is wrong with setCacheControl

zekageri, 2023-12-14:
Something was very sus from the start when i started to use this lib.
I use setCacheControl to every request.

zekageri, 2023-12-14:
**Here are the response headers after one hour before cache delete**

[image]

**And after one hour after cache delete**

[image]

hoeken, 2023-12-14:
Just pushed a potential fix.  Looks like there was a missing call to set the response code to 304 when checking the etags

zekageri, 2023-12-14:
Yeah, it keeps logging these

PsychicStaticFileHander.cpp:144: _filename = /Manifest/favicon.ico
PsychicStaticFileHander.cpp:173: "No cache hit" = No cache hit
PsychicStaticFileHander.cpp:174: _last_modified =
PsychicStaticFileHander.cpp:175: _cache_control = max-age=3600
PsychicStaticFileHander.cpp:144: _filename = /Pages/HomePage/quickLights.css
PsychicStaticFileHander.cpp:173: "No cache hit" = No cache hit
PsychicStaticFileHander.cpp:174: _last_modified =
PsychicStaticFileHander.cpp:175: _cache_control = max-age=3600
PsychicStaticFileHander.cpp:144: _filename = /947.js
PsychicStaticFileHander.cpp:173: "No cache hit" = No cache hit
PsychicStaticFileHander.cpp:174: _last_modified =
PsychicStaticFileHander.cpp:175: _cache_control = max-age=3600

hoeken, 2023-12-14:
Ah sorry I forgot to mention I left those in there to help tracking down
the bug if that doesn't fix it.

Let me know if it's fixed and I'll remove them.

zekageri, 2023-12-15:
Testing

zekageri, 2023-12-15:
It seems to work. I have tested it on

**Android**
- Chrome
- Firefox

**Windows**
- Edge
- Chrome
- Brave

closed 2023-12-15 by zekageri