hoeken/PsychicHttp #25: A lot of times the website is not loading on mobile.

zekageri, 2023-12-14:
It just stops answering to the requests.
I have a lot of files and endpoints but it worked with Async webserver. ( At least with the forked one )

hoeken, 2023-12-14:
How many files are you serving for a single request?
Does it stop responding completely?

It might be worth looking into using something like gulp to compress all your html + css + js into a single file or at least into a single html + js + css file.  The ESP is capable, but it does have limitations.

I use gulp in my other project: https://github.com/hoeken/yarrboard-firmware/blob/main/gulpfile.mjs

Once I have the combined html file, I then write it out as a header file to be #included.  That way I can skip the LittleFS stuff entirely and have a single .bin for OTA.  I'm not sure what FS you're using or if you're pulling from SD.  It still would be worth it to compress down to a less files.

Its pretty easy to integrate with your build process too.  I just call the gulp command every time I build the firmware.

zekageri, 2023-12-14:
I'm using webpack and all my files are compressed and gzipped. I have a lot of css files because i did not integrated it to the webpack build process. It just builds my js files for now. With forked Async i had no trouble with this. Gulp sounds good hovewer

closed 2024-01-04 by zekageri