hoeken/PsychicHttp #26: Problem with PsychicUploadHandler

zekageri, 2023-12-14:
So i have setup an upload handler like this

```cpp
uploadHandler = new PsychicUploadHandler();
uploadHandler->onUpload([this](PsychicRequest *request, const String& filename, uint64_t index, uint8_t *data, size_t len, bool last) {
    request->loadParams();
    // Logic here...
    return ESP_OK;
});
uploadHandler->onRequest([this](PsychicRequest *request){
    tempFile.close();
    return request->reply(200);
});
```

But the server throws a 400 bad requests with the content of `Request body must be less than 16384 bytes!`

[image]
[image]

I tought that it supposed to handle the files chunked and not have this limitation.

zekageri, 2023-12-14:
I see that this limitation is in the `handleRequest` method of `PsychicWebHandler`

```cpp
esp_err_t PsychicWebHandler::handleRequest(PsychicRequest *request)
{
  /* Request body cannot be larger than a limit */
  if (request->contentLength() > request->server()->maxRequestBodySize)
  {
    ESP_LOGE(PH_TAG, "Request body too large : %d bytes", request->contentLength());

    /* Respond with 400 Bad Request */
    char error[60];
    sprintf(error, "Request body must be less than %u bytes!", request->server()->maxRequestBodySize);
    httpd_resp_send_err(request->request(), HTTPD_400_BAD_REQUEST, error);

    /* Return failure to close underlying connection else the incoming file content will keep the socket busy */
    return ESP_FAIL;
  }

  //get our body loaded up.
  esp_err_t err = request->loadBody();
  if (err != ESP_OK)
    return err;

  //load our params in.
  request->loadParams();

  //okay, pass on to our callback.
  if (this->_requestCallback != NULL)
    err = this->_requestCallback(request);

  return err;
}
```

So how can we upload more than `request->server()->maxRequestBodySize` ?

hoeken, 2023-12-14:
No that should work, the UploadHandler overrrides the handleRequest method and doesn't call the parent.  My guess is that you might have gotten the .on() call wrong and its defaulting to a standard WebHandler.

I did just push a commit that adds the loadParams() to the UploadHandler so you can drop that from your own onUpload()

hoeken, 2023-12-14:
```cpp
server.on("/upload/*", HTTP_POST, &uploadHandler);
```

zekageri, 2023-12-14:
Because it was already a pointer i did not have to use a & operator on the on method. And it does not even compile with it if i keep using the pointer

zekageri, 2023-12-15:
Something is odd for me. It is not working i can't upload more than 16384 bytes.

zekageri, 2023-12-15:
I did this now

```cpp
PsychicUploadHandler *uploadHandler = new PsychicUploadHandler();
uploadHandlerInit(uploadHandler);
```

```cpp
void HsHServer::uploadHandlerInit(PsychicUploadHandler *uploadHandler){
    uploadHandler->onUpload([this](PsychicRequest *request, const String& filename, uint64_t index, uint8_t *data, size_t len, bool last) {
        if (!request->hasParam("path")) {
            Serial.println("[UploadHandler] - No path provided");
            return request->reply(500);
        }
        return ESP_OK;
    });
    uploadHandler->onRequest([this](PsychicRequest *request){
        tempFile.close();
        return request->reply(200);
    });
}
```

Same problem.

I will check it with static class

```cpp
PsychicUploadHandler uploadHandler;

void init(){
    server.on("/upload/*", HTTP_POST, &uploadHandler);
}

uploadHandler.onUpload([this](PsychicRequest *request, const String& filename, uint64_t index, uint8_t *data, size_t len, bool last) {
    if (!request->hasParam("path")) {
        Serial.println("[UploadHandler] - No path provided");
        return request->reply(500);
    }
    return ESP_OK;
});
uploadHandler.onRequest([this](PsychicRequest *request){
    tempFile.close();
    return request->reply(200);
});
```

zekageri, 2023-12-15:
It isn't working. Keeps sending me a `400 bad request` response

zekageri, 2023-12-15:
[image]

hoeken, 2023-12-22:
Are you still having issues with this?  Lmk and I will dig into it deeper.

zekageri, 2023-12-22:
Yes for some reason it just doesnt work. I too will look into it today

zekageri, 2024-01-03:
The upload handler started to behave strange.
If i pass more than 16k body the server can't find the endpoint.
I have multiple requests on the frontend to the /upload endpoint.
One has a body below 16k and the upload works fine.
The other has 56k and it throws a 404 response.
The situation is the same.

```cpp
inline PsychicUploadHandler uploadHandler;

uploadHandler.onUpload([this](PsychicRequest *request, const String& filename, uint64_t index, uint8_t *data, size_t len, bool last){
    if (!request->hasParam("path")) {
        Serial.println("[UploadHandler] - No path provided");
        return request->reply(500);
    }
    boolean isUserPartition = true;
    if (request->hasParam("partitionLabel") && request->getParam("partitionLabel")->value() != "user") { isUserPartition = false; }
    tempPartition = isUserPartition ? db.userFS : db.sysFS;
    if (!index){
        tempFile = tempPartition.open(request->getParam("path")->value().c_str(), FILE_WRITE);
        Serial.printf("[UploadHandler] - %s\n", request->getParam("path")->value().c_str());
    }else{
        tempFile = tempPartition.open(request->getParam("path")->value().c_str(), FILE_APPEND);
    }
    if(!tempFile || !tempFile.write(data, len)) { return ESP_FAIL; }
    return ESP_OK;
});
uploadHandler.onRequest([this](PsychicRequest *request){
    tempFile.close();
    return request->reply(200);
});

server.on("/upload/*", HTTP_POST, &uploadHandler);
```

**Request with more than 16k**

[image]

**Request below 16k**

[image]

zekageri, 2024-01-03:
So i have removed the wildcard endpoint and doing just this

`server.on("/upload", HTTP_POST, &uploadHandler);`

The server still thinks that this is a regular handler and not an upload handler. It limits me to the `MAX_REQUEST_BODY_SIZE` again.

zekageri, 2024-01-03:
i didn't know i should set maxUploadSize O_o Will try

zekageri, 2024-01-03:
If i don't set it should default to `MAX_UPLOAD_SIZE` isnt it? which is `(2048*1024)`

zekageri, 2024-01-03:
Interesting, now the server did not find the handler, returns 404

```cpp
inline PsychicUploadHandler uploadHandler;

uploadHandler.onUpload([this](PsychicRequest *request, const String& filename, uint64_t index, uint8_t *data, size_t len, bool last){}
uploadHandler.onRequest([this](PsychicRequest *request){}
server.on("/upload/*", HTTP_POST, &uploadHandler);
```

I don't understand what am i doing wrong.

proddy, 2024-01-03:
things to check
- you have enough max_uri_handlers set
- does it find it with just /upload ?
- uri_match_fn is not changed and still the default httpd_uri_match_wildcard
- try /upload/? which also matches /upload
- doesn't matter but using pointers and not inline declarations is tidier

zekageri, 2024-01-03:
Max uri handlers set to 1000
I have tried with /upload , /upload/ and /upload/*
the inline is there because i have this declared in a header file so i can use it in other files. C++ 17 compiles just once if there is an inline

zekageri, 2024-01-03:
I have these congiured now.

```
#define MAX_HANDLER_COUNT 1000
#define STACK_SIZE 33768

server.config.max_uri_handlers = MAX_HANDLER_COUNT;
server.config.stack_size = STACK_SIZE;
server.config.max_open_sockets = 13;
server.maxUploadSize = MAX_UPLOAD_SIZE; // Comes from Psychic lib
server.maxRequestBodySize = MAX_REQUEST_BODY_SIZE; // Comes from Psychic lib
```

proddy, 2024-01-03:
and the server.listen() is after the WiFi is running? Anything before will be ignored. I ran into this problem before, because with AsyncWS you could initialize the server->on() in the class initialization

zekageri, 2024-01-03:
I use Ethernet but yes.
I found the problem.

I had a request like this: `http://hsh.local/upload/?path=/graphs.json&partition=user` which worked fine
I tought that it works because of the smaller size.
The hardware upload request looked like this: `http://hsh.local/upload?path=/hwConfigBackup.json&partition=use`

notice any difference which is not the path? Yeah, there is no "/" after the upload in the url

This works: `http://hsh.local/upload/?path=/hwConfigBackup.json&partition=use`

zekageri, 2024-01-03:
Oh and it does not work even if my handler looks like this `server.on("/upload", HTTP_POST, &uploadHandler);`

proddy, 2024-01-03:
which is why I suggested `/upload/?` as this matches all paths ;-)

johnnytolengo, 2024-01-30:
Hi @zekageri how you make it work? I am days struggling with [this issue](https://github.com/hoeken/PsychicHttp/issues/69)  the onUpload() callback is called only once on the first file POST, then it's not called anymore :(

zekageri, 2024-01-30:
How do you call the endpoint from client side?