hoeken/PsychicHttp #29: Serve static data (not via FS)

proddy, 2023-12-18:
Thanks again for this great library. I've started my port from AsyncWebServer.

Is it possible to extend the library to support static data? Reason is that I'm generating all my HTML and placing into Flash mem for speed. I'm following the example on https://github.com/me-no-dev/ESPAsyncWebServer?tab=readme-ov-file#send-binary-content-from-progmem

Also another nice addition to your new Json response handling is to add `PsychicEndpoint * PsychicHttpServer::on(const char * uri, http_method method, PsychicJsonRequestCallback fn)` so it can be called via lambda on the `server->on()`

proddy, 2023-12-18:
thinking about this again, I maybe able to achieve this via

```C
        PsychicWebHandler * handler = new PsychicWebHandler();
        handler->onRequest([contentType, content](PsychicRequest * request) {
            return request->reply(200, contentType.c_str(), content);
        });
        server->on(uri.c_str(), HTTP_GET, handler);
```

hoeken, 2023-12-18:
Glad you're enjoying it.  You can serve that static data by just passing in your const char *:

In my other project I'm serving a 100kb index.html with the code below.  I'm using gulp to generate the header file that contains the compressed and combined html + css + js and also generates a sha1 hash that I use for Etag for caching.  Gulp file here: https://github.com/hoeken/yarrboard-firmware/blob/main/gulpfile.mjs

```cpp
  server.on("/", HTTP_GET, [](PsychicRequest *request) {

    //Check if the client already has the same version and respond with a 304 (Not modified)
    if (request->header("If-Modified-Since").indexOf(last_modified) > 0)
      return request->reply(304);
    //What about our ETag?
    else if (request->header("If-None-Match").equals(index_html_gz_sha))
      return request->reply(304);
    else
    {
        PsychicResponse response(request);
        response.setCode(200);
        response.setContentType("text/html");

        // Tell the browswer the contemnt is Gzipped
        response.addHeader("Content-Encoding", "gzip");

        // And set the last-modified datetime so we can check if we need to send it again next time or not
        response.addHeader("Last-Modified", last_modified);
        response.addHeader("ETag", index_html_gz_sha);

        //add our actual content
        response.setContent(index_html_gz, index_html_gz_len);

        return response.send();
    }
  });
```

proddy, 2023-12-18:
thats perfect. Thanks. I'll keep going with my port. I just added MsgPack as I need that too. If it all works I'll share my notes and do a couple of PRs.

hoeken, 2023-12-20:
Looking forward to it.  Happy to accept pull requests

bbasil2012, 2023-12-21:
I used this method:
server.on("/about",HTTP_GET,[](PsychicHttpServerRequest *request){
    PsychicHttpServerResponse response(request);
    response.addHeader("Content-Encoding", "gzip");
    response.setContentType("text/html");
    response.setCode(200);
    response.setContent(banner,sizeof(banner));
    return response.send();
  });

Where "banner" is gziped html file placed to static array:
const uint8_t banner[] PROGMEM = {0x1f,0x8b,...}

proddy, 2023-12-21:
yes, same here. works fine.

closed 2023-12-21 by proddy