hoeken/PsychicHttp #30: OTA Update

bbasil2012, 2023-12-21:
Whats about OTA ?

hoeken, 2023-12-21:
Its definitely possible.  There are many ways to do it.  Anything in particular?

bbasil2012, 2023-12-21:
Yes. My firmware file size exceeds 200K
I'm getting a message can't upload a file larger than 200K

hoeken, 2023-12-21:
Ah, yeah that's a configurable setting:

```cpp
server.maxUploadSize = bytes
```

bbasil2012, 2023-12-21:
Ok, I will try )
Very great lib!!!

bbasil2012, 2023-12-23:
It's worked !!! (c)
Here is my example code:

const uint32_t maxSketchSpace = 0x140000;
server.maxUploadSize=maxSketchSpace;
server.on("/update", HTTP_POST)->onUpload(updateHandler);

esp_err_t updateHandler(PsychicHttpServerRequest *request, const String& filename, uint64_t index, uint8_t *data, size_t len)
{
  if(!index){
    Serial.println("Begin OTA update");
    if(!Update.begin(maxSketchSpace)){
        Update.printError(Serial);
    }
  }
  if(!Update.hasError()){
      Serial.print(".");
      if(Update.write(data, len) != len){
        Update.printError(Serial);
      }
  }
  if (len<FILE_CHUNK_SIZE){
     if(Update.end(true)){
        Serial.printf("\nUpdate Success in: %u Bytes\n", index+len);
        ESP.restart();
      } else {
        Update.printError(Serial);
      }
  }
  return ESP_OK;
}

And example with curl:
curl -X POST -k -T firmware.bin https://<ip address or fqdn>/update

bbasil2012, 2023-12-23:
Done

Mr-HaleYa, 2024-01-04:
Great job!

Close the issue if you have solved your question

closed 2024-01-04 by bbasil2012