hoeken/PsychicHttp #31: [Question]  about corrupted certificates

bbasil2012, 2023-12-21:
If for some reason the certificates are corrupted on FS, is there any way to switch back to http?

13:50:16.693 -> E (18811) esp-tls-mbedtls: mbedtls_x509_crt_parse returned -0x2180
13:50:16.693 -> E (18812) esp-tls-mbedtls: Failed to set server pki context
13:50:16.693 -> E (18812) esp-tls-mbedtls: Failed to set server configurations, returned [0x8015] (ESP_ERR_MBEDTLS_X509_CRT_PARSE_FAILED)
13:50:16.693 -> E (18823) esp-tls-mbedtls: create_ssl_handle failed, returned [0x8015] (ESP_ERR_MBEDTLS_X509_CRT_PARSE_FAILED)

hoeken, 2023-12-21:
The call to server.listen() will return an esp_err_t which bubbles up from the calls to ESP-IDF http(s) server.  The ESP-IDF docs are usually pretty good on possible return values, but it just says 'success' on the docs for this one.

I haven't tested, but I'm 99% sure that you can just do something like this:

```cpp
if (server.listen(443, cert, key) != ESP_OK)
  server.listen(80);
```

bbasil2012, 2023-12-21:
Ок. I will try it!

hoeken, 2024-08-10:
I'm closing all these old issues.  Please re-open it if needed.

closed 2024-08-10 by hoeken