hoeken/PsychicHttp #32: esp_https_server.h: No such file or directory

proddy, 2023-12-21:
Hi all, has anyone else encountered recent compile issues? Not sure if it was my pio setup (I have removed all folders and re-installed) but recently I'm getting the below compile error why I never had before. It worked yesterday!

```
In file included from .pio/libdeps/default/PsychicHttp/src/PsychicHttp.cpp:1:
.pio/libdeps/default/PsychicHttp/src/PsychicHttp.h:19:10: fatal error: esp_https_server.h: No such file or directory
```

Looks like the latest espressif32 (https://github.com/platformio/platform-espressif32.git) doesn't have the HTTPS server enabled anymore?

Also using a fresh example from https://github.com/hoeken/PsychicHttp/tree/master/examples/platformio with 

```
platform = espressif32
framework = arduino
```

hoeken, 2023-12-21:
I hate to say it, but it might be a problem with your setup.  There's a
github action that compiles that example on each commit.  I just pushed a
version with ssl enabled and it passed just fine:
https://github.com/hoeken/PsychicHttp/actions/runs/7290588206/job/19867714507

----

Zach Smith

proddy, 2023-12-21:
fixed it, I had removed and re-installed pio again on my Windows WSL. I think the vscode and local pio environments were conflicting.

closed 2023-12-21 by proddy

proddy, 2023-12-21:
for what it's worth, it wasn't the multi-setup but platformio getting mixed up between the espressif libraries. I was switching between `platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5` and `platform = https://github.com/platformio/platform-espressif32.git`

solution is to use `pio package uninstall` and let pio find the install the correct packages.

hoeken, 2023-12-21:
Have you gotten ESP-IDF v5 working with platformio + arduino?  Seems to be some drama between platformio and espressif...

proddy, 2023-12-22:
yes, kind of, I'm using Jason's special library he built for Tasmota for Arduino 3.0/IDF 5.1.(alpha 3):

`platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.12.10/platform-espressif32.zip`

But it's optimized to remove some libraries that you may need (like wifi_secure, mbedtsl, etc) so won't work for all.

There's an interesting thread over at github.com/platformio/platform-espressif32/issues/ 1238

T94T, 2024-04-30:
Try to set the platform version explicitly in `platformio.ini` like this `platform = espressif32 @ 6.6.0`.