hoeken/PsychicHttp #33: Help needed with the new PsychicJsonRequestCallback()

proddy, 2023-12-21:
I'm caught in a snag, with POSTs and JSON callbacks.

If I use your example code (which runs flawlessly) and take the `/api` endpoint as an example as shown below:

```C
webServer.on("/rest/api", HTTP_POST, [](PsychicRequest * request) {
        //load our JSON request
        StaticJsonDocument<1024> json;
        String                   body = request->body();
        DeserializationError     err  = deserializeJson(json, body);

        //create our response json
        StaticJsonDocument<128> output;
        output["msg"]    = "status";
        output["status"] = "success";
        output["millis"] = millis();

        //work with some params
        if (json.containsKey("foo")) {
            String foo    = json["foo"];
            output["foo"] = foo;
        }

        //serialize and return
        String jsonBuffer;
        serializeJson(output, jsonBuffer);
        return request->reply(200, "application/json", jsonBuffer.c_str());
});
```

it's fine.

But if I replace the `on()` with the json callback like

```C
webServer.on("/rest/api", HTTP_POST, [](PsychicRequest * request, JsonVariant & json_input) {
 // same code
}
```

it goes into a endless loop when called. Debugging at IDF level doesn't tell me much either.

I'm stuck. Any ideas?

hoeken, 2023-12-22:
I havent actually added in that .on() convenience call, ran into some weird #include issues trying to do it.  I'll give it another try.

hoeken, 2023-12-22:
Okay, so the #include hierarchy might be a bit of a ratsnest and makes me want to rip my hair out, but I moved things around until they worked and now there is that .on() convenience call.  Update git and give it a try.

```cpp
    //api - json message passed in as post body
    // curl -i -X POST -H "Content-Type: application/json" -d '{"foo":"bar"}' http://psychic.local/api
    server.on("/api", HTTP_POST, [](PsychicRequest *request, JsonVariant &json)
    {
      JsonObject input = json.as<JsonObject>();

      //create our response json
      PsychicJsonResponse response = PsychicJsonResponse(request);
      JsonObject output = response.getRoot();

      output["msg"] = "status";
      output["status"] = "success";
      output["millis"] = millis();

      //work with some params
      if (input.containsKey("foo"))
      {
        String foo = input["foo"];
        output["foo"] = foo;
      }
  
      return response.send();
    });
```

proddy, 2023-12-22:
That worked, thanks!

I forgot to mention earlier I had already extended the on() to support ny adding to

**PsychicHttpServer.cpp:**
```C
PsychicEndpoint * PsychicHttpServer::on(const char * uri, http_method method, PsychicJsonRequestCallback fn) {
    PsychicJsonHandler * handler = new PsychicJsonHandler(fn);

    return on(uri, method, handler);
}
```

**PsychicHttpServer.cpp:**
```C
typedef std::function<esp_err_t(PsychicRequest * request, JsonVariant & json)> PsychicJsonRequestCallback;

PsychicEndpoint * on(const char * uri, http_method method, PsychicJsonRequestCallback onRequest);
```

to make things match AsyncWS.

I also added `-D ARDUINOTRACE_ENABLE=0` to platformio.ini to remove those DUMP() and TRACE() debugs outputting to Serial.

And added msgpack support.

When I have it all running I'll drop in a few PRs.

great work so far, thanks!

hoeken, 2023-12-22:
The ArduinoTrace stuff is either temporary or sprinkled into places where the code should never reach to help in diagnosing potential bugs.  msgpack looks cool, that would definitely be a nice addition.

closed 2023-12-22 by hoeken