hoeken/PsychicHttp #36: Websocket example with xQueue

kbesson, 2023-12-23:
Hello !

First, thank you so much for this library !

I was pulling my hair with AsyncWebServer... It was unstable with random crash on reconnection.  It's working way better with PsychicHttp, page loading is fast, no more random crash !

Now I try to set a websocket server with xQueue but I don't understand exactly how to implement it. I've check the docs and your reply on issue #11 (https://github.com/hoeken/PsychicHttp/issues/11#issuecomment-1848828317).

My client connect successfully, but the queue doens't seems to be processed.

Can you post a complete working example for this usecase ?

Thank you !

hoeken, 2023-12-24:
I'm glad its working for you :)

My real world example with websockets is here:

https://github.com/hoeken/yarrboard-firmware/blob/main/src/server.cpp
and
https://github.com/hoeken/yarrboard-firmware/blob/main/src/protocol.cpp

I will try to pull out the important bits into a proper example for you.

hoeken, 2023-12-24:
@kbesson here you go: https://github.com/hoeken/PsychicHttp/tree/master/examples/websockets

kbesson, 2023-12-25:
Wow, that was quick ! 😃

Thanks ! I will try this tomorrow !

kbesson, 2023-12-27:
Hey @hoeken

Just wanted to thank you for your amazing work !

I've implemented your example in my project, multi clients with message queue is working really well, free heap increased and my webUI is pretty fast with no crash.

I hope people will start to migrate massively from AsyncWebServer to PsychicHTTP, it's a far better solution.

I think you can close this issue. :)

Thanks again !

proddy, 2023-12-27:
That's a great outcome @kbesson . My heap mem has increased by 10kb when I switched from AsyncWebServer to this library and now figuring out what that could be. Do you have an open source code you're willing to share?

kbesson, 2023-12-27:
Not at this moment, I still need to cleanup a lot of things and work on the Modbus implementation before publishing it. The project is a webUI for a solar MPPT charger with Modbus RTU connectivity, so I prefer to double check everything before make it public.

Regarding your issue, witch fork of AsyncWebServer did you use ? Are you using xQueue ? How many endpoints do you have ?

proddy, 2023-12-28:
I was using a modified fork of https://github.com/yubox-node-org/ESPAsyncWebServer with also a highly customized AsyncTCP library, and have around 80 endpoints, mostly because of the dual GET and POST for reading/writing settings. I'm not using xQueue, just plain server.on() and EventSource right now. I'll need to go deeper into what is using all the additional heap.

hoeken, 2023-12-28:
@kbesson so glad its working well for you.  let me know if you have any other issues.

closed 2023-12-28 by hoeken