hoeken/PsychicHttp #38: Possible to send multiple images (stream) - e.g. webcam on ESP32CAM?

HowardsPlayPen, 2023-12-24:
Background: my aim is to use an ESP32CAM as a webcam using the commonly found code on the web - but to also include ElegantOTA for simple OTA updates (which brought in ESPAsyncWebServer). Ideally I wanted to be able to stream the images - and handle more than one client (the commonly available code only handles one client at a time) 

So this library looks great - better / more performant than ESPAsyncWebServer (and still supported) 

Question: I can see the documentation say:
      "You can not send more than one response to a single request." 
which seems to categorically rule out my request(? Unless there is a distinction between multiple Response objects versus sending multiple blobs over a single connection?)

I could use WebSockets - so again this library looks great if that is the case.

Apologies if this is a pointless request or asked before (I could not find it and I believe I went through all Issues)

Summary:
  Can I cache Response objects and send multiple blobs (e.g. call setContent / send multiple times?).

hoeken, 2023-12-24:
Websockets xQueue example: https://github.com/hoeken/PsychicHttp/tree/master/examples/websockets

HowardsPlayPen, 2023-12-27:
I have now got a local version of ElegantOTA updated to have an extra option to support PsychicHttp and it is working (using the Response option you separately mentioned for adding gzip headers) - so I am now coming back to look at this question again.
My understanding (possibly wrong I admit) is that HTTP1.1 should just mean keeping the socket connection alive and send new "reply()" with further messages (e.g. images) - that is what the existing code for the ESP32-CAM seems to do anyway.

I will use the WebSockets going forward for all manner of uses - but for the webcam side your mention of implementing a new handler sounds useful. Could you give a few lines of description of how you would do it (I say this as I just read that one should not cache handlers but instead cache socket handles(?) - so I would appreciate a simple pseudo code example of how you would see a handler being retained for sending multiple responses?

Thanks for all your help so far

hoeken, 2023-12-28:
@HowardsPlayPen if you have a link to a blog post or something describing what you'd like to do that would help me understand it better.

that being said, I would probably start with the EventSource stuff.  It has a custom handler, client, and response to generate the specific EventSource information.  You might not need all of that, since the Response generates its own HTTP header, the client has custom functions to send events, and the handler overrides the onOpen, onClose, and handleResponse to add its own custom hooks.

Hard to say exactly what would need to change without seeing something with more details on what you're trying to do.  Can you send me that esp32-cam code you're looking to replicate?

HowardsPlayPen, 2023-12-28:
The current cam code only responds to a single client at a time - so has a loop taking over the main thread. The HTTP header at the beginning is therefore telling the client to expect multiple parts and each new image has a Header showing it is a new part (but clearly this is obvious when you read the code so I do not mean to teach you to suck eggs..)

```

WiFiClient client = server.client();
String response = "HTTP/1.1 200 OK\r\n";
response += "Content-Type: multipart/x-mixed-replace; boundary=frame\r\n\r\n";
server.sendContent(response);

while (1)
  {
      cam.run();
      if (!client.connected())
          break;
      response = "--frame\r\n";
      response += "Content-Type: image/jpeg\r\n\r\n";
      server.sendContent(response);

      client.write((char *)cam.getfb(), cam.getSize());
      server.sendContent("\r\n");
      if (!client.connected())
          break;
  }
```

All I want to do is remove the above while(1) loop and merely push a new Handler / Request into a container to hold all active connections - then wait for the next cam image to be available, i.e.

```
cam.run()
```

Then use Handler.reply() for each of the active connections using the buffer 

```
      Handler.reply( (char *)cam.getfb(), cam.getSize());
```
One example GitHub project is:

https://github.com/rzeldent/esp32cam-rtsp/blob/main/src/main.cpp

see line 232 handle_stream()

I will have a look at the EventSource code though as that might be show the obvious solution.

hoeken, 2023-12-28:
Okay yep I see how it works and yeah you should definitely be able to
modify the event source stuff to do this.  Send that first set of headers
in the handler and then modify the client to send the stuff in the while
loop and you should be able to handle multiple clients no problem.

Let me know if you run into any issues.

DaeMonSxy, 2024-01-16:
I would be more than happy to see some esp32cam - webserver codes in examples. 
there are meny codes for non async, then few async-webserver codes- non of them seem to be stable to me and usefulll for daily use.

06GitHub, 2024-06-04:
Has someone already implemented **video webcam streaming with PsychichHttp** ? If yes, could example be shared ?

This **ESPAsyncWebServer** code provided by me-no-dev https://gist.github.com/me-no-dev/d34fba51a8f059ac559bf62002e61aa3 works OK (for my project).

I would like to migrate this ESPAsyncWebServer code to **PsychicHttp** (if possible ?), or use another working solution for video webcam streaming, so maybe someone has already done the job ?

Thanks

06GitHub, 2024-06-16:
For those interested, example below handles **streaming video with PsychicHttp.**
Limitation : **handles one client** (ie video stream is not sent to all clients connected to server, only to client which triggers the request).

Code, screenshots and output are provided.

**CONFIGURATION**
Arduino IDE 1.8.19
arduino-esp32 v2.0.17
ESP32-S3

**CODE**
```
/*********
  Camera video streaming with PsychicHttp server
  CREDITS
  https://github.com/hoeken/PsychicHttp/blob/master/src/PsychicResponse.cpp
  https://gist.github.com/me-no-dev/d34fba51a8f059ac559bf62002e61aa3
  https://RandomNerdTutorials.com/esp32-cam-video-streaming-web-server-camera-home-assistant/
*********/

char* TAG="CAM"; // ESP_LOG

#include "Arduino.h"
#include <WiFi.h>
#include "esp_camera.h"
// PsychicHttp https://github.com/hoeken/PsychicHttp
#include <ESPmDNS.h>
#include <PsychicHttp.h>
PsychicHttpServer server; // main server object http (not https)

//Replace with your network credentials
const char* ssid = "MYSSID";
const char* password = "MYPASSWORD";

// # camera module pinout (OV5640)
#define PWDN_GPIO_NUM    42
#define RESET_GPIO_NUM   -1
#define XCLK_GPIO_NUM    -1  // XLCL is not provided by ESP32, it is embedded on breakout board (crystal)
#define SIOD_GPIO_NUM    40
#define SIOC_GPIO_NUM    41
#define Y9_GPIO_NUM      46
#define Y8_GPIO_NUM      39
#define Y7_GPIO_NUM      2
#define Y6_GPIO_NUM      38
#define Y5_GPIO_NUM      6
#define Y4_GPIO_NUM      14
#define Y3_GPIO_NUM      48
#define Y2_GPIO_NUM      47
#define VSYNC_GPIO_NUM   4
#define HREF_GPIO_NUM    17
#define PCLK_GPIO_NUM    45

// content_type, header and boundary definition
#define PART_BOUNDARY "123456789000000000000987654321"
static const char* _STREAM_CONTENT_TYPE = "multipart/x-mixed-replace;boundary=" PART_BOUNDARY;
static const char* _STREAM_BOUNDARY = "\r\n--" PART_BOUNDARY "\r\n";
static const char* _STREAM_PART = "Content-Type: image/jpeg\r\nContent-Length: %u\r\n\r\n";

void startCameraServer(){
  server.listen(80); // start PsychicHttp server on http
  
  // endpoints
  server.on("/stream",[](PsychicRequest*req) {
    ESP_LOGI(TAG,"/stream");
    camera_fb_t * fb = NULL;
    esp_err_t res = ESP_OK;
    size_t _jpg_buf_len = 0;
    uint8_t * _jpg_buf = NULL;
    char * part_buf[64];
  
[... 186 lines cut ...]
[ 22521][I][psychic_videostream_OK.ino:100] operator()(): [CAM] header : part_buf Content-Type: image/jpeg
Content-Length: 8464

 hlen 50
[ 22533][I][psychic_videostream_OK.ino:101] operator()(): [CAM] result sending header ESP_OK
[ 22549][I][psychic_videostream_OK.ino:110] operator()(): [CAM] result sending frame is ESP_OK
[ 22558][I][psychic_videostream_OK.ino:119] operator()(): [CAM] boundary _STREAM_BOUNDARY 
--123456789000000000000987654321

[ 22569][I][psychic_videostream_OK.ino:120] operator()(): [CAM] result sending boundary ESP_OK
[ 22652][I][psychic_videostream_OK.ino:73] operator()(): [CAM] fb 0x3d8009c0 width 640 fb->buf 0x3d800a10 fb->len 8513
[ 22662][I][psychic_videostream_OK.ino:90] operator()(): [CAM] _jpg_buf 0x3d800a10 _jpg_buf_len 8513
[ 22671][I][psychic_videostream_OK.ino:100] operator()(): [CAM] header : part_buf Content-Type: image/jpeg
Content-Length: 8513

 hlen 50
[ 22684][I][psychic_videostream_OK.ino:101] operator()(): [CAM] result sending header ESP_OK
[ 22700][I][psychic_videostream_OK.ino:110] operator()(): [CAM] result sending frame is ESP_OK
[ 22709][I][psychic_videostream_OK.ino:119] operator()(): [CAM] boundary _STREAM_BOUNDARY 
--123456789000000000000987654321

[ 22720][I][psychic_videostream_OK.ino:120] operator()(): [CAM] result sending boundary ESP_OK
[snip]

```

hitecSmartHome, 2024-06-16:
How does this not trigger the watchdog?