hoeken/PsychicHttp #39: mbedtls_ssl_handshake returned -0x7780 with https request

myqel, 2023-12-25:
Thanks for that great work!
I just at the beginning to try PsychicHttp as a replacement for AsyncWebServer in a project I work on as I want to serve HTTPS, too.
For that project I need a captive portal (only AP, no network connection at all). IDE is VS Code / PlatformIO.
I used your encryption certificates in LittleFS for a first trial.
I struggled a whole day now with the following error during the AP connection establishment and with each new broswer request:
_E (169799) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (169799) esp_https_server: esp_tls_create_server_session failed
[169827][E][PsychicHttpServer.cpp:284] closeCallback(): [psychic] No client record 55_
I tested your server sketch to see how it should work (better), but I get the same error in the same situation.
Finally the page is loaded as https - I guess - ... Do you have an idea, what is going wrong?
Cheers; Michael

hoeken, 2023-12-28:
Looks like some kind of certificate error:  https://forums.mbed.com/t/error-0x7780-during-handshake/6883

hoeken, 2024-08-10:
I'm closing all these old issues.  Please re-open it if needed.

closed 2024-08-10 by hoeken