hoeken/PsychicHttp #40: ElegantOTA + reply() + gzip header

HowardsPlayPen, 2023-12-26:
Background:
I am updating ElegantOTA code in my workspace to use PyschicHttp instead of AsyncWebServer (in a test code build - so I can prove it works before integrating with my code).  

Issues:
1) In ElegantOTA it sends a response using (gzipped) html coming from memory - so uses send(..., content_len) as it knows the length of the memory block. I could use reply(...) but it would then internally use str_len() and this concerns me as I do not know if the normal string terminator could ever appear in a gzip block.  
  My preference would be to add another reply() method taking the extra parameter for content_len - does this ring alarm bells for you or is it safe to add?

2) The block being sent is a gzipped block so the current code (to be replaced) does:
        response->addHeader("Content-Encoding", "gzip");

   but as far as I can see the request object I have does not allow Headers to be added - and internally the reply() code sets the headers in the response object (although not currently an examples of gzip). 
  I can see that the header functionality could be added to Request or have another way of getting it to the Response object being used. Do you have any views + recommendations?

Obviously I accept I might be missing some obvious points or raising the wrong issues - so apologies in advance if this is the case.

hoeken, 2023-12-26:
Yes!  Finally an easy one :)

1. Yeah shouldn't be a problem, probably an oversight on my part for not adding that.  If you want to add that and send a PR I would happily accept it.

2. For some reason ESPAsync confuses requests and responses.  Requests come from the client, responses come from the server.  The request->reply() function is a convenience function that internally creates a response and sends it off.  If you need to send custom headers, probably best to create the response, add your headers + content then call response->send() at the end.  LMK if you need clarification.  Here's an example from my code with gziped content, etc:

```cpp
  PsychicResponse response(request);
  response.setCode(200);
  response.setContentType("text/html");

  // Tell the browswer the contemnt is Gzipped
  response.addHeader("Content-Encoding", "gzip");

  // And set the last-modified datetime so we can check if we need to send it again next time or not
  response.addHeader("Last-Modified", last_modified);
  response.addHeader("ETag", index_html_gz_sha);

  //add our actual content
  response.setContent(index_html_gz, index_html_gz_len);

  return response.send();
```

HowardsPlayPen, 2023-12-26:
Perfect! I shall add the reply code and send a PR (you are a hero and I thank you for the quick response)

I shall give your code a go for the Response - it looks good to me.

(Thanks!)

closed 2023-12-26 by HowardsPlayPen