hoeken/PsychicHttp #41: Multi Core support similarly to AsyncTCP?

Abasz, 2023-12-27:
I was wondering whether multi core support is planned or not. The AsyncTCP library does this by using the `xTaskCreateUniversal` function that creates a task in a way that if there is free core to run the task on chooses that if not uses the `xTaskCreate` (like your code).

Currently the webserver on my dual core esp32 runs on the same core as the main loop, and I wanted to off load it to the second one to improve overall performance.

Thanks

proddy, 2023-12-27:
See Zach's response to this in https://github.com/hoeken/PsychicHttp/issues/12#issuecomment-1868370757

Multi-threading is native in IDF 5.1.x on Arduino 3.0 (which is still in Alpha). We had issues getting it compiled. Feel free to hack around. I was using:

```
platform = https://github.com/platformio/platform-espressif32.git
; platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
platform_packages = https://github.com/espressif/arduino-esp32.git#3.0.0-alpha2
```

zekageri, 2023-12-27:
I think async TCP multi core is not the same. Async TCP just runs every request in a separate task on a core synchronously. The idf threaded solution would fire a new task for every request. Is my understanding right?

zekageri, 2023-12-27:
The current physic implementation does the same as async tcp

Abasz, 2023-12-27:
> I think async TCP multi core is not the same. Async TCP just runs every request in a separate task on a core synchronously. The idf threaded solution would fire a new task for every request. Is my understanding right?

Yes you are correct. Basically AsyncTCP just offloads the server to a separate core and creates a loop there and the loop runs on that core. Its not creating a thread for each request.

Abasz, 2023-12-27:
> ent physic implementation does the same as async tcp. One task on one core.

Yes but that task is on the "main" core i.e. core 1, while on a dual core esp32 core 0 does basically nothing. This is the reason for my question.

zekageri, 2023-12-27:
I think that the -1 which specified on task create for the core is not doing what you think. In my understanding its just picks an available core at create time and the task will run on that core which was available on creation. I dont think that the scheduler can pass the task to different cores at runtime.

Abasz, 2023-12-27:
> I think that the -1 which specified on task create for the core is not doing what you think. In my understanding its just picks an available core at create time and the task will run on that core [...]

I assumed this, based on the docs:

xCoreID: If the value is tskNO_AFFINITY, the created task is not pinned to any CPU, and the scheduler can run it on any core available. Values 0 or 1 indicate the index number of the CPU which the task should be pinned to. Specifying values larger than (portNUM_PROCESSORS - 1) will cause the function to fail.

But its possible you are correct. Actually in my code I specifically pin it to core 0 and the callbacks e.g. on websocket or sending the page is executed from that core (my main code runs on core 1). So I never tested the run time behaviour.

zekageri, 2023-12-27:
In my tests the async tcp peflrformed the same with core 0 and -1. :/

Abasz, 2023-12-27:
> In my tests the async tcp peflrformed the same with core 0 and -1. :/

That is very interesting as for me when I check the `xPortGetCoreID()` I get 0 if I initialize it with 0, showing that it is being run on core 0. If you check the implementation of the `xTaskCreateUniversal`, it shows that when the core ID is -1 xTaskCreate and when called with zero or 1, it calls `xTaskCreatePinnedToCore`

Abasz, 2023-12-27:
So it turn out there is a core_id property on the `httpd_ssl_config_t` `httpd` property that can set on which core server should run. If I set it to zero it runs on core 0. 

Currently this can only be set from within the the class, but when I have more time I may submit a PR (there are other interesing settings like max_open_sockets, etc. that are non-accessible that could be exposed via setters)

zekageri, 2023-12-27:
Yeah, the whole config object is exposed

hoeken, 2023-12-28:
@Abasz  trying to catch up, responses below

1. we use esp-idf http server under the hood, and they start the server with this call.  It is pinned to the core you set with server.config.core_id

```cpp
static inline int httpd_os_thread_create(othread_t *thread,
                                 const char *name, uint16_t stacksize, int prio,
                                 void (*thread_routine)(void *arg), void *arg,
                                 BaseType_t core_id)
{
    int ret = xTaskCreatePinnedToCore(thread_routine, name, stacksize, arg, prio, thread, core_id);
    if (ret == pdPASS) {
        return OS_SUCCESS;
    }
    return OS_FAIL;
}
```

2. as @zekageri pointed out, the config struct that is passed into esp-idf is a public member of the PsychicHttpServer class so you can adjust those parameters how you want.  Just make sure to make your changes before you call server.listen().  
3. Also worth noting that if you use SSL, it changes to server.ssl_config with the server.ssl_config.httpd being the same as server.config.  This is an esp-idf thing so we just have to work with hit.
4. As noted in another post, we should be able to get true multi-threaded support working once Arduino is upgraded to the current esp-idf v5.1.2.  We will have more control over that too, so the possibility of running on all cores is there.

Abasz, 2023-12-28:
Thanks for the above, I totally missed that the settings struct is public property... That makes life easier of course.

closed 2023-12-28 by Abasz

proddy, 2024-01-01:
old topic, but on point 4, [Arduino 6.5.0](https://github.com/platformio/platform-espressif32/releases/tag/v6.5.0) is out with IDF 5.1.2 so I tried `ENABLE_ASYNC` and played with task setting (see https://github.com/hoeken/PsychicHttp/issues/28#issuecomment-1872934088) but no luck yet