hoeken/PsychicHttp #42: Error on loading a multifile page: esp-tls-mbedtls: create_ssl_handle failed, returned [0x8017] (ESP_ERR_MBEDTLS_SSL_SETUP_FAILED)

Abasz, 2023-12-27:
I have a web page that I would like to serve from the ESP32 (its actually an angular app) over https that has an index.html and 5 other files (js and css). 

The index.html file loads correctly but for some reason I always receive the following error for some of the files, so the page never really loads. It totally random which file gets loaded and which gets this error.

E (93906) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (93908) esp-tls-mbedtls: create_ssl_handle failed, returned [0x8017] (ESP_ERR_MBEDTLS_SSL_SETUP_FAILED)
E (93908) esp_https_server: esp_tls_create_server_session failed
[ 93944][I][PsychicHttpServer.cpp:262] closeCallback(): [psychic] Client disconnected 52
[ 93944][E][PsychicHttpServer.cpp:285] closeCallback(): [psychic] No client record 52

I use self signed certificates, tried chrome as well as firefox. Also tried mobile chrome, which performed better as in that case on 1 out of 10 reloads I was able to load the page properly.

zekageri, 2023-12-27:
Probably insufficient ram becauee of the tls

Abasz, 2023-12-27:
> Probably insufficient ram becauee of the tls

You are right... removing stuff (e.g. Bluetooth) increases the free memory and allows proper loading... 

Thanks for pointing me into the right direction. 

Do you think it would be feasible to implement a que that checks free memory and if its low it parks the request and once memory is freed up it completes?

zekageri, 2023-12-27:
Yes it would be absolutely doable. I had a fork Async TCP which did almost the same. When it couldnt serve the request for any reason it sent an http header to the client to try again after a few seconds. It would be really good for this lib too.

Abasz, 2023-12-27:
> Yes it would be absolutely doable. I had a fork Async TCP which did almost the same. When it couldnt serve the request for any reason it sent an http header to the client to try again after a few seconds. It would be really good for this lib too.

I have been thinking about this and I think there is a chicken and egg situation here: I cannot send a header without TLS handshake, but I cannot have TLS handshake because there is not enough memory. The request needs to be parked prior to the TLS handshake so there is no usage of memory I think.

zekageri, 2023-12-27:
Yeah, it should be done by the lib instead of user code. I dont know how could we do this with tls however

hoeken, 2023-12-28:
@Abasz you could probably implement this code in the server.onOpen() callback.  i'm not 100% sure but I think you could call client.close() to close the connection if memory is low.

Also, make sure you're using the current github version of the code.  I reduced the open sockets available to 2 because of these issues.  You might want to try reducing that to 1 and see if that solves your problem as well:  ```server.ssl_config.httpd.max_open_sockets = 2;```

Abasz, 2023-12-28:
> You might want to try reducing that to 1 and see if that solves your problem as well: `server.ssl_config.httpd.max_open_sockets = 2;`

Yes, I have reduced the open sockets to 1 that helped. Also I reduced the number of files to 1 fully bundled HTML (inline js and css) helped too as the browser makes only one call.

>  you could probably implement this code in the server.onOpen() callback. i'm not 100% sure but I think you could call client.close() to close the connection if memory is low.

This could work, assuming that there is no concurrent client requests (as in that case close may potentially occur while the client is still downloading data, or a new WS connection force terminates the previous one). If there is only one request at the time, (i.e. new connection requests are not received until the previous one is fully completed) this should actually work. But otherwise this could cause instability in my view for clients as an inflight transfer may be terminated when a new one comes in. 

That is why I was thinking about parking until memory is available rather than force closing, but for now I am not sure how this could be implemented.  as currently I only see this via delaying the completion of the onOpen method, but I am not sure how this can be done in a non-blocking way.

hoeken, 2024-08-10:
I'm closing all these old issues.  Please re-open it if needed.

closed 2024-08-10 by hoeken