hoeken/PsychicHttp #44: [Question] understanding file response.

Chris--A, 2023-12-30:
My question is in regards to this close:

https://github.com/hoeken/PsychicHttp/blob/26b49f73dca48fa333361fb4a4cfeebca802eeda/src/PsychicFileResponse.cpp#L154

There may be other reasons I'm not seeing, but this appears to be redundant as the destructor closes the file. Additionally, the send function only closes the file when its size exceeds the chunk size, and leaves open if size < chunk.

If you have a concept started for a stream response, please let me know where I can help, as it will be of a huge benefit to reduce heap usage. I'm envisioning a `PsychicStreamResponse` class that could be used internally by the file response, or along side the chunked options. I would find great benefit using it directly to stream huge data sets (JSON from sqlite3).

Tonight when I have more free time, I will have a look and see if I can start a PR.

hoeken, 2023-12-30:
Yep, that looks like a bug.  I'm a bit busy now, but I can def get to this in the new year if you haven't sent a PR by then.

As for a stream response, I don't have anything in the works yet, but would be open to one.  It shouldn't be too hard to implement.  Is there one in ESPAsync we can use for a reference, or in another project?  If you have example code you could share that uses that it would help as well.

Chris--A, 2023-12-31:
I have a branch underway to fix this issue and a potential stream candidate. Will do some testing and create a PR.

PR #45 (2024-01-07): Addition of PsychicStreamResponse & update to PsychicFileResponse.
First, this PR makes a few changes to `PsychicFileResponse`. There are a few instance variables that are not needed, and a few redundant calls. Resolves #44 

The main feature is a new class `PsychicStreamResponse` to allow using the `Print` interface to stream raw data.

I stayed away from copying what ESPAsyncWebServer does, especially as their implementation is limited in size (https://github.com/me-no-dev/ESPAsyncWebServer/issues/179#issuecomment-310047127).

This implementation makes use of the existing `ChunkPrinter` usin [... cut]
closed 2024-01-07 by hoeken