hoeken/PsychicHttp #46: larger ram usage compared to espasyncwebserver

rjjrbatarao, 2024-01-01:
I've just migrated to PsychicHttp and in comparison espasyncwebserver has free memory of 210000 compared to psychichttp has only 170000 free. 

im using simple http server
```
#include <PsychicHttp.h>
PsychicHttpServer server;
```
the handlers
```
  // Optional, defines the default entrypoint
  server.on("/", HTTP_GET, [](PsychicRequest * request) {
    PsychicResponse response(request);
    response.setCode(200);
    response.setContentType("text/html");
    response.addHeader("Content-Encoding", "gzip");
    response.addHeader("Cache-Control", "public, max-age=2678400");
    response.setContent(static_files::f_index_html_contents, static_files::f_index_html_size);
    return response.send();
  });

  // Create a route handler for each of the build artifacts
  for (int i = 0; i < static_files::num_of_files; i++)
  {
    server.on(static_files::files[i].path, HTTP_GET, [i](PsychicRequest * request) {
      PsychicResponse response(request);
      response.setCode(200);
      response.setContentType(static_files::files[i].type);
      response.addHeader("Content-Encoding", "gzip");
      response.addHeader("Cache-Control", "public, max-age=2678400");
      response.setContent(static_files::files[i].contents, static_files::files[i].size);
      return response.send();
    });
  }

```

rjjrbatarao, 2024-01-01:
closing this issue it was caused by WiFi.useStaticBuffers(true); that i've recently added

closed 2024-01-01 by rjjrbatarao

proddy, 2024-01-02:
Hi @rjjrbatarao - can you roughly say much free mem is used between using AsyncWS and PsyhicHtttp? I'm still struggling to figure out why PsychicHttp takes up more heap mem and is also has a slower response time after my port.

rjjrbatarao, 2024-01-04:
@proddy my project is around 150+ kb gzipped html, I can't say how much since my project has no rest apis and functions so far its just static webserver at the meantime and memory usage with async vs psychic is almost thesame, our clue is when i start building the rest endpoints we would see some differences may be leaks on json etc who knows. download of assets + render is constant @ 400ms, hmm I forgot to check async probably faster or not?