hoeken/PsychicHttp #48: request->body() is incomplete?

zekageri, 2024-01-02:
I have a handler which looks like this:

```cpp
esp_err_t HardwareHandler::saveHardware(PsychicRequest *request) {
    DynamicJsonDocument doc(request->contentLength() * 3);
    DeserializationError error = deserializeJson(doc, request->body()); // Code fails to deserialize. InvalidInput
    if (error) {
        Serial.println(request->body());
        return request->reply(500,"text/plain",error.c_str());
    }
    boolean status = hwConfig.updateModule(doc);
    return request->reply(status ? 200 : 400);
}
```

I can't deserialize the incoming json because it is incomplete.
I adjusted the `MAX_REQUEST_BODY_SIZE` to `(30*1024)` in `PsychicCore.h` but no success.

The handler is called like this

```cpp
esp_err_t HardwareHandler::handleRequest(PsychicRequest *request) {
    request->loadBody();
    request->loadParams();
    if (request->path() == "/hardwares/save" && request->method() == HTTP_POST) {
        return saveHardware(request);
    }
    return ESP_OK;
}
```

If i print the body raw i get garbage at the end. The request should fit into 10k at max.
I also have plenty of RAM left.
Here are the headers

[image]

Some garbage data from the print of the raw request body

```cpp
���{"moduleAddress":13,"name":"b���OUT 29","displayName":"R32-Oz���9","type":0,"current":0,"faur���de":0,"relayCount":206,"regNJ���r":1,"pinNumber":12,"readablB���Number":29,"state":false,"vaZ���:0,"id":"R32_CF4F6760"},"R32R���42A8B":{"moduleAddress":13,"����":"R32-OUT 30","displayName"����2-OUT 30","type":0,"current"����faultCode":0,"relayCount":15����egNumber":1,"pinNumber":13,"����ablePinNumber":30,"state":fa����"value":0,"id":"R32_8F842A8B����R32_745132C1":{"moduleAddres����3,"name":"R32-OUT 31","displ궪�me":"R32-OUT 31","type":0,"cⶪ�nt":0,"faultCode":0,"relayCo����:100,"regNumber":1,"pinNumbe򶪪4,"readablePinNumber":31,"s tʶ��:false,"value":0,"id":"R32_7¶��2C1"},"R32_6CD83A42":{"modulڶ��ress":13,"name":"R32-OUT 32"Ҷ��splayName":"R32-OUT 32","typ*���,"current":0,"faultCode":0,""���yCount":22022,"regNumber":1,:���Number":15,"readablePinNumbe2���2,"state":false,"value":0,"i
���R32_6CD83A42"}}}
```

zekageri, 2024-01-03:
Here is the ArduinoJson assistant output for the object

[image]

zekageri, 2024-01-03:
Some strange thing is happening. Doesn't matter if i increase the `MAX_REQUEST_BODY_SIZE` to even `(100*1024)` the request body is incomplete. What is going on?

proddy, 2024-01-03:
just quick thoughts here
- are you sure you're using the PsychicJsonRequestCallback - otherwise the json payload on the POST won't be identified, then you only need to send the jsonobject in your saveHardware()
- I'm having a lot of fun with PR #45 and chunk streaming. See an example at https://github.com/hoeken/PsychicHttp/issues/37#issuecomment-1874075859

zekageri, 2024-01-03:
I don't use it, it was not available before.
I have a custom class made to handle specific requests like this one

```cpp
server.on("/hardwares/*", HTTP_GET, &hardwareHandler);
server.on("/hardwares/*", HTTP_POST, &hardwareHandler);

esp_err_t HardwareHandler::handleRequest(PsychicRequest *request) {
    request->loadBody();
    request->loadParams();
    if (request->path() == "/hardwares/scann") {
        return scann(request);
    } else if (request->path() == "/hardwares/save" && request->method() == HTTP_POST) {
        return saveHardware(request);
    }
   // More handlers...
    return ESP_OK;
}

esp_err_t HardwareHandler::saveHardware(PsychicRequest *request) {
    DynamicJsonDocument doc(request->contentLength() * 2);
    DeserializationError error = deserializeJson(doc, request->body());
    if (error) {
        Serial.println(request->body());
        return request->reply(500,"text/plain",error.c_str());
    }
    int status = hwConfig.updateModule(doc) ? 200 : 400;
    return request->reply(status);
}
```

zekageri, 2024-01-03:
Thanks for the example but I don't send json in this handler, i send 200 or 400 or 500, i want to accept it.

zekageri, 2024-01-03:
I will try it like this and see what happens

```cpp
server.on("/hardwares/save", HTTP_POST, [](PsychicRequest *request){
    return hardwareHandler.saveHardware(request);
});
```

zekageri, 2024-01-03:
Same, `InvalidInput` error message from ArduinoJson...

zekageri, 2024-01-03:
Will check it like this

```cpp
server.on("/hardwares/save", HTTP_POST, [](PsychicRequest *request,JsonVariant &json){
    //return hardwareHandler.saveHardware(request);
    serializeJson(json, Serial);
    return ESP_OK;
});
```

zekageri, 2024-01-03:
Well, it prints nothing.

proddy, 2024-01-03:
if you convert the jsonvariant into a doc?

```
    if (!json.is<JsonObject>()) {
            StaticJsonDocument<1024> input_doc;
            JsonObject                                 input = input_doc.to<JsonObject>();
            serializeJson(input, Serial);
            return hardwareHandler.saveHardware(request, input);
        }
```

ArduinoJson 7 which is just out makes this easier, but above is the old way

Chris--A, 2024-01-03:
I started with V7, however, I found that I was getting some extra heap fragmentation from a small amount of use. I switched back to V6 so I could use static buffers. I might change things soon, ordered 2x new ESP32-S3 with 16MB flash and 8MB PSRAM, so I can be a bit more flexible.

zekageri, 2024-01-03:
> if you convert the jsonvariant into a doc?
>
> ```
>     if (!json.is<JsonObject>()) {
>             StaticJsonDocument<1024> input_doc;
>             JsonObject                                 input = [...]

Will Check!

zekageri, 2024-01-03:
> I started with V7, however, I found that I was getting some extra heap fragmentation from a small amount of use. I switched back to V6 so I could use static buffers. I might change things soon, ordered 2x new ESP32-S3 with 16MB flash and 8MB PSRAM, so I can be a bit more flexible.

Wow, thats not what we are used to from ArduinoJson

Chris--A, 2024-01-03:
> Wow, thats not what we are used to from ArduinoJson

I'll have to do more testing, likely a me issue, but I do also prefer stack allocations where I can which is a V6 feature.

proddy, 2024-01-03:
> > Wow, thats not what we are used to from ArduinoJson
> 
> I'll have to do more testing, likely a me issue, but I do also prefer stack allocations where I can which is a V6 feature.

same here, I use StaticJsonDocument for anything less than 1024 bytes, because of fragmentation (and max alloc buffer size).

zekageri, 2024-01-03:
With this line

server.on("/hardwares/save", HTTP_POST, [this](PsychicRequest *request, JsonVariant &json){});

The server just throws a 400 bad request to the client and i can't see anyting now.

zekageri, 2024-01-03:
I'm pulling my hair out

zekageri, 2024-01-03:
[image]
[image]

zekageri, 2024-01-03:
This is my handler now but does not print anything. My handler doesn't even called.

```cpp
server.on("/hardwares/save", HTTP_POST, [this](PsychicRequest *request, JsonVariant &json){
    Serial.print("RAW BODY: ");
    Serial.println(request->body());
    Serial.print("Json Variant: ");
    serializeJson(json, Serial);
    if (!json.is<JsonObject>()) {
        StaticJsonDocument<1024> input_doc;
        JsonObject input = input_doc.to<JsonObject>();
        Serial.println("Input Object: ");
        serializeJson(input, Serial);
    }
    return request->reply(500);
});
```

proddy, 2024-01-03:
also add `-D ARDUINOTRACE_ENABLE=1` as there are some extra debugging in the lib which could help you

proddy, 2024-01-03:
and if you're getting 400, it's probably because the json is incorrectly formatting. Look at the code in PsychicWebHandler::handleRequest() and add some more debugging lines to show the error code from ArduinoJson

zekageri, 2024-01-03:
Will try this thank!

zekageri, 2024-01-03:
It writes nothing to the Serial. O_o

zekageri, 2024-01-03:
handleRequest has some debugging already but it prints nothing

```cpp
esp_err_t PsychicWebHandler::handleRequest(PsychicRequest *request)
{
  //lookup our client
  PsychicClient *client = checkForNewClient(request->client());
  if (client->isNew)
    openCallback(client);

  /* Request body cannot be larger than a limit */
  if (request->contentLength() > request->server()->maxRequestBodySize)
  {
    ESP_LOGE(PH_TAG, "Request body too large : %d bytes", request->contentLength());

    /* Respond with 400 Bad Request */
    char error[60];
    sprintf(error, "Request body must be less than %u bytes!", request->server()->maxRequestBodySize);
    httpd_resp_send_err(request->request(), HTTPD_400_BAD_REQUEST, error);

    /* Return failure to close underlying connection else the incoming file content will keep the socket busy */
    return ESP_FAIL;
  }

  //get our body loaded up.
  esp_err_t err = request->loadBody();
  if (err != ESP_OK)
    return err;

  //load our params in.
  request->loadParams();

  //okay, pass on to our callback.
  if (this->_requestCallback != NULL)
    err = this->_requestCallback(request);

  return err;
}
```

zekageri, 2024-01-03:
Sometimes i just get 400 bad request but sometimes it does do something.
Sometimes my cb gets called and passes an incomplete body

**Here is the result ( trimmed down the raw body print )**

```cpp
RAW BODY: "R32_C2B889A2":{"moduleAddress":13,"name":"R32-OUT 10","displayName":"R32-OUT 10","type":0,"current":0,"faultCode":0,"relayCount":247,<b�?
Json Variant: "M8_083A02B2"
Input Object:
{}
```

This is the handler

```cpp
server.on("/hardwares/save", HTTP_POST, [this](PsychicRequest *request, JsonVariant &json){
    Serial.print("RAW BODY: ");
    Serial.println(request->body());
    Serial.print("Json Variant: ");
    serializeJson(json, Serial);
    if (!json.is<JsonObject>()) {
        StaticJsonDocument<1024> input_doc;
        JsonObject input = input_doc.to<JsonObject>();
        Serial.println("\nInput Object: ");
        serializeJson(input, Serial);
    }
    return request->reply(500);
});
```

proddy, 2024-01-03:
I just tried your code above, in the standard test and it works. it prints out the json. Only the Input Object not because of the !json is incorrect

hoeken, 2024-01-03:
@zekageri it looks like the json you're sending might be bad or have non-ascii characters.  To debug where that is happening, can you have your sending code print that json, save it to a file.  Verify it with an online json parser, and then once you know its good, use curl to emulate the request.  If its still malformed its an issue in PsychicHttp.

```
curl -i -X POST -H "Content-Type: application/json" --data-binary "@/path/to/filename" http://psychic.local/api
```

zekageri, 2024-01-03:
Json is produced by a javascript client and stringified when sending. The transfer is happening with standard javascript fetch request. I have printed the object in the console on client side and i also verified it with ArduinoJson assistant. The dev console also confirms it is good.

hoeken, 2024-01-03:
Can you try the curl command and post the results?

zekageri, 2024-01-04:
Yap i will in a couple of hours. Btw this object and the front end code is not modified after i transitioned to psychic from asynctcp. And it worked before.

zekageri, 2024-01-04:
```curl
curl -i -X POST -H "Content-Type: application/json" --data-binary "@D:\test.json" http://hsh.local/upload/?path=/tempModule.json&partition=user
HTTP/1.1 200 OK
Content-Type: text/plain
Content-Length: 15
Server: HsH Server

Upload success!'partition' is not recognized as an internal or external command,
operable program or batch file.
```

Seems to me that curl doesn't know how to handle params? :D

zekageri, 2024-01-04:
Strange thing is that i have switched to upload handling instead of normal web request handler ( i tought that the size was the problem ) but the result is the same. It is trully looks like that my object is malformed but it is not. I attach the json object as a file here.

This is the object what is causing the issue.
[test.json](https://github.com/hoeken/PsychicHttp/files/13827213/test.json)

zekageri, 2024-01-04:
[image]

zekageri, 2024-01-04:
Will try a curl on a regular json endpoint in a minute

```cpp
server.on("/hardwares/newSave", HTTP_POST, [this](PsychicRequest *request, JsonVariant &json){
    serializeJsonPretty(json, Serial);
    return request->reply(200);
});
```

zekageri, 2024-01-04:
Same file with this endpoint

**ESP Serial:**
`"state"`
**Curl**
```
curl -i -X POST -H "Content-Type: application/json" --data-binary "@D:\test.json" http://hsh.local/hardwares/newSave
HTTP/1.1 200 OK
Content-Type: text/plain
Content-Length: 2
Server: HsH Server

OK
```

zekageri, 2024-01-04:
Maybe this is the problem in the lib

#if ARDUINOJSON_VERSION_MAJOR == 5
  #define ARDUINOJSON_5_COMPATIBILITY
#else
  #ifndef DYNAMIC_JSON_DOCUMENT_SIZE
    #define DYNAMIC_JSON_DOCUMENT_SIZE 4096
  #endif
#endif

#ifndef JSON_BUFFER_SIZE
  #define JSON_BUFFER_SIZE 4*1024
#endif

zekageri, 2024-01-04:
No, it wasnt. :(

zekageri, 2024-01-04:
Why is it jsonVariant? Why not JsonDocument in the cb?

zekageri, 2024-01-04:
If i print the raw `request->body()` it is also malformed in the `esp_err_t PsychicJsonHandler::handleRequest(PsychicRequest *request)`

```cpp
esp_err_t PsychicJsonHandler::handleRequest(PsychicRequest *request)
{
  //process basic stuff
  PsychicWebHandler::handleRequest(request);

  if (_onRequest)
  {
    #ifdef ARDUINOJSON_5_COMPATIBILITY
      DynamicJsonBuffer jsonBuffer;
      JsonVariant json = jsonBuffer.parse();
      if (!json.success())
        return request->reply(400);
    #else
      DynamicJsonDocument jsonBuffer(this->_maxJsonBufferSize);
      DeserializationError error = deserializeJson(jsonBuffer, request->body());
      if (error)
        return request->reply(400);

      Serial.println(request->body());
      Serial.println();
      serializeJsonPretty(jsonBuffer, Serial);
      JsonVariant json = jsonBuffer.as<JsonVariant>();
    #endif

    return _onRequest(request, json);
  }
  else
    return request->reply(500);
}
```

The start of the object is missing and the end of it.

zekageri, 2024-01-04:
Since String uses malloc() and i have set malloc to allocate from my ext ram, i really doubt that it would be insufficient memory. But i will check my free ram too.

zekageri, 2024-01-04:
As you can see i have `136640 byte` heap free and `3118471 byte` ext ram free before the curl request.
And almost the same after.

```cpp
[SYSTEM] - [HEAP][1]
Free:   136640
Min:    136640
Max:    136640
Contig: 90100
Diff:   32

[SYSTEM] - [EXT][1]
Free:   3118471
Min:    3118471
Max:    3118471
Contig: 3014644
Diff:   1552

[Server] - Connection #54 connected from 192.168.10.114
"name""name"

[SYSTEM] - [HEAP][1]
Free:   136672
Min:    136672
Max:    136672
Contig: 90100
Diff:   32

[SYSTEM] - [EXT][1]
Free:   3120023
Min:    3120023
Max:    3120023
Contig: 3014644
Diff:   1552
```

proddy, 2024-01-04:
I'd recommend you take the example 'platformio' code,  modify the `main.cpp` with just your code to help isolate the problem. Then we can all 'chip in' and make it work. That's what I did when I ran into memory & performance issues.

zekageri, 2024-01-04:
Yeah, i have no other choice left. Thanks. WIll try!

zekageri, 2024-01-04:
So i have copyed the platformio example and added this handler

```cpp
server.on("/testSave", HTTP_POST, [](PsychicRequest *request, JsonVariant &json) {
    // Because it can split out some strings which is valid JSON
    static const int MIN_LENGTH = 20;
    Serial.println();
    int len = serializeJsonPretty(json, Serial);
    Serial.println();
    return request->reply(len>MIN_LENGTH?200:500, "text/plain", len>MIN_LENGTH?"OK":"Failed");
});
```

On the client side i have created a button and a text area field where you can copy a json object or whatever json. When you click a button it makes a request to this endpoint which will print it out. I used my JSON from the file i uploaded here earlier and the result is the same. It just wont deserialize properly.

Here is the test repo: https://github.com/zekageri/Psychic_Upload_Test

zekageri, 2024-01-04:
On the front end side i do this

```js
const testJsonButton = document.querySelector("#testJsonBtn");
const testJsonField = document.querySelector("#testJson");

testJsonButton.addEventListener('click', async function(){
    if( testJsonField.value.length == 0 ){
        alert("No test json provided");
        return;
    }
    const jsonText = JSON.parse( testJsonField.value );
    const response = await fetch("/testSave",{
        method: "POST",
        headers: {
            "Content-Type": "application/json"
        },
        body: JSON.stringify(jsonText)
    });
    alert( response.ok?"Success":"Failed" );
    testJsonField.value = "";
}, false);
```

parsing the value of the textarea because i have an actual object in my tests and we can stringify it again to make it more real case.

zekageri, 2024-01-04:
I even got a crash now

```cpp
Core  1 register dump:
PC      : 0x4008cf01  PS      : 0x00060430  A0      : 0x8021a37c  A1      : 0x3ffdcdc0
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3f82ad70
A10     : 0x00000001  A11     : 0x0000007d  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x0000000e  A15     : 0x00060023  SAR     : 0x00000015  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4008cf01  LEND    : 0x4008cf11  LCOUNT  : 0xffffffff

Backtrace: 0x4008cefe:0x3ffdcdc0 0x4021a379:0x3ffdcdd0 0x40221416:0x3ffdd0e0 0x40221452:0x3ffdd170 0x4011e7b1:0x3ffdd1b0 0x400d888c:0x3ffdd200 0x400d9a11:0x3ffdd330 0x40133ee5:0x3ffdd380 0x40133f21:0x3ffdd3e0 0x4014d4d1:0x3ffdd400 0x4014dda1:0x3ffdd470 0x40198475:0x3ffdd4f0 0x4019726c:0x3ffdd520 0x40197a9a:0x3ffdd5c0 0x4019663c:0x3ffdd5e0 0x40233446:0x3ffdd600 0x40196761:0x3ffdd620

  #0  0x4008cefe:0x3ffdcdc0 in strlen at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:82

ELF file SHA256: bbf7c306adf71612

Rebooting..
```

zekageri, 2024-01-04:
Oh the crash was my bad because i passed a badly deserialized json string coming from the handler.

hoeken, 2024-01-04:
> curl -i -X POST -H "Content-Type: application/json" --data-binary ***@***.***:\test.json" http://hsh.local/upload/?path=/tempModule.json&partition=user

Put the URL in quotes.

zekageri, 2024-01-04:
Can you elaborate please?

proddy, 2024-01-04:
'click' to "click"

zekageri, 2024-01-04:
Am i this dumb or are you just messing with me

zekageri, 2024-01-04:
Every one of the urls are in quotes. On the frontend side as well as on the endpoints. Even in the curl command. I dont know what you are referring to

proddy, 2024-01-04:
oh, your example was JS code, then single-quote is ok

zekageri, 2024-01-05:
So I still don't understand.

zekageri, 2024-01-05:
The event handler is worked regardless of the type of the code.
It posts the json text fine. The problem was again with the body loading of the library.
It does this even with the platformio test sketch.

zekageri, 2024-01-05:
- I have enough internal and external ram
- The Json message fits to the default body size
- The body text is malformed even in the `handleRequest` method of the `PsychicWebHandler` class in the lib's code
- I have confirmed with the test sketch that it is not working properly

proddy, 2024-01-05:
> I started with V7, however, I found that I was getting some extra heap fragmentation from a small amount of use. I switched back to V6 so I could use static buffers. I might change things soon, ordered 2x new ESP32-S3 with 16MB flash and 8MB PSRAM, so I can be a bit more flexible.

FYI I just ported ArduinoJson to V7 and the memory footprint is identical. Which is comforting. Both Available Heap and Max Alloc are the same. It does feel slightly nippier, especially processing MsgPacks, but maybe that's just my brain wanted to see some overall improvement from all the silly hours I spent this week.

zekageri, 2024-01-05:
I'm porting it to V7 just now too. But that incomplete request body just makes me angryer. That is the only thing holding me back right now.

proddy, 2024-01-05:
I'll see if I can take a look at your example code today, still in https://github.com/zekageri/Psychic_Upload_Test ?

zekageri, 2024-01-05:
Yes, thank you very much. Use the file i uploaded here.

[9 more replies]

closed 2024-01-08 by zekageri