hoeken/PsychicHttp #50: WebSocket handler core panic

T94T, 2024-01-03:
We really like this implementation of an http/https web server. Thanks a lot!

Currently we are experiencing a problem that is causing the ESP32 to crash.

WebSocket client: https://github.com/hoeken/PsychicHttp/blob/26b49f73dca48fa333361fb4a4cfeebca802eeda/examples/websockets/data/www/index.html

The ESP32 runs the repository's example code:
```
server.on("/ws", &websocketHandler);
websocketHandler.onFrame([](PsychicWebSocketRequest *request, httpd_ws_frame *frame) {
  Serial.printf("[socket] #%d sent: %s\n", request->client()->socket(), (char *)frame->payload);
  return request->reply(frame);
});

```
Receiving WebSocket messages works fine without any errors. As soon as a packet is sent to the ESP32, it crashes:

```
[ 29135][I][PsychicWebSocket.cpp:175] handleRequest(): [psychic] frame len is 5
[ 29136][I][PsychicWebSocket.cpp:191] handleRequest(): [psychic] Got packet with message: TheMessage
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

```

```
#0  0x40111ae9:0x3ffd66d0 in httpd_sess_get at /esp_http_server/src/httpd_sess.c:173
#1  0x40112e19:0x3ffd6720 in httpd_ws_get_fd_info at /esp_http_server/src/httpd_ws.c:521
#2  0x400dba35:0x3ffd6740 in PsychicWebSocketHandler::handleRequest(PsychicRequest*) at .pio/libdeps/esp32-wroom/PsychicHttp/src/PsychicWebSocket.cpp:204 (discriminator 7)
#3  0x400dc0b9:0x3ffd6810 in PsychicEndpoint::requestCallback(httpd_req*) at .pio/libdeps/esp32-wroom/PsychicHttp/src/PsychicEndpoint.cpp:72
#4  0x401115b5:0x3ffd6890 in httpd_req_new at /esp_http_server/src/httpd_parse.c:776
#5  0x40111e2e:0x3ffd6930 in httpd_sess_process at /esp_http_server/src/httpd_sess.c:419 (discriminator 15)
#6  0x4010fb9c:0x3ffd6950 in httpd_process_session at /esp_http_server/src/httpd_main.c:178
    (inlined by) httpd_process_session at /esp_http_server/src/httpd_main.c:163
#7  0x4016e03a:0x3ffd6970 in httpd_sess_enum at /esp_http_server/src/httpd_sess.c:50 (discriminator 1)
#8  0x40110c40:0x3ffd6990 in httpd_server at /esp_http_server/src/httpd_main.c:228
    (inlined by) httpd_thread at /esp_http_server/src/httpd_main.c:250
```

The PlatformIO example (https://github.com/hoeken/PsychicHttp/tree/master/examples/websockets), on the other hand, works without any problems.

playmiel, 2024-01-11:
I did a search in the code and in http idf to find the source of the problem but I couldn't find any ideas?

T94T, 2024-02-01:
Thank you for looking into it. We have not found a solution either.

johnnytolengo, 2024-02-01:
Take a look on the heap memory to be sure that's not full.

playmiel, 2024-02-02:
I haven't been able to test it yet, but it's probably due to the "CORE_DEBUG_LEVEL=5" that I'd have to change to "CORE_DEBUG_LEVEL=2" or even less to fix the problem. Can you test it?

T94T, 2024-02-14:
Just tried the `CORE_DEBUG_LEVEL=2` setting and it does not crash anymore.
On the other hand `server.config.stack_size = 8192;` has no effect and will still cause a core panic.

T94T, 2024-04-26:
related to: #74

hoeken, 2024-08-10:
I think this is fixed, please re-open if necessary.

closed 2024-08-10 by hoeken