hoeken/PsychicHttp #52: Change authentication values on endpoint at runtime?

zekageri, 2024-01-04:
I have an endpoint like this

```cpp
server.serveStatic("/", db.sysFS, "/admin")->setCacheControl(ONE_MONTH_CACHE_TIME)
        .setAuthentication(config.adminUsername, config.adminPassword);
```

The password is changing dinamically at runtime.
Can i pass it as a reference to the char array or should i create an endpoint for it like this?

```cpp
server.on("/Pages/AdminPage/index.html", HTTP_GET, [](PsychicRequest *request){
    if( !request->authenticate(config.adminUsername, config.adminPassword) ){
        return request->requestAuthentication(BASIC_AUTH, "Server", "Admin page needs authentication!");
    }
    PsychicFileResponse response(request,db.sysFS,ADMIN_PATH,"text/html");
    response.addHeader("Content-Encoding", "gzip");
    response.addHeader("Cache-Control", ONE_MONTH_CACHE_TIME);
    return response.send();
});
```

hoeken, 2024-01-07:
version 2 will let you change that at runtime

hoeken, 2024-08-10:
Umm, I'm not sure what I was writing there, but I looked in the code now and you should just be able to call handler->setAuthentication() again to update those.  You'll need to keep the pointer around if you want to do that.  Also, I'm not 100% sure if its thread safe to do that within a callback.  Give it a shot and see if it works.

hoeken, 2024-08-11:
I just tested this in v2-dev, and you can update the authentication at runtime:

```cpp
// how to do basic auth
//  curl -i --user admin:admin http://psychic.local/auth-basic
server.on("/auth-basic", HTTP_GET, [](PsychicRequest* request) {
  request->endpoint()->setAuthentication(app_user, "pass");
  return request->reply("Auth Basic Success!"); 
})->setAuthentication(app_user, app_pass, BASIC_AUTH);
```

Note: you will need to call it on every endpoint / every handler that uses auth as they store a copy of the username/password internally. You should be able to do this no problem by keeping references to your handlers / endpoints around if you need that.

closed 2024-08-11 by hoeken