hoeken/PsychicHttp #53: HTTPS doesnt cache?

zekageri, 2024-01-04:
When i switch to https ( painfully slow ) and wait for the files to load and refresh the page i see in the console that the files are not cached. Is this intentional?

Chris--A, 2024-01-05:
This may be obvious, but it caught me out... I had the dev tools open and "disable cache" was selected. Realized my issue after a coffee.

If not, can you confirm the cache headers are received in the browser?

I wrote my own handler for a few reasons:

- I need to lock the SD card.
- development for the web pages needs to show changes (when updates uploaded). I use `must-revalidate`.
- `PsychicStaticFileHandler` sets the etag as the file size. This will hurt when a change is made and size is equal. I use file time.

Here is my default handler:

```C++
server.on("/*", HTTP_GET, [](PsychicRequest *request) {
  esp_err_t result = ESP_OK;
  String path = "/www";

  path += (request->url() == "/") ? "/index.html" : request->url();
  Serial.printf("HTTP GET [%s] -> [%s] ", request->url(), path);

  //Lock SD card
  std::lock_guard<std::mutex> lck(SDMutex);
  File file = SD.open(path);

  if(file){
    String etagStr(file.getLastWrite(), 10);

    if(request->header("If-None-Match").equals(etagStr)){
      result = request->reply(304);
    }else{
      PsychicFileResponse response(request, file, path);
      response.addHeader("Cache-Control", "must-revalidate");
      response.addHeader("ETag", etagStr.c_str());
      result = response.send();
    }
    file.close();
  }else{
    result = request->reply(404);
  }
  return result;
});  
```

zekageri, 2024-01-05:
Oh the file time for etag is 200 IQ. I will check the headers!

zekageri, 2024-01-05:
Little curious about this endpoind.

What if the client does not want to request a file?
It will throw a 404 for any endpoint which is not a file.

Where do you unlock the SD guard? In a separate close handler?

Why don't you use the static file filter like : 

`server.serveStatic("/", SD, "/").setFilter(YOUR_FILTER_FN);`

Do you use HTTPS too?

zekageri, 2024-01-05:
It does have cache control header

[image]

zekageri, 2024-01-05:
It caches a png logo for example but not the bigger scripts.

zekageri, 2024-01-05:
Here is a logo header

[image]

It is loaded from cache after second refresh but not the login.js

zekageri, 2024-01-05:
HTTPS also returns 404 for requests that are normally served fine with HTTP.

zekageri, 2024-01-05:
[image]

zekageri, 2024-01-05:
It has etag and cache control header

Chris--A, 2024-01-05:
> What if the client does not want to request a file? It will throw a 404 for any endpoint which is not a file.

I have found that it searches top to bottom for a compatible endpoint. For example, my `/api/*` route below will match before the `/*` route due to the order that I add them:

```C++
server.on("/api/*", [](PsychicRequest *request) {...}

server.on("/testRoute", [](PsychicRequest *request) {...}

server.on("/*", HTTP_GET, [](PsychicRequest *request) {...}
```

> Where do you unlock the SD guard? In a separate close handler?

`std::lock_guard<std::mutex>` works with the RAII principle, when the variable goes out of scope (destructor called) the mutex is released.

> Why don't you use the static file filter like :
> `server.serveStatic("/", SD, "/").setFilter(YOUR_FILTER_FN);`

I might be wrong, but the filter appears to be a boolean check to see if the route should handle the request. I'm not sure if it has the ability to add customizations to the handler itself. I don't think I'm understanding what I'd use the filter for in your question.

> Do you use HTTPS too?

I only messed with it in a test app, I haven't got it in my project yet, I've been lazy but will give it a go tonight.

Chris--A, 2024-01-05:
> ![image](https://private-user-images.githubusercontent.com/37899881/294442446-3ade5cb3-aaed-4b01-9795-14b00fead294.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ0NDY1MTAsIm5iZiI6MTcwNDQ0NjIxMCwicGF0aCI6Ii8zNzg5OTg4MS8yOTQ0NDI0NDYtM2FkZTVjYjMtYWFlZC00YjAxLTk3OTUtMTRiMDBmZWFkMjk0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA1VDA5MTY1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIwMDYxODc0YTliMTYwYjc0MzhmZDYyOTJlZjI4NGMzZTEzOGFhYzU3ZWYwZjU0M2I4YjEyMmEzYjkzMTkwOTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.EOXZBZpezOpZmPutmmx0wpConERiWpnG2rU5AoLVvU4)

Can you see the `If-None-Match` header being sent with the etag to the server in the requests headers section?

zekageri, 2024-01-05:
> std::lock_guard<std::mutex> works with the RAII principle, when the variable goes out of scope (destructor called) the mutex is released.
Ahh i see. Really cool.

> Can you see the If-None-Match header being sent with the etag to the server in the requests headers section?
Not checked yet.

Will try with this `/*` middleware soon.
Some files are cached in https some are don't. Interesting.

zekageri, 2024-01-05:
I created this endpoint quickly to see the header values

```cpp
server.on("/*", HTTP_GET, [](PsychicRequest *request) {
    Serial.printf("If-None-Match header: %s\n", request->header("If-None-Match").c_str());
    return ESP_OK;
});
```
It prints nothing: 

`If-None-Match header: `

Chris--A, 2024-01-05:
> I created this endpoint quickly to see the header values
> 
> ```c++
> server.on("/*", HTTP_GET, [](PsychicRequest *request) {
>     Serial.printf("If-None-Match header: %s\n", request->header("If-None-Match").c_str());
>     return ESP_OK;
> });
> ```
> 
> It prints nothing:
> 
> `If-None-Match header: `

Hey, sorry, I meant from the browser, it looks like the browser isn't sending it.

[image]

I'm just setting up the HTTPS stuff now, will let you know if it works for me.

Chris--A, 2024-01-05:
I'll have to look at the HTTPS tomorrow. It just kind of explodes.. I get one file of data back, and it throws an error. After that the connection is severed. 

```
E (63765) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (63770) esp_https_server: esp_tls_create_server_session failed
```

zekageri, 2024-01-05:
It does this for me too.
I had to increase the stack size and lower the `max_open_sockets`

hoeken, 2024-08-10:
Closing these old issues, please re-open if needed.

closed 2024-08-10 by hoeken