hoeken/PsychicHttp #54: Multiple serveStatic  Handlers

kevinkerker, 2024-01-04:
@hoeken 

``` 
//* Cache static files for 1 day
  httpServer.serveStatic("/static/", SD, "/www/static/").setCacheControl("max-age=86.400");

  httpServer.serveStatic("/", SD, "/www/").setDefaultFile("index.html");
  httpServer.serveStatic("/config/", SD, (std::string(CONFIG_LOCATION) + "/").c_str());
  httpServer.serveStatic("/log-small", SD, "/log-small.csv");
```

If i define multiple serveStatic handler only the last one is active.
Are there any restrictions for serveStatic handlers?

T94T, 2024-01-04:
Try to call the `listen()` method before all `serveStatic()` instructions.

kevinkerker, 2024-01-04:
Thank you all. It was my bad i pulled the last release 1.0.1 via platformIO. But there are huge changes to the code. Now i pull it with 

# or using GIT Url (the latest development version)
lib_deps = https://github.com/hoeken/PsychicHttp

closed 2024-01-04 by kevinkerker

hoeken, 2024-01-04:
I will try to push a v1.1 this week but for now it's best to use the GitHub
code.