hoeken/PsychicHttp #55: request->send_P method

orkungedik, 2024-01-05:
Hi,

I am trying to convert my project from ESPAsyncWebServer to PsychicHttp for SSL support reason. By using ESPAsyncWebServer I can send response to the client by `request->send_P(200, "text/html", page_html, page_html_processor);` . So I can provide page_html_processor parameter in send_P method in order to replace variables defined in static html. But I can't find a matching method with send_P in PsychicHttp library. I found **reply** method but it hasn't a callback function parameter like AwsTemplateProcessor callback = nullptr.

Is there any way to use html processor paramater in PsychicHttp?

Thank you,

Orkun Gedik

orkungedik, 2024-01-05:
Additionally, implementations are missing, below;

```AsyncCallbackWebHandler& on(const char* uri, WebRequestMethodComposite method, ArRequestHandlerFunction onRequest, ArUploadHandlerFunction onUpload);
    AsyncCallbackWebHandler& on(const char* uri, WebRequestMethodComposite method, ArRequestHandlerFunction onRequest, ArUploadHandlerFunction onUpload, ArBodyHandlerFunction onBody);
```

Chris--A, 2024-01-06:
I'm working on a solution for templating at the moment. Might have a candidate tonight.

hoeken, 2024-08-10:
Template processor from @Chris--A is merged and live in v1.2

closed 2024-08-10 by hoeken