hoeken/PsychicHttp #57: Missing AsyncCallbackWebHandler& on implementations

orkungedik, 2024-01-07:
Hi,

Multipart body implementations are does not exist.

AsyncCallbackWebHandler& on(const char* uri, WebRequestMethodComposite method, ArRequestHandlerFunction onRequest, ArUploadHandlerFunction onUpload);

AsyncCallbackWebHandler& on(const char* uri, WebRequestMethodComposite method, ArRequestHandlerFunction onRequest, ArUploadHandlerFunction onUpload, ArBodyHandlerFunction onBody);

Library needs multipart support not only on upload and download files, but also on large HTML pages.

Thank you for your support.

Orkun Gedk

hoeken, 2024-08-10:
Multipart support has been added at some point, here is the code from the platformio example.

```cpp
    // a little bit more complicated multipart form
    PsychicUploadHandler* multipartHandler = new PsychicUploadHandler();
    multipartHandler->onUpload([](PsychicRequest* request, const String& filename, uint64_t index, uint8_t* data, size_t len, bool last)
      {
      File file;
      String path = "/www/" + filename;

      //some progress over serial.
      Serial.printf("Writing %d bytes to: %s\n", (int)len, path.c_str());
      if (last)
        Serial.printf("%s is finished. Total bytes: %d\n", path.c_str(), (int)index+(int)len);

      //our first call?
      if (!index)
        file = LittleFS.open(path, FILE_WRITE);
      else
        file = LittleFS.open(path, FILE_APPEND);
      
      if(!file) {
        Serial.println("Failed to open file");
        return ESP_FAIL;
      }

      if(!file.write(data, len)) {
        Serial.println("Write failed");
        return ESP_FAIL;
      }

      return ESP_OK; });

    // gets called after upload has been handled
    multipartHandler->onRequest([](PsychicRequest* request)
      {
      if (request->hasParam("file_upload"))
      {
        PsychicWebParameter *file = request->getParam("file_upload");

        String url = "/" + file->value();
        String output;

        output += "<a href=\"" + url + "\">" + url + "</a><br/>\n";
        output += "Bytes: " + String(file->size()) + "<br/>\n";
        output += "Param 1: " + request->getParam("param1")->value() + "<br/>\n";
        output += "Param 2: " + request->getParam("param2")->value() + "<br/>\n";
        
        return request->reply(output.c_str());
      }
      else
        return request->reply("No upload."); });

    // wildcard basic file upload - POST to /upload/filename.ext
    //  use http://psychic.local/ to test
    server.on("/multipart", HTTP_POST, multipartHandler);
```

If this does not work for you, please re-open an issue with the specific problem.

closed 2024-08-10 by hoeken

mathieucarbou, 2024-08-10:
@hoeken : this is linked to #118 I think.
`loadParams()` should also multipart content because body parameters can also be sent in a multipart content.