hoeken/PsychicHttp #58: Server fails to process requests around 6kb with POST.

zekageri, 2024-01-08:
I have made a test sketch forked from the PIO example of this repo which you can find here: https://github.com/zekageri/Psychic_Upload_Test

**In the test sketch:**
- I have added a `textarea` for a json string and a button to POST it with javascript fetch API.
- I have added an endpoint to process this json data

**Here is a test json which i have tried but any json will be failed at around 6k**
https://github.com/hoeken/PsychicHttp/files/13827213/test.json

Here are the js and c++ code of the test:

```cpp
server.on("/testSave", HTTP_POST, [](PsychicRequest *request, JsonVariant &json) {
    // MIN_LENGTH  is a random low number because it can split out some strings which is valid JSON
    static const int MIN_LENGTH = 20;
    Serial.println();
    int len = serializeJsonPretty(json, Serial);
    Serial.println();
    return request->reply(len>MIN_LENGTH?200:500, "text/plain", len>MIN_LENGTH?"OK":"Failed");
});
```

**Javascript**

```js
// Get the button and the textarea
const testJsonButton = document.querySelector("#testJsonBtn");
const testJsonField = document.querySelector("#testJson");

// Add an event listener to the button
testJsonButton.addEventListener('click', async function(){
    // Can't post without data inside the textarea
    if( testJsonField.value.length == 0 ){
        alert("No test json provided");
        return;
    }
    // Parse the json and log it raw and parsed
    // So we have an idea what should be seen on the backend
    const jsonText = JSON.parse( testJsonField.value );
    console.log("Raw json: ", testJsonField.value);
    console.log("Parsed json: ", jsonText);
  
    // Send the request with fetch API to the test endpoint
    const response = await fetch("/testSave",{
        method: "POST",
        headers: {
            "Content-Type": "application/json"
        },
        body: JSON.stringify(jsonText)
    });
    // Log out the response
    alert( response.ok?"Success":"Failed" );
    testJsonField.value = "";
}, false);
```

**On the backend i set the sizes on the server like this**

```cpp
// MAX_UPLOAD_SIZE and MAX_REQUEST_BODY_SIZE are coming from the lib
server.maxUploadSize = MAX_UPLOAD_SIZE;
server.maxRequestBodySize = MAX_REQUEST_BODY_SIZE;
```

So far it doesn't matter if i set it to a higher size or does not set it at all.
The request will be incomplete at around 6kb even if the `MAX_REQUEST_BODY_SIZE` is 16k at min. 
I tried logging the raw `request->body()` before my handler is called ( in the lib's code ) but the result is the same.
I have upgraded to `ArduinoJson V7` which will allocate memory dinamically but it didn't help since the problem is not with the parsing of the json string but it is further back.

I even tried to `POST` a file with `CURL` in windows but the result is the same.
Here is the `CURL` output

```CURL
curl -i -X POST -H "Content-Type: application/json" --data-binary "@D:\test.json" http://hsh.local/upload/?path=/tempModule.json&partition=user
HTTP/1.1 200 OK
Content-Type: text/plain
Content-Length: 15
Server: HsH Server

Upload success!
```

The `CURL` is tested with the uploadhandler but the result is the same.
I have validated my test json with more than one online **JSON validator**

[image]

The result is always valid. ( which i can also see in the front end since the JSON parse is not failing )

As you can see i have `136640 byte` heap free and `3118471 byte` ext ram free before the curl request.
And almost the same after.

Here is a log of my heap and external ram before and after CURL. 
As you can see it logs out "name""name" from the json POST request.

```cpp
[SYSTEM] - [HEAP][1]
Free:   136640
Min:    136640
Max:    136640
Contig: 90100
Diff:   32

[SYSTEM] - [EXT][1]
Free:   3118471
Min:    3118471
Max:    3118471
Contig: 3014644
Diff:   1552

[Server] - Connection #54 connected from 192.168.10.114
"name""name"

[SYSTEM] - [HEAP][1]
Free:   136672
Min:    136672
Max:    136672
Contig: 90100
Diff:   32

[SYSTEM] - [EXT][1]
Free:   3120023
Min:    3120023
Max:    3120023
Contig: 3014644
Diff:   1552
```

**To summarize:**

- I have enough internal and external ram
- The Json message fits to the default body size
- The body text is malformed even in the `handleRequest` method of the `PsychicWebHandler` class in the lib's code
- I have confirmed with the test sketch that it is not working properly ( which i have linked above )
- Doesn't matter if I `POST` the json from a file with `CURL` or if I `POST` the raw text with `fetch` from the client side.
- It also doesnt matter if I set the content type to `application/json` or regular text. It will be text anyway.
- Confirmed that it is not an ArduinoJson issue because the raw body is malformed before the lib can parse it

I'm totally lost on this issue.
Otherwise (almost) everything works flawlessly and the server is stable and fast and responsive.

zekageri, 2024-01-08:
The funny thing is that sometimes it can process the request.

proddy, 2024-01-08:
I said I would help but work caught up on me, and I'm out of office this week travelling. I'll see if I play around later this week

zekageri, 2024-01-08:
That would be wonderfull. Thank you.

zekageri, 2024-01-08:
Would be good if @hoeken could take a look at this too!

zekageri, 2024-01-08:
So i have refactored the `loadBody()` method and it seems to work fine now.

```cpp
esp_err_t PsychicRequest::loadBody()
{
  esp_err_t err = ESP_OK;

  this->_body = String();

  size_t remaining = this->_req->content_len;
  size_t actuallyReceived = 0;
  char *buf = (char *)malloc(remaining + 1);
  if (buf == NULL) {
    ESP_LOGE(PH_TAG, "Failed to allocate memory for body");
    return ESP_FAIL;
  }

  while (remaining > 0) {
    int received = httpd_req_recv(this->_req, buf + actuallyReceived, remaining);

    if (received == HTTPD_SOCK_ERR_TIMEOUT) {
      continue;
    }
    else if (received == HTTPD_SOCK_ERR_FAIL) {
      ESP_LOGE(PH_TAG, "Failed to receive data.");
      err = ESP_FAIL;
      break;
    }

    remaining -= received;
    actuallyReceived += received;
  }

  buf[actuallyReceived] = '\0';
  this->_body = String(buf);
  free(buf);
  return err;
}
```

zekageri, 2024-01-08:
More testing required

hoeken, 2024-01-08:
@zekageri I'm so sorry this has been such a problem for you.  I will try my best to make time to test this and get any changes pulled.  This next month is a bit crazy for me though.  I live on a sailboat and we are planning on sailing from Panama to Miami next week and my engine just started giving me comms errors and shutting down randomly.  The sail itself will take 1-2 weeks and then the boat show prep is going to be pretty intense too I imagine.

I will carve out some hours in the evening though, got a few other PsychicHttp PRs to accept.

zekageri, 2024-01-09:
Yeah, i can confirm that this was the problem.
I have refactored the `loadBody(`) method in the following ways

```cpp
esp_err_t PsychicRequest::loadBody()
{
  esp_err_t err = ESP_OK;

  this->_body = String();

  size_t remaining = this->_req->content_len;
  size_t actuallyReceived = 0;
  char *buf = (char *)malloc(remaining + 1);
  if (buf == NULL) {
    ESP_LOGE(PH_TAG, "Failed to allocate memory for body");
    return ESP_FAIL;
  }

  while (remaining > 0) {
    int received = httpd_req_recv(this->_req, buf + actuallyReceived, remaining);

    if (received == HTTPD_SOCK_ERR_TIMEOUT) {
      continue;
    }
    else if (received == HTTPD_SOCK_ERR_FAIL) {
      ESP_LOGE(PH_TAG, "Failed to receive data.");
      err = ESP_FAIL;
      break;
    }

    remaining -= received;
    actuallyReceived += received;
  }

  buf[actuallyReceived] = '\0';
  this->_body = String(buf);
  free(buf);
  return err;
}
```

1. Memory Allocation Check:
  In the modified version, a check for the success of memory allocation is added after malloc. If the memory allocation fails, an error message is logged, and the function returns ESP_FAIL.

```cpp
char *buf = (char *)malloc(remaining + 1);
if (buf == NULL) {
  ESP_LOGE(PH_TAG, "Failed to allocate memory for body");
  return ESP_FAIL;
}
```

2. Tracking Received Data:
  The variable actuallyReceived is introduced to keep track of the amount of data received in each iteration of the while loop. This variable is then used to update the buffer pointer (buf + actuallyReceived) in the httpd_req_recv function call.

```cpp
size_t actuallyReceived = 0;
int received = httpd_req_recv(this->_req, buf + actuallyReceived, remaining);
```
This change ensures that the received data is appended correctly to the buffer.

3. Logging Errors:
  Additional error logging is added to provide more information about failure conditions.

```cpp
if (received == HTTPD_SOCK_ERR_TIMEOUT) {
  continue;
}
else if (received == HTTPD_SOCK_ERR_FAIL) {
  ESP_LOGE(PH_TAG, "Failed to receive data.");
  err = ESP_FAIL;
  break;
}
```

4. Cleanup:
  The cleanup phase is improved by updating the null termination step to use actuallyReceived instead of this->_req->content_len. This ensures that only the actually received data is null-terminated.

```cpp
buf[actuallyReceived] = '\0';
```

Overall, these changes enhance the robustness and readability of the code, handling memory allocation failures more gracefully, providing additional error information, and ensuring correct tracking of received data.

closed 2024-01-09 by zekageri

zekageri, 2024-01-09:
Nah i think it was because of incorrect use of the httpd_req_recv() IDF function.

it was used like this

`int received = httpd_req_recv(this->_req, buf, this->_req->content_len);`

instead of this

`int received = httpd_req_recv(this->_req, buf + actuallyReceived, remaining);`

It overwrited the buffer's content before. But now if i always pass the start pointer of the next chunk it works.

proddy, 2024-01-09:
that's a great find - not sure I would have discovered that. kudos!

zekageri, 2024-01-09:
I have a couple of days in that. :D But now i have found that something is also wrong with the `PsychicUploadHandler`.
It produces the same symptoms. Can't upload really large files... Regular handler works with the fix hovewer. Will investigate...

zekageri, 2024-01-09:
Oh yeah, as I'm looking into it closer, it has the same pattern as the loadBody()

```cpp
if ((received = httpd_req_recv(request->request(), buf, min(remaining, FILE_CHUNK_SIZE))) <= 0)
{
  /* Retry if timeout occurred */
  if (received == HTTPD_SOCK_ERR_TIMEOUT)
    continue;
  //bail if we got an error
  else if (received == HTTPD_SOCK_ERR_FAIL)
  {
    ESP_LOGE(PH_TAG, "Socket error");
    err = ESP_FAIL;
    break;
  }
}
```

The `_basicUploadHandler` also uses the `httpd_req_recv`

zekageri, 2024-01-09:
`_basicUploadHandler` also has no check if malloc failed or not but at least it has an `index` variable which tracks the bytes

proddy, 2024-01-09:
> Can't upload really large files...

that's interesting. How large? I've been uploading firmware .bin files > 2MB with no issues.

zekageri, 2024-01-10:
Can you please give me an example on an uploadhandler cb? For me it even corrupts 20kb files

zekageri, 2024-01-10:
This is how i use it

```cpp
void HsHServer::uploadHandlerInit(){
    uploadHandler.onUpload([this](PsychicRequest *request, const String& filename, uint64_t index, uint8_t *data, size_t len, bool last) {
        if (!request->hasParam("path")) {
            uploadStatus = false;
            return request->reply(500,"text/plain","No path provided");
        }
        if (!index){
            uploadStatus = true;
            getPartitionObject(request);
            tempFile = tempPartition.open(request->getParam("path")->value().c_str(), FILE_WRITE);
        }else{
            tempFile = tempPartition.open(request->getParam("path")->value().c_str(), FILE_APPEND);
        }

        if(!tempFile || !tempFile.write(data, len)){ uploadStatus = false; return ESP_FAIL; }
        return ESP_OK;
    });
    uploadHandler.onRequest([this](PsychicRequest *request){
        return request->reply(uploadStatus?200:500,"text/plain",uploadStatus?"Upload success!":"Upload failed!");
    });
}

void HsHServer::getPartitionObject(PsychicRequest *request){
    tempFile.close();
    if (request->hasParam("partitionLabel") && request->getParam("partitionLabel")->value() != "user") {
        tempPartition = db.sysFS;
        return;
    }
    tempPartition = db.userFS;
}
```

Basically it opens a file on a specific paftition and just writes the data to it.

zekageri, 2024-01-10:
I'm testing with this sample json file
[testUpload.json](https://github.com/hoeken/PsychicHttp/files/13886337/testUpload.json)

**With this CURL command**

`curl -i -X POST -H "Content-Type: application/json" --data-binary "@D:\testUpload.json" http://hsh.local/upload/?path="/testUpload.json"`

And the result is a corrupted json in the fs

```json
[
    {
        "quiz": {
            "sport": {
                "q1": {
                    "question": "Which one is correct team name in NBA?",
                    "options": [
                        "New York Bulls",
                        "Los Angeles Kings",
                        "Golden State Warriros",
                        "Huston Rocket"
                    ],
                    "answer": "Huston Rocket"
                }
            },
            "maths": {
                "q1": {
                    "question": "5 + 7 = ?",
                    "options": [
                        "10",
                        "11",
                        "12",
                        "13"
                    ],
                    "answer": "12"
                },
                "q2": {
                    "question": "12 - 8 = ?",
                    "options": [
                        "1",
                        "2",
                        "3",
                        "4"
                    ],
                    "answer": "4"
                }
            }
        }
    },
    {
        "quiz": {
            "sport": {
                "q1": {
                    "question": "Which one is correct team name in NBA?",
                    "options": [
                        "New York Bulls",
                        "Los Angeles Kings",
                        "Golden State Warriros",
                        "Huston Rocket"
                    ],
                    "answer": "Huston Rocket"
                }
            },
            "maths": {
                "q1": {
                    "question": "5 + 7 = ?",
                    "options": [
                        "10",
                        "11",
                        "12",
[... 183 lines cut ...]
                "q1": {
                    "question": "5 + 7 = ?",
                    "options": [
                        "10",
                        "11",
                        "12",
                        "13"
                    ],
                    "answer": "12"
                },
                "q2": {
                    "question": "12 - 8 = ?",
                    "options": [
                        "1",
                        "2",
                        "3",
                        "4"
                    ],
                    "answer": "4"
                }
            }
        }
    }
]
```

zekageri, 2024-01-10:
Yeah, it fixed the problem. Strange.