hoeken/PsychicHttp #59: depends on Arduino?

marchingband, 2024-01-08:
I am looking for a server to use in an esp-idf project ... I don't see any obvious dependency on Arduino ... what do you think?
thanks!

Chris--A, 2024-01-08:
Hi @marchingband 

From the readme:

> PsychicHttp is a webserver library for ESP32 + **Arduino framework**

You don't need to use the IDE, but it does make use of core Arduino components.

marchingband, 2024-01-08:
ok thanks!

closed 2024-01-08 by marchingband

marchingband, 2024-01-08:
If it compiled under esp-idf I would be happy to contribute back any new features I developed.
It looks like there is not a great esp-idf server, so I will likely look to your code for some inspiration.
If I come up with anything PR worthy I will keep in touch, but unfortunately I don't have a ton of time ATM to contribute.
I will let you know if that changes! Thanks for the invitation.

hoeken, 2024-01-08:
PsychicHttp uses the esp-idf http server under the hood which is very stable... it is just very basic and makes you write a lot of extra code to use it.

marchingband, 2024-01-08:
Yes. That's why I am on this hunt for a good lib to help, but I can't use Arduino.