hoeken/PsychicHttp #62: HTTPS is unusable.

zekageri, 2024-01-11:
If i use the following config

```cpp
server.ssl_config.httpd.max_uri_handlers = 100;
server.ssl_config.httpd.stack_size = 32000;
server.ssl_config.httpd.max_open_sockets = 8;
```

The HTTPS is just not working. The web page fails to load and i get a bunch of errors like these

```cpp
[SYSTEM] - heap_caps_calloc was called but failed to allocate 4 bytes with 0x804 capabilities.
E (151500) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x4310
E (151500) esp_https_server: esp_tls_create_server_session failed   
[SYSTEM] - heap_caps_calloc was called but failed to allocate 4 bytes with 0x804 capabilities.
E (160968) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x4310
E (160968) esp_https_server: esp_tls_create_server_session failed   
[SYSTEM] - heap_caps_calloc was called but failed to allocate 4 bytes with 0x804 capabilities.
E (163687) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x4310
E (163688) esp_https_server: esp_tls_create_server_session failed   
[SYSTEM] - heap_caps_calloc was called but failed to allocate 4 bytes with 0x804 capabilities.
E (179263) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x4310
E (179264) esp_https_server: esp_tls_create_server_session failed   
[SYSTEM] - heap_caps_calloc was called but failed to allocate 4 bytes with 0x804 capabilities.
E (182024) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x4310
E (182025) esp_https_server: esp_tls_create_server_session failed 
```

If i set the `server.ssl_config.httpd.max_open_sockets` to **1** it can load some files but not all of them ( max 4-5 ) and the requests results in error.

Are there some settings i have missed?

hoeken, 2024-08-10:
@philippebourcier HTTPS on the esp32 is a big challenge that mostly comes down to memory issues.

As for http + https working together, you have to create a 2nd server listening on http.  In the example, it is like this:

```
// this creates a 2nd server listening on port 80 and redirects all requests HTTPS
PsychicHttpServer* redirectServer = new PsychicHttpServer();
redirectServer->config.ctrl_port = 20424; // just a random port different from the default one
redirectServer->onNotFound([](PsychicRequest* request) {
  String url = "https://" + request->host() + request->url();
  return request->redirect(url.c_str());
});
```

hoeken, 2024-08-11:
@zekageri I think this might actually be a memory issue.  Setting that max_uri_handlers to 100 is going to pre-allocate a ton of memory.  if you try the v2-dev branch, we use the new meta endpoints setup where you no longer have to pre-declare those uri handlers limit.

hoeken, 2024-08-11:
HTTPS is slow and sucks, but it does work.  Closing this unless there's a more specific problem we can troubleshoot.

closed 2024-08-11 by hoeken