hoeken/PsychicHttp #65: Core Panic with WebSocketServer and CORE_DEBUG_LEVEL=5

theelims, 2024-01-14:
Hi, first congratulations to this library. I hope this really becomes a popular alternative to the unstable ESPAsyncWebServer world. This is what everyone has been waiting for. 

This relates to issue #20.

When `httpd_ws_get_fd_info(request->server(), httpd_req_to_sockfd(request->request())` is called the ESP32-S3 crashes. 

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4204c348  PS      : 0x00060730  A0      : 0x8204d604  A1      : 0x3fca2680
A2      : 0x3fc9ab4c  A3      : 0x00000039  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x00004629  A7      : 0x420d25e0  A8      : 0x000003e8  A9      : 0x00000069
A10     : 0x1125427d  A11     : 0x000fffff  A12     : 0x3c0e1636  A13     : 0x00000062  
A14     : 0x3c0e18a5  A15     : 0x3c0e1648  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x000003e8  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffe  

Backtrace: 0x4204c345:0x3fca2680 0x4204d601:0x3fca26d0 0x420103c1:0x3fca26f0 0x42010bd1:0x3fca27c0 0x4204be45:0x3fca2840 0x4204c66e:0x3fca28e0 0x4204b470:0x3fca2900 0x420d3c06:0x3fca2920 0x4204b584:0x3fca2940

  #0  0x4204c345:0x3fca2680 in httpd_sess_get at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_http_server/src/httpd_sess.c:173
  #1  0x4204d601:0x3fca26d0 in httpd_ws_get_fd_info at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_http_server/src/httpd_ws.c:521
  #2  0x420103c1:0x3fca26f0 in PsychicWebSocketHandler::handleRequest(PsychicRequest*) at .pio/libdeps/esp32-s3-devkitc-1/PsychicHttp/src/PsychicWebSocket.cpp:214 (discriminator 7)  
  #3  0x42010bd1:0x3fca27c0 in PsychicEndpoint::requestCallback(httpd_req*) at .pio/libdeps/esp32-s3-devkitc-1/PsychicHttp/src/PsychicEndpoint.cpp:72
  #4  0x4204be45:0x3fca2840 in httpd_req_new at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_http_server/src/httpd_parse.c:776
  #5  0x4204c66e:0x3fca28e0 in httpd_sess_process at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_http_server/src/httpd_sess.c:419 (discriminator 15)
  #6  0x4204b470:0x3fca2900 in httpd_process_session at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_http_server/src/httpd_main.c:178
      (inlined by) httpd_process_session at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_http_server/src/httpd_main.c:163
  #7  0x420d3c06:0x3fca2920 in httpd_sess_enum at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_http_server/src/httpd_sess.c:50 (discriminator 1)
  #8  0x4204b584:0x3fca2940 in httpd_server at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_http_server/src/httpd_main.c:228
      (inlined by) httpd_thread at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_http_server/src/httpd_main.c:250

ELF file SHA256: 9b0f12f59e6c4cfb
```

Commenting out the line `ESP_LOGI(PH_TAG, "ws_handler: httpd_handle_t=%p, sockfd=%d, client_info:%d", request->server(),  httpd_req_to_sockfd(request->request()), httpd_ws_get_fd_info(request->server(), httpd_req_to_sockfd(request->request())));` or changing to `-D CORE_DEBUG_LEVEL=2` solves the issue.

pletch, 2024-02-11:
Suspect the default stack size for the psychichttp httpd server task is too small at only 4K with this ESP_LOGI instruction.   

Increasing the psychic http stack size by calling httpd_config option config.stack_size = 6144 prior to server.listen seems to address this problem.

See [https://www.esp32.com/viewtopic.php?t=9834](https://www.esp32.com/viewtopic.php?t=9834) and [https://www.esp32.com/viewtopic.php?t=10443](https://www.esp32.com/viewtopic.php?t=10443).

#74 is essentially the same issue.

mhaberler, 2024-08-10:
this is now fixed in https://github.com/hoeken/PsychicHttp/issues/74 and can be closed

closed 2024-08-10 by hoeken