hoeken/PsychicHttp #69: Upload never finish the first File request

johnnytolengo, 2024-01-28:
setup:  esp32, platformio espressif32@6.5.0

I want to upload multiple files, and one call is done after the previous is finished (200) like this:
```
http://ipadress_here/api/savefile?filename=/image/agv3b05c.png
http://ipadress_here/api/savefile?filename=/image/37u3b45c.png
http://ipadress_here/api/savefile?filename=/image/3323b32c.png
etc..
```
my upload handler:
```
PsychicUploadHandler *uploadHandler2 = new PsychicUploadHandler();
	uploadHandler2->onUpload([this](PsychicRequest *request, const String& filename, uint64_t index, uint8_t *data, size_t len, bool last) {
    //if (request->hasParam("ord")) {
				DK_LOGD(MODULE_NAME,"[UploadHandler] - DEBUG 1 ");
        //return request->reply(200);
    //}
    return ESP_OK;
	});
	uploadHandler2->onRequest([this](PsychicRequest *request){
		DK_LOGD(MODULE_NAME,"[UploadHandler] - DEBUG 2 ");
			return request->reply(200);
	});

	_asyc_webserver->on("/api/savefile", HTTP_POST, uploadHandler2);
```
The "[UploadHandler] - DEBUG 1 " is printed only once in the Serial Monitor and the "[UploadHandler] - DEBUG 2 " is printed on each call, 
that means that the handler is not recognize that the next call is another file.

I tried to use the "Content-Disposition header" option and  the filename paramater (example above), none of these options prints the "[UploadHandler] - DEBUG 1 " on each file detected.

I don't know what I am doing wrong or what I misunderstood. 

Anyone can help me with it?

johnnytolengo, 2024-01-29:
I was all day trying to find out where is the problem but I could't see where is the problem.
What I see is that the callback (onUpload) [_handleUploadByte](https://github.com/hoeken/PsychicHttp/blob/master/src/PsychicUploadHandler.cpp#L205)  is never called

anyone can help me?

johnnytolengo, 2024-01-30:
@hoeken @ could I please you your help to find out why the uploadCallback is not called?  I followed all the instructions but  the uploadCallback is called only on the first file POST, after that is not called anymore.
I spent about 2 days looking and debugging :(
What I am doing wrong?

On first POST http://localhost:3000/api/save?filename=config.tmp  the callback is called (onUpload):
[image]

Next POST http://localhost:3000/api/save?filename=logo512.png callback is not called anymore only the onRequest is called:
[image]

Thank you!

zekageri, 2024-01-30:
How do you call the endpoint from client side?

johnnytolengo, 2024-01-30:
I call the endpoint with Axios (from chrome browser), the strange thing is that onUpload() callback is called only the first time.

zekageri, 2024-01-30:
You must wait for the first one to finish. The esp cant process two concurrent requests.

johnnytolengo, 2024-01-30:
That's what I am doing, I send the first file when I got 200 then I post next one, and so on.

johnnytolengo, 2024-01-30:
This is my code:
```
PsychicUploadHandler *uploadHandler = new PsychicUploadHandler();

uploadHandler->onUpload([this](PsychicRequest *request, const String& filename, uint64_t index, uint8_t *data, size_t len, bool last) {
String real_filename = filename;
DK_LOGD(MODULE_NAME,"[UploadHandler] - DEBUG 1 ");
	
if (last)
	DK_LOGD(MODULE_NAME,"%s is finished. Total bytes: %d\n", real_filename.c_str(), (int)index+(int)len);

if (!index){
	DK_LOGD(MODULE_NAME,"FIRST RUN!!");
}
else{
	DK_LOGD(MODULE_NAME,"RUN %i" , (int)index);
}

return ESP_OK;
});

uploadHandler->onRequest([this](PsychicRequest *request){
_tempFile.close();
return request->reply(200);
});

_asyc_webserver->on("/api/save", HTTP_POST, uploadHandler);
```

zekageri, 2024-01-30:
Hmmm... Stange.

zekageri, 2024-01-30:
I don't have that problem. I also have file uploads. I even implemented multi file uploads like this on client side.

zekageri, 2024-01-30:
Will check if it is working as intended

zekageri, 2024-01-30:
Oh well, it also doesn't work for me too. Will check on it.
I hope @hoeken see this

johnnytolengo, 2024-01-30:
I am now alone :)  I am still here trying to find out where is the bug.

johnnytolengo, 2024-01-30:
I spent many days trying to fix that issue but without success, it seems that the problem within the "[_parseMultipartPostByte](https://github.com/hoeken/PsychicHttp/blob/master/src/PsychicUploadHandler.cpp#L219)"  probably in the parsing header section or so..  to be honest I don't know well how that workflow runs. 
I think that  @hoeken  can help us to fix that issue very quickly, he knows where is the end process flag,  in order to execute the callback.

johnnytolengo, 2024-01-31:
Hi @zekageri , could you take a look on that bug? I would like at least to know where I should look in order to fix it.
Thanks.

zekageri, 2024-01-31:
Iam currently in the middle of something else but i will definietly investigate it asap because i need that functionality too.

johnnytolengo, 2024-01-31:
great! meanwhile I will be also looking for a solution. please let me know on any update. 
Thank you!

johnnytolengo, 2024-02-01:
@zekageri  I could't find the bug, I feel like rebuilding the PsychicUploadHandler, but I don't know if it worth it maybe I am paranoid.

zekageri, 2024-02-01:
Sad. Will look into it tomorrow for sure!

johnnytolengo, 2024-02-05:
Could someone fix this issue?

hoeken, 2024-02-05:
Hey @johnnytolengo sorry about the long delay here.  I was on a sailing trip and its really hard to code when the whole world is moving around you.  I'm back on land now and I will get this sorted out ASAP.

johnnytolengo, 2024-02-07:
Hi @hoeken, do you know where can be the problem? maybe I can help :)

zekageri, 2024-02-07:
Well, I cant figure it out. :D

johnnytolengo, 2024-02-07:
@zekageri the same here.. I have not clue what is causing that issue, I spent days trying to figuring out without success :(

hoeken, 2024-02-07:
Can you provide some example client code that reproduces the bug?

johnnytolengo, 2024-02-07:
@hoeken  I created a simple testing project you can clone from [HERE](https://github.com/johnnytolengo/postTest.git)
after clone you can run it with yarn: 
yarn install
yarn start

johnnytolengo, 2024-02-08:
@hoeken could you test it? do you need some help?

johnnytolengo, 2024-02-09:
I can't see the light at the end of the tunnel  :)

hoeken, 2024-02-09:
So sorry about the delay here. I was on a boat delivery from Panama to
Florida and this last week was hectic getting it ready for the Miami boat
show. Today is the last day of prep on that and then things calm down for a
while again.

johnnytolengo, 2024-02-09:
No problem, I hope you can figure it out ASAP, I'll really appreciate that. After that I can contribute a few PR that will include some extra features.

johnnytolengo, 2024-02-12:
@zekageri could you take a look on that?

johnnytolengo, 2024-02-16:
Hi @hoeken  do you think that you can fix it within this weekend?  sorry that I'm putting a bit of pressure on it but this feature is for me critical. if you can't just tell me. :) 
Thank you!

johnnytolengo, 2024-02-19:
nobody?  :(

johnnytolengo, 2024-02-19:
@hitecSmartHome it should work but it doesn't, the process does not recognize when a new file is coming, it keeps going on the same loop. :(

johnnytolengo, 2024-02-20:
@hoeken could you tell us short description of that upload process in order to find out where is the issue?

hoeken, 2024-02-20:
@johnnytolengo sorry about not getting to this, life has been crazy lately.  no time for projects.

the upload stuff is all handled by the PsychicUploadHandler.cpp

```PsychicUploadHandler::handleRequest``` is the main hook
```PsychicUploadHandler::_multipartUploadHandler``` is the multipart hook.  it loops over each byte and calls the parser ```_parseMultipartPostByte```

The parser is copied from ESPAsyncWebserver and is a state machine type parser.  It calls ```itemWriteByte``` on each processed byte and if its a file then calls ```_handleUploadByte``` where the callback is finally called.

johnnytolengo, 2024-02-20:
@hoeken I have been looking at all those functions but, I can't find is where is the flag (inside multipart) that stops the previous file loop and tells to the script that it started a new file processing?

hoeken, 2024-02-23:
I merged the PR from gb88, let me know if this fixes it for you.  Thanks @gb88 !

johnnytolengo, 2024-02-23:
Thank you so much @gb88

closed 2024-02-23 by johnnytolengo