hoeken/PsychicHttp #70: Different Version Arduino And Github

hasankorkmazdev, 2024-02-01:
The version in the Arduino library manager seems very incomplete. Please make a new update.The version in the Arduino library manager seems very incomplete. Please make a new update.
[image]

theelims, 2024-02-01:
The easiest way is to download the zip file from github (Green CODE button) --> Download ZIP and import that into Arduino IDE. With the next official release the Arduino library will update itself.

Personal recommendation: switch to VSCode and Platformio.io

hasankorkmazdev, 2024-02-11:
Thanks for advice I try arduino after i research platform.io :)

closed 2024-08-10 by hoeken