hoeken/PsychicHttp #73: Incosistent filter() behaviour with ESPAsyncWebServer

theelims, 2024-02-03:
I am using the `filter()` callback to authenticate a Websocket connection. A a short lived JWT token is supplied as a search parameter like so:
```
/ws/lightState?access_token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c2VybmFtZSI6ImFkbWluIiwiYWRtaW4iOnRydWV9.7j33pzYbZ-Uf_irNQxzPG7UGTpKMV-VhTPTLA1RV_oM
```

The first issue I encountered when porting my code from ESPAsyncWebserver was that the search parameters are not resolved on calling the filter function yet. Only on `handleRequest` the parameters are evaluated. This can be fixed by calling `request->loadParams();` inside the filter function. However with the current way of concatenating the parameters in `loadParams()` it will introduce a bug with the parameters being present twice unable to evaluate properly when called the second time in `handleRequest`. For my use case I can ignore that bug.

The second issue I ran into was that the filter is called with every received frame again. But this time without headers or parameters. Thus my filter function would returns `false` and the websocket connection is closes with the first received message. I found a hack around this, as the request is initialized with the defaults like an empty string for the URI or method HTTP_DELETE (=0). This is not nice and potentially opens up an attack vector for my app. 

@hoeken Is this deviation from the ESPAsyncWebserver desired? If not it would be really nice if the request would be fully initialized when calling the filter function already. And received websocket frames don't need to go through that filter every time, as the connection passed the filter when it had been established.

hoeken, 2024-08-09:
you're correct, we should be passing a more or less full request to the filter.  I'll put this on my list for 2.0

hoeken, 2024-08-10:
i'm pushing a fix for this shortly - it will parse the parameters in the query string before filter(), and you won't need to call loadParams() to access them.  loadParams() will now actually only load parameters from the POST body.

that being said, i don't think we should doing anything with the body, so no post variables will be available.  The more we process the request before we get to filter(), the slower every single request is.

the filter() function is more for filtering on external things like wifi status, etc.

mathieucarbou, 2024-08-10:
Filters are much more used to select which handler to execute on the same endpoint (url + method) based on external conditions (not in the request). Everything linked to the request should be processed as part of the handler. That works, but...

But on the other hand @hoeken , filters can also be used to handle authentication in a common fashion. I've already seen that in a project and also this is very common in the web servers in java land where filters are used heavily to do some processing on the request that is not related to the application itself (auth, logging, etc)

@hoeken : there is also the discussion in the discord to consider, where filters are called on each websocket frame instead of just once when a connection happens.

So even if I agree with the perf loss, I disagree because it does not fit the conventional use of a filter, which is often used to apply a common processing on every request of an app 

It could also be adding informational headers, processing jwt tokens, oauth, etc.

hoeken, 2024-08-10:
@mathieucarbou That makes sense, but I think loading the request body before calling filters is a potentially expensive operation, not to mention some of the handlers further on have different size limits, error handling, and code to deal with buffered reading.

If needed, you can still call loadBody() and loadParams() to get POST data in the filter callback.

You can also access headers, cookies, session data, endpoint + handler objects, etc.

The only thing in your list you can't do is add headers.  

I'm not sure if you're talking request headers or response headers, but request headers is because esp-idf doesn't give us access to a list of headers.  we can only look up single headers.

as for the response headers, the response object isn't created until inside the handler callback due to the # of potential response objects that can be created.

There's a lot going on here, but I think currently its working pretty well.  Happy to address specific issues though.

closed 2024-08-10 by hoeken

mathieucarbou, 2024-08-10:
I was talking about the request headers and parameters, that I think we should have access to in filters.

We can have both world also: we can have 2 different kind of filters: a request filter and a filter which is not based on request:

```c++
typedef std::function<bool()> PsychicFilterFunction;
typedef std::function<bool(PsychicRequest* request)> PsychicRequestFilterFunction;
```

then in the handler and rewrite, something like that (supposing we cannot have both at the same time, which I would disagree)

```c++
    union {
        PsychicFilterFunction _requestFilter;
        PsychicRequestFilterFunction _filter;
    };
```

I would prefer: 

```c++
        PsychicFilterFunction _requestFilter;
        PsychicRequestFilterFunction _filter;
```

if _filter si set, call it.
then if _requestFilter is set, parse params and headers and call it.

mathieucarbou, 2024-08-10:
@hoeken : I have opened a draft PR showing how it could be done: https://github.com/hoeken/PsychicHttp/pull/145