hoeken/PsychicHttp #74: web socket onFrame crashes with log level i

ricdijk, 2024-02-10:
PsychicWebSocket.cpp lines 204 and 205:

`ESP_LOGI(PH_TAG, "ws_handler: httpd_handle_t=%p, sockfd=%d, client_info:%d", request->server(),
    httpd_req_to_sockfd(request->request()), httpd_ws_get_fd_info(request->server(), httpd_req_to_sockfd(request->request())));`

With log level 'Information', the library crashes on the call 'httpd_ws_get_fd_info(request->server(), httpd_req_to_sockfd(request->request()))'.

I simply took the the argument out of the log_i, not the solution, however works for me.

T94T, 2024-02-19:
Is there already a solution for this?

ricdijk, 2024-02-19:
No, I commented the lines out (did not need info for this part of the library). Solved the issue for me

dzungpv, 2024-03-01:
I have the same crash. Remove the logs on the PsychicWebSocket.cpp line 204 and it work fine. Here is the log for @hoeken  investigate:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4016c561  PS      : 0x00060930  A0      : 0x801d6466  A1      : 0x3ffd7e10  
0x4016c561: enum_function at /Users/vietdzung/esp/esp-idf-v4.4.6/components/esp_http_server/src/httpd_sess.c:86

A2      : 0x7fffffff  A3      : 0x3ffd7e60  A4      : 0x00000000  A5      : 0x3ffc4ad8  
A6      : 0x00000001  A7      : 0x0000000c  A8      : 0x00000004  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00060d23  A12     : 0x00060d20  A13     : 0x00060d23  
A14     : 0x0000000a  A15     : 0x3ffde7b8  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x80000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4016c55e:0x3ffd7e10 0x401d6463:0x3ffd7e40 0x4016c35d:0x3ffd7e60 0x4016e161:0x3ffd7eb0 0x4010cba0:0x3ffd7ed0 0x4010d196:0x3ffd7f90 0x4016be49:0x3ffd8010 0x4016c815:0x3ffd8040 0x4016aad4:0x3ffd8060 0x401d6463:0x3ffd8090 0x4016ae17:0x3ffd80b0 0x4016aebb:0x3ffd8100 0x4008d3fd:0x3ffd8120
0x4016c55e: enum_function at /Users/vietdzung/esp/esp-idf-v4.4.6/components/esp_http_server/src/httpd_sess.c:83

0x401d6463: httpd_sess_enum at /Users/vietdzung/esp/esp-idf-v4.4.6/components/esp_http_server/src/httpd_sess.c:50 (discriminator 1)

0x4016c35d: httpd_sess_get at /Users/vietdzung/esp/esp-idf-v4.4.6/components/esp_http_server/src/httpd_sess.c:181

0x4016e161: httpd_ws_get_fd_info at /Users/vietdzung/esp/esp-idf-v4.4.6/components/esp_http_server/src/httpd_ws.c:521

0x4010cba0: PsychicWebSocketHandler::handleRequest(PsychicRequest*) at /Users/vietdzung/dev/projects/esp32/wsproject/components/PsychicHttp/src/PsychicWebSocket.cpp:204 (discriminator 8)

0x4010d196: PsychicEndpoint::requestCallback(httpd_req*) at /Users/vietdzung/dev/projects/esp32/wsproject/components/PsychicHttp/src/PsychicEndpoint.cpp:72

0x4016be49: httpd_req_new at /Users/vietdzung/esp/esp-idf-v4.4.6/components/esp_http_server/src/httpd_parse.c:776

0x4016c815: httpd_sess_process at /Users/vietdzung/esp/esp-idf-v4.4.6/components/esp_http_server/src/httpd_sess.c:419 (discriminator 9)

0x4016aad4: httpd_process_session at /Users/vietdzung/esp/esp-idf-v4.4.6/components/esp_http_server/src/httpd_main.c:178 (discriminator 9)

0x401d6463: httpd_sess_enum at /Users/vietdzung/esp/esp-idf-v4.4.6/components/esp_http_server/src/httpd_sess.c:50 (discriminator 1)

0x4016ae17: httpd_server at /Users/vietdzung/esp/esp-idf-v4.4.6/components/esp_http_server/src/httpd_main.c:228

0x4016aebb: httpd_thread at /Users/vietdzung/esp/esp-idf-v4.4.6/components/esp_http_server/src/httpd_main.c:250 (discriminator 15)

0x4008d3fd: vPortTaskWrapper at /Users/vietdzung/esp/esp-idf-v4.4.6/components/freertos/port/xtensa/port.c:142
```

mhaberler, 2024-08-07:
the issue is an invalid argument list in `PsychicWebSocketHandler::handleRequest`  - [three format conversions,. four arguments](https://github.com/hoeken/PsychicHttp/blob/master/src/PsychicWebSocket.cpp#L204-L205):

````c++
  ESP_LOGI(PH_TAG, "ws_handler: httpd_handle_t=%p, sockfd=%d, client_info:%d", request->server(),
    httpd_req_to_sockfd(request->request()), httpd_ws_get_fd_info(request->server(), httpd_req_to_sockfd(request->request())));

````

mhaberler, 2024-08-07:
the cause is an invalid argument list: [three format conversions, four arguments](https://github.com/hoeken/PsychicHttp/blob/master/src/PsychicWebSocket.cpp#L204-L205):

```c++
  // reformatted for clarity
  ESP_LOGI(PH_TAG, "ws_handler: httpd_handle_t=%p, sockfd=%d, client_info:%d", 
    request->server(),
    httpd_req_to_sockfd(request->request()), 
    httpd_ws_get_fd_info(request->server(),  
    httpd_req_to_sockfd(request->request())));

```

commit 58509c3d (2024-08-10): uri is now parsed on update in PsychicResponse and will be available in filter() - fixes #74
closed 2024-08-08 by hoeken