hoeken/PsychicHttp #75: Send content larger than 1.7kB

BCsabaEngine, 2024-02-11:
I successfully integrated lib into my project (previously Async used). When I need to send a large response (larger than 1.7kB?), I get a reboot...

I tried chunk, double send
```
      response.send();
      return response.send();
```
but I cannot solve the problem. It is a dinamic content, so I cannot serve it from littleFs.

Reboot reason:
```
0x400d8c3f:0x3ffd9030 0x40107cf1x3ffd8f80 0x400d8acd:0x3ffd8fb0 ed reason: Stack canary watchpoint triggered (httpd) 
Core  1 register dump:
PC      : 0x4016d0e3  PS      : 0x00060936  A0      : 0x80164d45  A1      : 0x3ffd82c0  
A2      : 0x3ffd9314  A3      : 0x3ffd8600  A4      : 0x3ffd84e4  A5      : 0x00002000 0x400d8c3f:0x3ffd9030 0x40107cf1:0x3ffd90a0 0x40106a44:0x3ffd90d0 0x4010734a:0x3ffd9170 0x40105eac:0x3ffd9190 0x40174d22:0x3ffd91b0 0x40105fd0:0x3ffd91d0

ELF file SHA256: 2dd1fbc798a33dd9

```

zekageri, 2024-02-11:
Post a little bit more code pls. We need to see how you want to send the data. The crash indicates it was a stack canary error so the task what you are using has not enough stack allocated for the data. It isnt the libs fault. But post more code.

BCsabaEngine, 2024-02-11:
I have successfully made a "huge" response with extending JSON example with a for cycle:
```
      JsonDocument output;
      output["msg"] = "status";
      output["status"] = "success";
      for (int i = 0; i < 1000; i++)
        output[String(i)] = i;
      output["millis"] = millis();
```
It makes a 9.6kB response. So it is OK!

BCsabaEngine, 2024-02-11:
Where I would like to compose a 2k array, I cannot serve it. The simpliest, shortest example is:

```
    server.on("/large", HTTP_GET, [](PsychicHttpServerRequest * request)
    {
      const uint8_t data[2000] = {};

      PsychicHttpServerResponse response(request);
      response.setContentType("text/html");
      response.setContent(data, 2000);
      return response.send();
    });
```
I tried without const, failed too.
```
    server.on("/large", HTTP_GET, [](PsychicHttpServerRequest * request)
    {
      uint8_t data[2000] = {};

      PsychicHttpServerResponse response(request);
      response.setContentType("text/html");
      response.setContent(data, 2000);
      return response.send();
    });
```

pletch, 2024-02-11:
It looks like you are allocating buffer with ~ 2K on the stack but the default stack size for the esp-http-server is only 4K leading to a stack overflow.  Try initializing http server with increased stack size by including config.stack_size = 6144 (or 8192) prior to calling listen.  See [httpd_config options](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html#overview).

Reference: [https://www.esp32.com/viewtopic.php?t=10443](https://www.esp32.com/viewtopic.php?t=10443)

BCsabaEngine, 2024-02-11:
Without any stacksize config, I solved the situation with these code. Copy content (const or not) to a dynamically allocated array and it can be sent. After that need to release memory. So we need to copy it to a pointer memory instead of const.
```
      const uint8_t data[8000] = {};

      uint8_t * ptr = (uint8_t *)malloc(sizeof(data));
      for (int i = 0; i < sizeof(data); i++)
        ptr[i] = data[i];
      ptr[0] = 'A';
      ptr[1] = 'B';

      PsychicHttpServerResponse response(request);
      response.setContentType("image/png");
      response.setContent(ptr, sizeof(data));
      auto resp = response.send();
      delete ptr;
      return resp;
```

proddy, 2024-02-11:
Have you tried streaming it?

BCsabaEngine, 2024-02-11:
> Have you tried streaming it?

No, how can I?

closed 2024-02-11 by BCsabaEngine

proddy, 2024-02-11:
> > Have you tried streaming it?
> 
> 
> 
> No, how can I?

https://github.com/hoeken/PsychicHttp/pull/45

zekageri, 2024-02-11:
Dinamically allocated memory uses the heap. Therefore it doesnt care about the stack size. Thats why its working. Streaming could be a solution or websockets. But you can just stick with malloc or increase the http stack size with the config object.

BCsabaEngine, 2024-02-11:
Thanks Geri and Proddy. I now understand the memory usage and I will try the stream.

BCsabaEngine, 2024-02-11:
> Have you tried streaming it?

It works beautifully, I **successfully downloaded 500k data** without using a temporary buffer. Writing the const array directly to the stream is a great solution. Thanks for bringing it to my attention!

_I feel obligated to note that the current master source code is significantly different from Arduino v1.0.1. The classes are different, it has streamResponse (which is why I had to work from source code instead of package), it is much faster (in my opinion), there is less debug log._

proddy, 2024-02-11:
Glad it works! All the credit must go to @Chris--A for his amazing addition to this project

BCsabaEngine, 2024-02-12:
I made a package with which a Svelte/React app can be served from ESP32. Only and exclusively with PsychicHttp!
https://github.com/BCsabaEngine/svelteesp32