hoeken/PsychicHttp #77: Assert failed: tcpip_send_msg_wait_sem with Ethernet

JC-Electronics-Design, 2024-02-13:
Hello,

I'm trying to build a webserver in combination with ethernet instead of WiFi.
While testing I stumbled upon the following error when the function listen() is called. 
```
Start test webserver

assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace: 0x40083539:0x3ffb1f10 0x40088999:0x3ffb1f30 0x4008dd4d:0x3ffb1f50 0x400e616e:0x3ffb2080 0x400f0fdd:0x3ffb20b0 0x400f103c:0x3ffb20d0 0x400e5fe5:0x3ffb2120 0x400f7408:0x3ffb2140 0x400d3899:0x3ffb21c0 0x400d38cf:0x3ffb21e0 0x400d3979:0x3ffb2220 0x400d2b3e:0x3ffb2240 0x400daa5a:0x3ffb2290

  #0  0x40083539:0x3ffb1f10 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x40088999:0x3ffb1f30 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x4008dd4d:0x3ffb1f50 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x400e616e:0x3ffb2080 in tcpip_send_msg_wait_sem at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:455 (discriminator 1)
  #4  0x400f0fdd:0x3ffb20b0 in netconn_apimsg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/api_lib.c:136
  #5  0x400f103c:0x3ffb20d0 in netconn_new_with_proto_and_callback at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/api_lib.c:166
  #6  0x400e5fe5:0x3ffb2120 in lwip_socket at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/sockets.c:1774
  #7  0x400f7408:0x3ffb2140 in socket at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/include/lwip/sockets.h:656
      (inlined by) httpd_server_init at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_http_server/src/httpd_main.c:268
      (inlined by) httpd_start at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_http_server/src/httpd_main.c:423
  #8  0x400d3899:0x3ffb21c0 in PsychicHttpServer::_startServer() at .pio/libdeps/esp32dev/PsychicHttp/src/PsychicHttpServer.cpp:97
  #9  0x400d38cf:0x3ffb21e0 in PsychicHttpServer::_start() at .pio/libdeps/esp32dev/PsychicHttp/src/PsychicHttpServer.cpp:81
  #10 0x400d3979:0x3ffb2220 in PsychicHttpServer::listen(unsigned short) at .pio/libdeps/esp32dev/PsychicHttp/src/PsychicHttpServer.cpp:68
  #11 0x400d2b3e:0x3ffb2240 in setup() at src/main.cpp:154
  #12 0x400daa5a:0x3ffb2290 in loopTask(void*) at /Users/jonathancaes/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: 4ced09abcc31700f

Rebooting...
```

Here is the code I'm running:
```
#include <Arduino.h>
// #include <WiFi.h>
#include <LittleFS.h>
#include <ESPmDNS.h>
#include <esp_sntp.h>
#include <Ethernet.h>
#include <PsychicHttp.h>

FS* filesystem =      &LittleFS;

byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

PsychicHttpServer server;
PsychicEventSource events;

void setup() {
    delay(500);
    Serial.begin(1000000);
    delay(500);
    Serial.println("Start test webserver");
    
    // WiFi.mode(WIFI_STA);
    // WiFi.begin(ssid, password);
    // if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    //     Serial.printf("WiFi Failed!\n");
    //     return;
    // }

    // Serial.print("IP Address: ");
    // Serial.println(WiFi.localIP());

    Ethernet.init(5);
    delay(100);

    if(Ethernet.linkStatus() != LinkON) {
        Serial.println("Check ethernet connection!");
        // #ifdef USE_DISPLAY
        // link_status_warning(true);
        // #endif
        while(Ethernet.linkStatus() != LinkON) {
            delay(100);
        }
    }

    if(!Ethernet.begin(mac)) {
        Serial.println("Retrieving ip address via DHCP failed!\nReset MCU!");
        // Wait 100ms before resetting W5500 and MCU
        delay(100);
        ESP.restart();
    }

    if(!LittleFS.begin()){
        Serial.println("An Error has occurred while mounting SPIFFS");
        return;
    }

    //setup server config stuff here
    server.config.max_uri_handlers = 20; //maximum number of uri handlers (.on() calls)

    server.listen(80);
    
    server.serveStatic("/index.html", LittleFS, "/index.html");

    server.serveStatic("/front", LittleFS, "/front.jpg");

    server.serveStatic("/global_style.css", LittleFS, "/global_style.css");
    server.serveStatic("/index_style.css", LittleFS, "/index_style.css");
    server.serveStatic("/script.js", LittleFS, "/script.js");
}

void loop() {
}
```

platformio.ini file:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 1000000
upload_speed = 1000000
board_build.filesystem = littlefs
monitor_filters = esp32_exception_decoder
lib_deps = 
	https://github.com/hoeken/PsychicHttp
	arduino-libraries/Ethernet@^2.0.0
```

Everything works when using WiFi but once I try using ethernet (W5500 chip) I start getting this issue.
I also tried the Ethernet2.h library from Adafruit but this has the same results.

Hope someone can help me with this issue. Maybe I missed some setting? Maybe I have to use another Ethernet library?
Is it possible to implement ethernet compatibility?

Pablo2048, 2024-02-14:
Hi, IIRC it seems like you have not initialized ip stack (lwip). Try to just start wifi ap to achieve this (I've seen method to initialize lwip somewhere, but I can not find it right now :-( ). Maybe at least try to use `WiFi.onEvent(onEvent);` as seen in Ethernet library example.

JC-Electronics-Design, 2024-02-15:
@Pablo2048 
Hi Pablo,
I've tried `WiFi.onEvent(onEvent);`, `WiFi.softAP("ssid", "pass");` and `esp_netif_init();`.
With all of these I don't get the assert failed error but am still not able to access the webpage via ethernet. 
Any other suggestions?

johnnytolengo, 2024-02-16:
tcpip_adapter_init(); //Call this before start any Ethernet device

johnnytolengo, 2024-02-16:
first of all you must be sure that your ethernet is working properly, did you test it?

hoeken, 2024-02-16:
> @johnnytolengo <https://github.com/johnnytolengo> I don't get any errors
> when including tcpip_adapter_init();, but am still not able to connect to
> the server.
> @Pablo2048 <https://github.com/Pablo2048> The device does have an ip
> address before I call, in this case, listen(80).
> I don't get the assert error and am able to ping the device with the ip
> address the device is assigned but when I request a webpage for example
> index.html the device does not receive the request. It seems like
> PsychicHttp does not look for/consider requests from coming from ethernet
> client.

Can you test with the esp-idf official webserver example:
https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/simple/README.md

If that works, then try to figure out what they are doing that we aren't as
PsychicHttp is built on top of this library.

It looks like they do quite a bit of config setup which is located in this
shared file:
https://github.com/espressif/esp-idf/blob/master/examples/common_components/protocol_examples_common/eth_connect.c

I've never worked with esp32 + eth so this is just my 5 minute take on it.

----

Zach Smith

JC-Electronics-Design, 2024-02-16:
Yes, I pinged the ip address and that worked no problem. With other similar code that communicates with an MQTT server (with the same setup) I have no problems and everything is stable. 
Would you like me to post the code when I'm home? Maybe you see something?

hoeken, 2024-02-16:
Take a look at the 2nd link I posted and try some of that code. It's
probably some sort of init thing.

JC-Electronics-Design, 2024-02-17:
@hoeken Sorry didn't see your reply at first. 
I'll try the example via the esp-idf and see if that works. 
I've never worked with the esp-idf though so this will take a bit longer to figure out.

JC-Electronics-Design, 2024-02-18:
Here an update on the matter.
I've installed the ESP-IDF with VS code and got everything up and running. Configured the http_server simple example with the SDK Configuration Editor. Build, installed and monitored and am able to connect to the sever requesting the /hello page.

At first I had some issue with the version the W5500 chip returned, but I found that the SPI clock speed was too high for my prototype setup. Once I reduced the clock speed I didn't have any problems and everything just worked.

Below you can find a debug log of the example application.
```
--- esp-idf-monitor 1.4.0 on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7172
load:0x40078000,len:15532
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3904
entry 0x40080640
I (29) boot: ESP-IDF v5.2-dirty 2nd stage bootloader
I (29) boot: compile time Feb 18 2024 13:07:36
I (29) boot: Multicore bootloader
I (34) boot: chip revision: v3.1
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1a3b8h (107448) map
I (140) esp_image: segment 1: paddr=0002a3e0 vaddr=3ffb0000 size=02460h (  9312) load
I (143) esp_image: segment 2: paddr=0002c848 vaddr=40080000 size=037d0h ( 14288) load
I (151) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=41374h (267124) map
I (245) esp_image: segment 4: paddr=0007139c vaddr=400837d0 size=0c1e4h ( 49636) load
I (273) boot: Loaded app from partition at offset 0x10000
I (273) boot: Disabling RNG early entropy source...
I (284) cpu_start: Multicore app
D (285) cpu_start: Pro cpu up
D (285) cpu_start: Starting app cpu, entry point is 0x40081230
0x40081230: call_start_cpu1 at /Users/jonathancaes/esp/v5.2/esp-idf/components/esp_system/port/cpu_start.c:183

D (0) cpu_start: App cpu up
D (294) clk: RTC_SLOW_CLK calibration value: 3189811
I (306) cpu_start: Pro cpu start user code
I (306) cpu_start: cpu freq: 160000000 Hz
I (306) cpu_start: Application information:
I (311) cpu_start: Project name:     simple
I (315) cpu_start: App version:      1
I (320) cpu_start: Compile time:     Feb 18 2024 13:07:34
I (326) cpu_start: ELF file SHA256:  2d2a8d560...
I (331) cpu_start: ESP-IDF:          v5.2-dirty
I (337) cpu_start: Min chip rev:     v0.0
I (341) cpu_start: Max chip rev:     v3.99 
I (346) cpu_start: Chip rev:         v3.1
D (351) memory_layout: Checking 9 reserved memory ranges:
D (356) memory_layout: Reserved memory range 0x3ff82000 - 0x3ff82000
D (363) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (369) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb3780
D (376) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (382) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (388) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (395) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at /Users/jonathancaes/esp/v5.2/esp-idf/components/xtensa/xtensa_vectors.S:2027
[... 327 lines cut ...]
D (8583) httpd_txrx: httpd_send_all: sent = 2
D (8593) w5500.mac: receive len=60
D (8593) httpd_txrx: httpd_send_all: sent = 2
D (8603) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:4 posted to loop 0x3ffb8074
D (8613) w5500.mac: receive len=60
D (8613) httpd_txrx: httpd_send_all: sent = 12
D (8623) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:6 posted to loop 0x3ffb8074
I (8633) example: Request headers lost
D (8633) httpd_sess: httpd_sess_process: httpd_req_delete
D (8643) httpd_sess: httpd_sess_process: success
D (8643) httpd: httpd_server: processing listen socket 54
D (8653) httpd: httpd_accept_conn: newfd = 58
D (8653) httpd_sess: httpd_sess_new: fd = 58
D (8653) httpd_sess: httpd_sess_new: active sockets: 2
D (8663) httpd: httpd_accept_conn: complete
D (8663) event: no handlers have been registered for event ESP_HTTP_SERVER_EVENT:2 posted to loop 0x3ffb8074
D (8663) httpd: httpd_server: doing select maxfd+1 = 59
D (9163) w5500.mac: receive len=333
D (9163) w5500.mac: receive len=353
D (9453) w5500.mac: receive len=226
D (10313) w5500.mac: receive len=419
D (10313) w5500.mac: receive len=439
```
I have not yet found why it is not working with the PsychicHttp library but thought I might post this log already. Maybe one of you, more clever people, can see what is going on.
I'll keep searching for the solution in the meantime.

Pablo2048, 2024-02-18:
Why it is not working is obvious: Forget about Arduino Ethernet library and use this one instead https://github.com/espressif/arduino-esp32/tree/master/libraries/Ethernet , start with this example https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_W5500_Arduino_SPI/ETH_W5500_Arduino_SPI.ino and after making it work just add PsychicHttp...

JC-Electronics-Design, 2024-02-18:
Ok, I'll try that. Thanks!

closed 2024-08-09 by hoeken