hoeken/PsychicHttp #80: ArduinoJson7

woodenplastic, 2024-03-07:
Hi,

do you plan to make ArduinoJson7 happen?

All the best
Adrian

zekageri, 2024-03-08:
It is supported

hoeken, 2024-03-08:
I really need to make a v1.1 release. Sorry I've been slacking.  Use git
for now plz.

papakpmartin, 2024-03-28:
I'm using...

```yaml
lib_deps = 
  bblanchon/ArduinoJson@^7.0.3
  https://github.com/hoeken/PsychicHttp
```

...and I'm seeing...

```
Library Manager: Installing bblanchon/ArduinoJson @ ^7.0.3
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@7.0.4 has been installed!
Library Manager: PsychicHttp@1.0.1+sha.e878a9f has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ ^6.21.4
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.21.5 has been installed!
Library Manager: Installing bblanchon/ArduinoTrace @ ^1.2.0
Unpacking  [####################################]  100%
Library Manager: ArduinoTrace@1.2.0 has been installed!
Library Manager: Installing plageoj/UrlEncode @ ^1.0.1
Unpacking  [####################################]  100%
Library Manager: UrlEncode@1.0.1 has been installed!
```

Then when I try to use 7-ish code like `JsonDocument doc;` I get errors like...

```
src/main.cpp:201:16: error: 'ArduinoJson::V6215PB2::JsonDocument::JsonDocument()' is protected within this context
```

...which look like it wants ArduinoJson 6-ish syntax.

`PsychicHttp@1.0.1+sha.e878a9f` looks like the latest greatest.

How might I get this working?

papakpmartin, 2024-03-28:
FWIW, I did also try adding...

```yaml
build_flags =
  -D ARDUINOJSON_VERSION_MAJOR=7
```

...but I see the same result.

hitecSmartHome, 2024-03-28:
Arduinojson 7 is definitely working with the latest PsychicHttp. Try to download both from github and put it into your lib folder.

woodenplastic, 2024-03-28:
lib_deps = 
	https://github.com/me-no-dev/AsyncTCP
	https://github.com/hoeken/PsychicHttp
	https://github.com/bblanchon/ArduinoStreamUtils
	bblanchon/ArduinoJson @ ^7.0.4
	
	thats working for me.

papakpmartin, 2024-04-03:
Just a heads up that even though I'm up and running using ArduinoJson7, PsychicHttp is nevertheless bringing in `bblanchon/ArduinoJson@^6.21.4` as a dependency.

```
Library Manager: Installing git+https://github.com/hoeken/PsychicHttp
git version 2.37.1
Cloning into '/Users/ken.martin/.platformio/.cache/tmp/pkg-installing-c5l4ux3r'...
remote: Enumerating objects: 166, done.
remote: Counting objects: 100% (166/166), done.
remote: Compressing objects: 100% (131/131), done.
remote: Total 166 (delta 36), reused 84 (delta 16), pack-reused 0
Receiving objects: 100% (166/166), 349.42 KiB | 3.18 MiB/s, done.
Resolving deltas: 100% (36/36), done.
Library Manager: PsychicHttp@1.0.1+sha.e878a9f has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ ^6.21.4
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.21.5 has been installed!
Library Manager: Installing bblanchon/ArduinoTrace @ ^1.2.0
Unpacking  [####################################]  100%
Library Manager: ArduinoTrace@1.2.0 has been installed!
Library Manager: Installing plageoj/UrlEncode @ ^1.0.1
Unpacking  [####################################]  100%
Library Manager: UrlEncode@1.0.1 has been installed!
```

hitecSmartHome, 2024-04-03:
Oh. It must be because of the library.json dependecy tree. It should be adjusted.

closed 2024-08-09 by hoeken