hoeken/PsychicHttp #82: PsychicHttp.h: No such file or directory on PlatformIO

mLamneck, 2024-04-07:
Hi, I'm the author of [this](https://github.com/mLamneck/SDDS_ESP_Extension) library and I consider using PsychicHttp as a replacement for the ESPAsyncWebServer dependency. However, when trying to compile the provided example with PlatformIO, I encounter a compile error:

> PsychicHttp.h: No such file or directory

Looking at the created `c_cpp_properties.json`, there is not much added in the path array. Last line added is about the ArduinoJson library. So perhaps this library is causing the issue.

Anyway, I just noticed that PsychicHttp only supports ESP32? Do you have any plans to support ESP8266 as well?

What about licensing when using this library in my own. I'm undecided between GPL or LGPL but I guess both would be possible!?

hoeken, 2024-04-08:
Is this from the GitHub main branch?  I'm a bit behind and need to make a
1.1 release ASAP.

Currently I'm only supporting esp32 but I would be happy to accept pulls
for esp8266 compatibility.

As far as licensing goes it's lgpl so feel free to use it for whatever,
commercial ok. We just ask any changes to the library itself get rolled
back into PsychicHttp.

mLamneck, 2024-04-08:
I tried

```
lib_deps = hoeken/PsychicHttp@^1.0.0
lib_deps = hoeken/PsychicHttp@^1.0.1
lib_deps = hoeken/PsychicHttp
```
and

`lib_deps = https://github.com/hoeken/PsychicHttp`

as well as the provided example for PlatformIO from your repository.

closed 2024-08-09 by hoeken