hoeken/PsychicHttp #85: invalid usage of httpd_ws_send_frame_async

florentbr, 2024-04-23:
Release 1.0.1, the websocket class is using `httpd_ws_send_frame_async` to send an arbitrary message :

```cpp
esp_err_t PsychicWebSocketClient::sendMessage(httpd_ws_frame_t * ws_pkt)
{
  return httpd_ws_send_frame_async(this->server(), this->socket(), ws_pkt);
} 
```

and `httpd_ws_send_frame` to send a message within a reply:

```cpp
esp_err_t PsychicWebSocketRequest::reply(httpd_ws_frame_t * ws_pkt)
{
  return httpd_ws_send_frame(this->_req, ws_pkt);
} 
```

From the IDF source it looks like `httpd_ws_send_frame_async` and `httpd_ws_send_frame_async` are only for replies and callbacks from the queue.

So sending an arbitrary message with `PsychicWebSocketHandler::sendAll` should either call `httpd_ws_send_data` or `httpd_ws_send_data_async` or queue the messages with [`httpd_queue_work`](https://github.com/espressif/esp-idf/blob/release/v5.3/examples/protocols/https_server/wss_server/main/wss_server_example.c)

hoeken, 2024-08-11:
fixed in cc81fa18af16ad459ee89957a6029f4793f8e4d9

closed 2024-08-11 by hoeken

florentbr, 2024-08-11:
I think #97 is related. I could be wrong but from I've seen from the IDF sources, messages need to be queued if they are not sent from a reply.

commit cc81fa18 (2024-08-11): use proper method to send websocket data outside of response.  fixes #85