hoeken/PsychicHttp #90: OTA update example for PsychicHttp

06GitHub, 2024-05-27:
**OTA update example for PsychicHttp**
For those interested, an example of OTA (Over The Air) update implementation for PsychicHttp is provided below, using Arduino IDE.

**Requirements**
Requirements for my project are :
- OTA update for code (firmware)
- OTA update for data (littlefs)
- manual restart of ESP32, triggered by User (no automatic restart after code or data file upload)

**Implementation**
OTA update relies on handler PsychicUploadHandler.

Screenshots and Code are provided below.

**Credits**
https://github.com/hoeken/PsychicHttp/blob/master/src/PsychicUploadHandler.cpp

https://github.com/hoeken/PsychicHttp/issues/30 

**Configuration**
Arduino IDE 1.8.19
arduino-32 v2.0.15
ESP32S3

**Files Structure**

```
psychichttp_ota
     data
          | update.html
     psychichttp_ota.ino
```

**SCREENSHOTS**
**Update code (firmware)**
[image]

[image]
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[   267][I][psychichttp_ota.ino:57] connectToWifi(): [OTA] [WiFi] WiFi is disconnected
[   775][I][psychichttp_ota.ino:60] connectToWifi(): [OTA] [WiFi] WiFi is connected, IP address 192.168.1.50
[  9208][I][PsychicHttpServer.cpp:236] openCallback(): [psychic] New client connected 51
[  9218][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 867274
[  9230][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 861984
[  9242][I][psychichttp_ota.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[  9252][I][psychichttp_ota.ino:133] operator()(): [OTA] update begin, filename code.bin
[  9395][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 856240
[  9407][I][psychichttp_ota.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[  9435][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 850496
[  9447][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 844752
[  9459][I][psychichttp_ota.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[  9488][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 839008
[  9500][I][psychichttp_ota.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[  9528][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 833264
[  9540][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 827520
[  9551][I][psychichttp_ota.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[snip]
[ 15625][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 13308
[ 15637][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 7564
[ 15647][I][psychichttp_ota.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[ 15790][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 1820
[ 15800][I][psychichttp_ota.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 1
[... 379 lines cut ...]
			document.getElementById("progressBar").value = Math.round(percent);		
			document.getElementById("status").innerHTML = Math.round(percent) + "% uploaded...";
			if (percent >= 100) {
				document.getElementById("status").innerHTML = "Writing file ...";
			}
		}
		
		function errorHandler(event) {
			document.getElementById("status").innerHTML = "<b style='color:red'>Error event catched by errorHandler !</b>";			
		}
		
		function esprestartButton() {
			/* restart ESP */
			var urltocall = "/restart";
			xhr=new XMLHttpRequest();
			xhr.onreadystatechange = function() {
				if (this.readyState == 4) document.getElementById("status").innerHTML = xhr.responseText;
			};
			xhr.open("POST", urltocall, false); // trigger restart request with file parameter
			xhr.send();	
		}
		
	</script>
</html>
```

hitecSmartHome, 2024-06-12:
I always got some weird crash on like 70-80% on upload.
Doesn't matter if i upload the firmware or the littlefs.bin

```cpp
Decoding stack results
0x4008381e: panic_abort at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_system\panic.c line 408
0x401c6b45: esp_system_abort at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_system\esp_system.c line 137
0x4008fbf6: abort at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\newlib\abort.c line 46
0x401c6c27: task_wdt_isr at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_system\task_wdt.c line 176
0x4008a157: vPortExitCritical at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\freertos\port\xtensa\include/freertos/portmacro.h line 571
0x40081a13: esp_intr_noniram_enable at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\freertos\port\xtensa\include/freertos/portmacro.h line 605
0x40084e7b: spi_flash_enable_interrupts_caches_and_other_cpu at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\spi_flash\cache_utils.c line 208
0x40085c89: cache_enable at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\spi_flash\spi_flash_os_func_app.c line 68
0x40085cc7: spi1_end at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\spi_flash\spi_flash_os_func_app.c line 131
0x400887a9: spiflash_end_default at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\spi_flash\esp_flash_api.c line 154
0x400853de: flash_end_flush_cache at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\spi_flash\esp_flash_api.c line 192
0x40085812: esp_flash_erase_region at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\spi_flash\esp_flash_api.c line 658
0x401e73cf: esp_partition_erase_range at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\spi_flash\partition.c line 540
0x401b83c1: EspClass::partitionEraseRange(esp_partition_t const*, unsigned int, unsigned int) at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp line 436
0x4019e5f7: UpdateClass::_writeBuffer() at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/Update/src/Updater.cpp line 237
0x4019e731: UpdateClass::write(unsigned char*, unsigned int) at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/Update/src/Updater.cpp line 329
0x4017d470: std::_Function_handler   >::_M_invoke(const std::_Any_data &, PsychicRequest *&&, const String &, unsigned long long &&, unsigned char *&&, unsigned int &&, bool &&) at src/Server/httpUpdater.cpp line 40
0x401a3ae1: std::function ::operator()(PsychicRequest*, String const&, unsigned long long, unsigned char*, unsigned int, bool) const at c:\users\pc\.platformio\packages\toolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h line 260
0x401a3bd6: PsychicUploadHandler::_handleUploadByte(unsigned char, bool) at lib/PsychicHttp/src/PsychicUploadHandler.cpp line 212
0x401a3c95: PsychicUploadHandler::_parseMultipartPostByte(unsigned char, bool) at lib/PsychicHttp/src/PsychicUploadHandler.cpp line 243
0x401a45da: PsychicUploadHandler::_multipartUploadHandler(PsychicRequest*) at lib/PsychicHttp/src/PsychicUploadHandler.cpp line 189
0x401a465b: PsychicUploadHandler::handleRequest(PsychicRequest*) at lib/PsychicHttp/src/PsychicUploadHandler.cpp line 58
0x401a4e7e: PsychicEndpoint::requestCallback(httpd_req*) at lib/PsychicHttp/src/PsychicEndpoint.cpp line 72
0x401c3906: httpd_uri at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_uri.c line 329
0x401c28ec: httpd_req_new at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_parse.c line 659
0x401c3006: httpd_sess_process at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_sess.c line 419
0x401c2088: httpd_process_session at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_main.c line 178
0x402601b2: httpd_sess_enum at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_sess.c line 53
0x401c2170: httpd_thread at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_main.c line 228
```

hitecSmartHome, 2024-06-12:
On regular firmware

```cpp
Decoding stack results
0x4008381e: panic_abort at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_system\panic.c line 408
0x401c6b45: esp_system_abort at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_system\esp_system.c line 137
0x4008fbf6: abort at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\newlib\abort.c line 46
0x401c6c27: task_wdt_isr at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_system\task_wdt.c line 176
0x401a3b9e: PsychicUploadHandler::_handleUploadByte(unsigned char, bool) at lib/PsychicHttp/src/PsychicUploadHandler.cpp line 207
0x401a3c95: PsychicUploadHandler::_parseMultipartPostByte(unsigned char, bool) at lib/PsychicHttp/src/PsychicUploadHandler.cpp line 243
0x401a45da: PsychicUploadHandler::_multipartUploadHandler(PsychicRequest*) at lib/PsychicHttp/src/PsychicUploadHandler.cpp line 189
0x401a465b: PsychicUploadHandler::handleRequest(PsychicRequest*) at lib/PsychicHttp/src/PsychicUploadHandler.cpp line 58
0x401a4e7e: PsychicEndpoint::requestCallback(httpd_req*) at lib/PsychicHttp/src/PsychicEndpoint.cpp line 72
0x401c3906: httpd_uri at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_uri.c line 329
0x401c28ec: httpd_req_new at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_parse.c line 659
0x401c3006: httpd_sess_process at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_sess.c line 419
0x401c2088: httpd_process_session at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_main.c line 178
0x402601b2: httpd_sess_enum at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_sess.c line 53
0x401c2170: httpd_thread at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_main.c line 228
```

06GitHub, 2024-06-12:
Are you using Arduino IDE or ESP-IDF ? If ESP-IDF, is version **ESP-IDF v4.4.7** ?

This OTA example has been developed on top of Arduino IDE 1.8.19 and arduino-esp32 v2.0.15.  Arduino-esp32 release v2.0.15 is based on **ESP-IDF v4.4.7** https://github.com/espressif/arduino-esp32/releases/tag/2.0.15

PsychichHttp version is v1.1.0. ESP is ESP-S3.

So maybe crash is due to some differences between this version and version you are using ?

Note that example described in this issue has been submitted as a PR https://github.com/hoeken/PsychicHttp/pull/94

hitecSmartHome, 2024-06-12:
I'm using ESP-Wrover-E with PSram.
Arduino is 2.0.17
IDF version is 4.4.7

I'm using Arduino as a component of IDF in PlatformIO

06GitHub, 2024-06-13:
I have done a new test today with (almost) same configuration as yours : 
- ESP-WRover with PSram (previously ESP32-S3)
- Arduino 2.0.17 (previously 2.0.15)

The only difference is that I am using Arduino IDE 1.8.19, which is built on top of IDF 4.4.7, and you are using ESP-IDF 4.4.7.
So both IDEs use same version IDF 4.4.7.

Test is ok : both code and firmware can be updated using OTA, see for instance below output for code update.

Which partition are you using ? Is partition size large enough for OTA, and contains otadata, app0 and app1 ? Maybe there is a glitch in partition you are using, which could prevent ota process to complete ?

I am using default partition called "Default 4Mb with SPIFFS"  on Arduino IDE :
Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x140000,
app1,     app,  ota_1,   0x150000,0x140000,
spiffs,   data, spiffs,  0x290000,0x160000,
coredump, data, coredump,0x3F0000,0x10000,

**output for code update**
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[   451][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   586][I][psychichttp_ota_issue90.ino:57] connectToWifi(): [OTA] [WiFi] WiFi is disconnected
[  1094][I][psychichttp_ota_issue90.ino:60] connectToWifi(): [OTA] [WiFi] WiFi is connected, IP address 192.168.1.31
[ 13175][I][PsychicHttpServer.cpp:236] openCallback(): [psychic] New client connected 51
[ 13186][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 936264
[ 13200][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 930973
[ 13212][I][psychichttp_ota_issue90.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[ 13223][I][psychichttp_ota_issue90.ino:133] operator()(): [OTA] update begin, filename code.bin
[ 13371][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 925229
[ 13385][I][psychichttp_ota_issue90.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[ 13414][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 919485
[ 13428][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 913741
[ 13439][I][psychichttp_ota_issue90.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[ 13474][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 907997
[ 13487][I][psychichttp_ota_issue90.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[ 13517][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 902253
[ 13530][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 896509
[ 13553][I][psychichttp_ota_issue90.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[ 13583][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 895073
[ 13596][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 889329
[ 13608][I][psychichttp_ota_issue90.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[ 13639][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 883585
[ 13652][I][psychichttp_ota_issue90.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[ 13682][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 877841
[ 13696][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 872097
[ 13708][I][psychichttp_ota_issue90.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[ 13739][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 866353
[snip]
[ 20500][I][psychichttp_ota_issue90.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[ 20531][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 21985
[ 20544][I][psychichttp_ota_issue90.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[ 20574][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 16241
[ 20587][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 10497
[ 20598][I][psychichttp_ota_issue90.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[ 20809][I][PsychicUploadHandler.cpp:164] _multipartUploadHandler(): [psychic] Remaining size : 4753
[ 20820][I][psychichttp_ota_issue90.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 0
[ 20851][I][psychichttp_ota_issue90.ino:128] operator()(): [OTA] updateHandler->onUpload _error 0 Update.hasError() 0 last 1
[ 21131][I][psychichttp_ota_issue90.ino:165] operator()(): [OTA] Update Success: 936032 written
[ 21140][I][psychichttp_ota_issue90.ino:184] operator()(): [OTA] Update code or data OK Update.errorString() No Error
[ 21154][I][PsychicHttpServer.cpp:236] openCallback(): [psychic] New client connected 52
[ 21163][I][PsychicHttpServer.cpp:262] closeCallback(): [psychic] Client disconnected 52
```

hitecSmartHome, 2024-06-14:
I have a custom partition table with several partitions, including app0 and app1.
The firmware goes on a regular route to either app0 or app1 which freeRTOS decides.
The file system goes to a "system" partition. I'm sure it is enough, the firmware download works with OTA and with http requests too. I wanted to implement this as a third option but I get crashes on 90% of the time.

Interestingly, the firmware succeeds sometimes and everything goes well 4 out of 10 times.
The file system can not make it a single time. Usually it halts at 80-90% upload.
The freeRTOS watchdog complains that the idle task has not enough time to switch tasks.

hoeken, 2024-08-09:
seems fixed?

closed 2024-08-09 by hoeken