hoeken/PsychicHttp #93: "No Server certificate supplied"

gradhocJcalvo, 2024-05-28:
Conditions: With new update from espressif on Arduino IDE enviroment, (3.0 board version). And using ESP32-S3. Default partition scheme. SSL hast to be enabled on example. With http it works fine.

BUG: E (1160) esp_https_server: No Server certificate supplied

-With example of the library.

-LittleFS  uploaded (no errors when reading it).

Thank you very much.

connornishijima, 2024-05-28:
I have this issue as well, documented in my own repo's issue here:

https://github.com/Lixie-Labs/Emotiscope/issues/86

(I'm also on espressif/arduino-esp32 3.0.0 / IDF 5.x)

mhaberler, 2024-07-23:
I have the same issue, see https://github.com/hoeken/PsychicHttp/issues/103

mhaberler, 2024-07-24:
https://github.com/hoeken/PsychicHttp/pull/104 should fix this problem

mhaberler, 2024-08-07:
@connornishijima @gradhocJcalvo

a fix has been merged, could you verify this works for you?

hoeken, 2024-08-09:
fixed

closed 2024-08-09 by hoeken