hoeken/PsychicHttp #95: #89 Pull request totally broke the lib

hitecSmartHome, 2024-05-30:
I have downloaded the new version with the new fixes.
My web server no longer showed up.
I reverted back to my "old" PsychicHttp version, applyed only the "memory leak fix" and it was the same.
Webserver never answers to the requests. 

First I did not notice it, but I cleared my cache on my browser and the web never loads. In fact it seems that the webserver never reaches the point to answer to requests.

I deleted the "memory leak fix" from the lib and all working fine again.

https://github.com/dzungpv/PsychicHttp/commit/ec80969f28c032305a82d8e65783b84ebe4dcbed

Chris--A, 2024-05-31:
@hoeken 
Rather than merging #96 as a fix, PR #89 should be reverted, and resubmitted as a new PR once thoroughly tested. Additionally to keep my TemplatePrinter included, reopen and merge #56 not only to maintain proper attribution to contributors (https://github.com/hoeken/PsychicHttp/pull/89#issuecomment-2141189840), but to ensure PR's are specific to a need, in cases like this when something needs to be reverted.

hoeken, 2024-05-31:
> proper attribution to contributors (#89 (comment)

Hey all, sorry about messing this one up. I was a bit overeager on the pull
requests and didn't fully test them.

I'll unwind this ASAP.

hitecSmartHome, 2024-06-04:
No problem. Is there a memory overflow or not?

closed 2024-06-14 by hitecSmartHome